strongtz/i915-sriov-dkms #1: SR-IOV cannot be turned on.

Alion548, 2022-09-27:
Thank you very much for making dkms, but I have a problem using it.

```bash
uname -a
Linux 5.19.11-xm1.0.fc36.x86_64 #1 SMP Sun Sep 25 00:27:08 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

dkms status                       
i915-sriov/dkms, 5.19.11-xm1.0.fc36.x86_64, x86_64: installed (original_module exists)

modinfo i915 | grep extra                                  
filename:       /lib/modules/5.19.11-xm1.0.fc36.x86_64/extra/i915.ko

cat /etc/default/grub                           
GRUB_TIMEOUT=1
GRUB_DISTRIBUTOR="$(sed 's, release .*$,,g' /etc/system-release)"
GRUB_DEFAULT=saved
GRUB_DISABLE_SUBMENU=true
GRUB_TERMINAL_OUTPUT="console"
GRUB_CMDLINE_LINUX="selinux=0 mitigations=off intel_iommu=on iommu=pt i915.enable_guc=7 rhgb quiet"
GRUB_DISABLE_RECOVERY="true"
GRUB_ENABLE_BLSCFG=true
GRUB_RECORDFAIL_TIMEOUT=$GRUB_TIMEOUT
```

I get the following error when enabling the SR-IOV feature.

```bash
echo 3 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs 
echo: write error

dmesg
[   50.838155] i915 0000:00:02.0: driver does not support SR-IOV configuration via sysfs
```

strongtz, 2022-09-27:
There might be some problem with your DKMS configuration, and you're probably **still using the old driver**.

> [    1.503002] i915 0000:00:02.0: [drm] Incompatible option enable_guc=7 - undocumented flag

The flag is certainly documented in this driver.

anyway, here's what should happen with this driver

```
[    4.626212] i915: loading out-of-tree module taints kernel.
[    4.640255] i915: module verification failed: signature and/or required key missing - tainting kernel
[    4.773789] Setting dangerous option enable_guc - tainting kernel
[    4.828300] i915 0000:00:02.0: i915_virtualization_probe: entry
[    4.829034] i915 0000:00:02.0: i915_sriov_probe: entry
[    4.829804] i915 0000:00:02.0: i915_sriov_probe: gen12_pci_capability_is_vf in
[    4.830666] i915 0000:00:02.0: i915_sriov_probe: dev_is_pf in
[    4.831490] i915 0000:00:02.0: pf_verify_readiness: entry
[    4.832287] i915 0000:00:02.0: pf_verify_readiness: pf_reduce_totalvfs
[    4.832659] i915 0000:00:02.0: i915_sriov_probe: I915_IOV_MODE_SRIOV_PF
[    4.833028] i915 0000:00:02.0: i915_sriov_probe returns: 3
[    4.833397] i915 0000:00:02.0: Running in SR-IOV PF mode
[    4.834368] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    4.835133] i915 0000:00:02.0: [drm] Transparent Hugepage mode 'huge=within_size'
[    4.835871] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=io+mem
[    4.839486] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
[    4.946002] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_70.1.1.bin version 70.1
[    4.946007] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc_7.9.3.bin version 7.9
[    4.948160] i915 0000:00:02.0: [drm] HuC authenticated
[    4.948460] i915 0000:00:02.0: [drm] GuC submission enabled
[    4.948461] i915 0000:00:02.0: [drm] GuC SLPC enabled
[    4.948775] i915 0000:00:02.0: [drm] GuC RC: enabled
[    4.949864] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    4.950286] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    4.979109] i915 0000:00:02.0: 7 VFs could be associated with this PF
```

Alion548, 2022-09-27:
Solved, xanmod does not support dkms.

closed 2022-09-27 by Alion548

xTechon, 2022-10-06:
I have the same issue, while I have a kernel that supports dkms, it would be very helpful to briefly list the process of switching from the old module to the dkms version. I am using systemd-boot for the drive encryption

```
uname -a
Linux DellXPS13 5.19.12-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 28 Sep 2022 13:21:25 +0000 x86_64 GNU/Linux

dkms status | grep i915
i915-sriov-dkms/5.15.49, 5.19.12-arch1-1, x86_64: installed (original_module exists)

modinfo i915 | grep filename
filename:       /lib/modules/5.19.12-arch1-1/updates/dkms/i915.ko.zst

modprobe --show-depends i915
insmod /lib/modules/5.19.12-arch1-1/kernel/drivers/char/agp/intel-gtt.ko.zst
insmod /lib/modules/5.19.12-arch1-1/kernel/drivers/media/cec/core/cec.ko.zst
insmod /lib/modules/5.19.12-arch1-1/kernel/drivers/gpu/drm/display/drm_display_helper.ko.zst
insmod /lib/modules/5.19.12-arch1-1/kernel/drivers/gpu/drm/ttm/ttm.ko.zst
insmod /lib/modules/5.19.12-arch1-1/kernel/drivers/acpi/video.ko.zst
insmod /lib/modules/5.19.12-arch1-1/kernel/drivers/gpu/drm/drm_buddy.ko.zst
insmod /lib/modules/5.19.12-arch1-1/updates/dkms/i915.ko.zst enable_guc=7

cat /boot/loader/entries/arch.conf
title Arch Linux
linux /vmlinuz-linux
initrd /intel-ucode.img
initrd /initramfs-linux.img
options cryptdevice=UUID=[censored]:vg0 root=[censored] resume=[censored] rw intel_pstate=no_hwp
options CONFIG_CFG80211_USE_KERNEL_REGDB_KEYS=yes intel_iommu=on iommu=pt i915.enable_guc=7
```
modinfo also shows documentation for flagging `enable_guc=4` but that gave the same error regardless (i'll skip the description)
```
modinfo i915 | grep enable_guc
parm:           enable_guc:...(-1=auto [default], 0=disable, 1=GuC submission, 2=HuC load, 4=SR-IOV PF) (int)
```
my log output, I have an encrypted drive, so I censored a few things
```
dmesg | grep i915
[    0.000000] Command line: initrd=\intel-ucode.img initrd=\initramfs-linux.img cryptdevice=UUID=[censored]:vg0 root=[censored] resume=[censored] rw intel_pstate=no_hwp CONFIG_CFG80211_USE_KERNEL_REGDB_KEYS=yes intel_iommu=on iommu=pt i915.enable_guc=7
[    0.032921] Kernel command line: initrd=\intel-ucode.img initrd=\initramfs-linux.img cryptdevice=UUID=[censored]:vg0 root=[censored] resume=[censored] rw intel_pstate=no_hwp CONFIG_CFG80211_USE_KERNEL_REGDB_KEYS=yes intel_iommu=on iommu=pt i915.enable_guc=7
[    0.876029] i915 0000:00:02.0: [drm] Incompatible option enable_guc=7 - undocumented flag
[    0.876335] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    0.876446] i915 0000:00:02.0: vgaarb: deactivate vga console
[    0.876476] i915 0000:00:02.0: [drm] Transparent Hugepage mode 'huge=within_size'
[    0.878276] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    0.879506] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/tgl_dmc_ver2_12.bin (v2.12)
[    1.023844] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_70.1.1.bin version 70.1
[    1.023847] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc_7.9.3.bin version 7.9
[    1.027756] i915 0000:00:02.0: [drm] HuC authenticated
[    1.028287] i915 0000:00:02.0: [drm] GuC submission enabled
[    1.028288] i915 0000:00:02.0: [drm] GuC SLPC enabled
[    1.028903] i915 0000:00:02.0: [drm] GuC RC: enabled
[    1.032153] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    1.047889] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    1.055570] fbcon: i915drmfb (fb0) is primary device
[    1.081821] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[   15.301784]  encrypted_keys trusted asn1_encoder tee tpm rng_core dm_mod serio_raw rtsx_pci_sdmmc atkbd crct10dif_pclmul mmc_core libps2 vivaldi_fmap crc32_pclmul ghash_clmulni_intel aesni_intel crypto_simd nvme cryptd xhci_pci rtsx_pci nvme_core xhci_pci_renesas i8042 serio i915 intel_gtt drm_buddy video drm_display_helper cec ttm ext4 crc32c_generic crc32c_intel crc16 mbcache jbd2
[   15.908531] i915 0000:00:02.0: [drm] Selective fetch area calculation failed in pipe A
[   16.033609] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[   16.059241] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[   16.353601] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[  616.158827] i915 0000:00:02.0: driver does not support SR-IOV configuration via sysfs
```
I also made sure my laptop GPU supports SR-IOV to begin with,
```
lspci -v
00:02.0 VGA compatible controller: Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] (rev 01) (prog-if 00 [VGA controller])
	DeviceName: Onboard - Video
	Subsystem: Dell Device 0991
	Flags: bus master, fast devsel, latency 0, IRQ 128, IOMMU group 1
	Memory at 6052000000 (64-bit, non-prefetchable) [size=16M]
	Memory at 4000000000 (64-bit, prefetchable) [size=256M]
	I/O ports at 3000 [size=64]
	Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
	Capabilities: [40] Vendor Specific Information: Len=0c <?>
	Capabilities: [70] Express Root Complex Integrated Endpoint, MSI 00
	Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable+ 64bit-
	Capabilities: [d0] Power Management version 2
	Capabilities: [100] Process Address Space ID (PASID)
	Capabilities: [200] Address Translation Service (ATS)
	Capabilities: [300] Page Request Interface (PRI)
[... cut ...]
My next step will be to to regenerate first if that works, then remove `i915` from the MODULES section and regenerate the initramfs, maybe it's loading the original i915 because I haven't regenerated initramfs since installing the dkms module.
Googling hasn't helped much either since I can hardly find anyone wanting to use a dkms version of module over an original one, and in i915's case, I'm fairly certain it's included with the Arch Linux `mesa` package which means I can't just uninstall `mesa` since so many other packages depend on it.

Alion548, 2022-10-07:
I add the following process:
1. kdms install
2. dracut -f /boot/initramfs-5.19.14-300.fc37.x86_64.img $(uname -r)
3. lsinitrd -k $(uname -r)
4. grub2-mkconfig
Where the second step img address is modified according to your actual one.
SR-IOV enabled successfully, but the VM iGPU reports an error (windows 10 21H2)