strongtz/i915-sriov-dkms #2: UHD 730 support

ra1nb0w, 2022-09-28:
Hi,
thank you very much for this dkms.
I am trying to use it with my UHD 730 but the `probe` fails.

```
[    0.050678] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-linux root=UUID=9be5a562-3abd-49f0-88fb-3669ae1346bb rw lsm=landlock,lockdown,yama,integrity,apparmor,bpf cryptdevice=UUID=749a2178-6bfc-457f-bfe5-2e43bb5f1100:cryptoroot:allow-discards,no-read-workqueue,no-write-workqueue cryptkey=rootfs:/root/.cryptoroot.keyfile ibt=off intel_iommu=on i915.enable_guc=7 loglevel=3 quiet
[    8.170255] i915 0000:00:02.0: i915_virtualization_probe: entry
[    8.170269] i915 0000:00:02.0: i915_sriov_probe: entry
[    8.170269] i915 0000:00:02.0: i915_sriov_probe: gen12_pci_capability_is_vf in
[    8.170272] i915 0000:00:02.0: i915_sriov_probe: dev_is_pf in
[    8.170272] i915 0000:00:02.0: pf_verify_readiness: entry
[    8.170273] i915 0000:00:02.0: VFs BAR not ready
[    8.170273] i915 0000:00:02.0: i915_sriov_probe: SR-IOV probe failed
[    8.170274] i915 0000:00:02.0: i915_sriov_probe returns: 1
[    8.170877] i915 0000:00:02.0: vgaarb: deactivate vga console
[    8.171487] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    8.173483] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
[    8.199557] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_70.1.1.bin version 70.1
[    8.199560] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc_7.9.3.bin version 7.9
[    8.202113] i915 0000:00:02.0: [drm] HuC authenticated
[    8.202425] i915 0000:00:02.0: [drm] GuC submission enabled
[    8.202425] i915 0000:00:02.0: [drm] GuC SLPC enabled
[    8.202692] i915 0000:00:02.0: [drm] GuC RC: enabled
[    8.203678] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    8.263937] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    8.352473] fbcon: i915drmfb (fb0) is primary device
[    8.471757] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
```

It should support SRIOV without any issue since it is the same as 770 except the frequency and the codec engine numbers. Maybe some checks?

thanks

strongtz, 2022-09-28:
Try enabling these in BIOS: - Resizable BAR - PCIe Native ASPM - SR-IOV - Above 4G Decoding

strongtz, 2022-09-29:
No, it can't directly output to a physical port. I'm not sure whether looking-glass will work, as it doesn't provide a framebuffer.

ra1nb0w, 2022-09-29:
ok. I tried adding the VF as vfio-pci but Windows fails to initialize the card (error 43) maybe for the framebuffer. How do you use the created VF? I would like to have a fully accelerated windows VM.

strongtz, 2022-09-29:
Honestly I have no idea either, and I only succeeded with a Linux VM. Someone on Reddit seems to have succeeded using Win10 though.
https://www.reddit.com/r/VFIO/comments/xoeika/sriov_on_intel_uhd_graphics_770_xe_12th_gen/

ra1nb0w, 2022-09-29:
after reading that xml and a few try&fail I succeed to enable correctly the VF graphic card and installing the Intel DCH driver. Unfortunately, [looking-glass](https://looking-glass.io/) seems to require an interface with a real output and not only rendering to start the host process.
Anyway, thanks for hit!

strongtz, 2022-09-29:
> after reading that xml and a few try&fail I succeed to enable correctly the VF graphic card and installing the Intel DCH driver. Unfortunately, [looking-glass](https://looking-glass.io/) seems to [...]

Wow, are you using W11 or W10? I myself still can't get it working xD

ra1nb0w, 2022-09-29:
W10 LTSC. You can find my image and xml at this link https://mega.nz/file/Je9hkbYQ#nAvhtvHsEIBS3TzavJYvq9hL2rkyfP4VJLoRWM4Byjc

strongtz, 2022-12-08:
@Priyantha Your iGPU is not supported. Only Xe architecture iGPU supports SR-IOV.