strongtz/i915-sriov-dkms #3: got error on boot

vvhh2002, 2022-09-29:
[    4.238574] ================================================================================
[    4.238579] UBSAN: array-index-out-of-bounds in /var/lib/dkms/i915-sriov-dkms/0.1/build/drivers/gpu/drm/i915/gt/intel_engine_user.c:61:43
[    4.238582] index 5 is out of range for type 'u8 [4]'
[    4.238584] CPU: 16 PID: 843 Comm: modprobe Tainted: P     U     O      5.19.7-1-pve #1
[    4.238586] Hardware name: Dell Inc. Precision 3460/08WXMX, BIOS 1.5.1 08/24/2022
[    4.238587] Call Trace:
[    4.238588]  <TASK>
[    4.238590]  dump_stack_lvl+0x49/0x63
[    4.238594]  dump_stack+0x10/0x16
[    4.238595]  ubsan_epilogue+0x9/0x3f
[    4.238596]  __ubsan_handle_out_of_bounds.cold+0x44/0x49
[    4.238598]  ? try_to_wake_up+0x22c/0x620
[    4.238601]  engine_cmp+0xb0/0x110 [i915]
[    4.238677]  ? intel_engine_reset_pinned_contexts+0x70/0x70 [i915]
[    4.238733]  list_sort+0x1bc/0x2a0
[    4.238736]  intel_engines_driver_register+0x7e/0x6a0 [i915]
[    4.238788]  ? intel_modeset_init+0x24f/0x2a0 [i915]
[    4.238868]  ? down_write+0x13/0x60
[    4.238870]  ? blocking_notifier_chain_register+0x40/0x70
[    4.238872]  i915_gem_driver_register+0x1b/0x30 [i915]
[    4.238945]  i915_driver_probe.cold+0x862/0xa8f [i915]
[    4.239028]  ? acpi_dev_found+0x66/0x80
[    4.239032]  i915_pci_probe+0xe2/0x1d0 [i915]
[    4.239090]  local_pci_probe+0x48/0x90
[    4.239093]  ? pci_match_device+0xe2/0x140
[    4.239095]  pci_device_probe+0xc8/0x270
[    4.239096]  really_probe+0x1d3/0x3b0
[    4.239098]  __driver_probe_device+0x119/0x190
[    4.239100]  driver_probe_device+0x23/0xc0
[    4.239101]  __driver_attach+0xbd/0x200
[    4.239102]  ? __device_attach_driver+0x120/0x120
[    4.239104]  bus_for_each_dev+0x7c/0xd0
[    4.239105]  driver_attach+0x1e/0x30
[    4.239106]  bus_add_driver+0x178/0x220
[    4.239107]  driver_register+0x91/0x100
[    4.239108]  __pci_register_driver+0x68/0x70
[    4.239110]  i915_pci_register_driver+0x23/0x30 [i915]
[    4.239164]  i915_init+0x3b/0xed [i915]
[    4.239221]  ? 0xffffffffc385c000
[    4.239223]  do_one_initcall+0x46/0x210
[    4.239225]  ? kmem_cache_alloc_trace+0x1a6/0x320
[    4.239227]  do_init_module+0x52/0x210
[    4.239229]  load_module+0x1d9c/0x21d0
[    4.239231]  __do_sys_finit_module+0xc8/0x140
[    4.239233]  __x64_sys_finit_module+0x1a/0x20
[    4.239234]  do_syscall_64+0x59/0x90
[    4.239236]  ? syscall_exit_to_user_mode+0x26/0x50
[    4.239237]  ? __x64_sys_mmap+0x33/0x50
[    4.239238]  ? do_syscall_64+0x69/0x90
[    4.239240]  ? syscall_exit_to_user_mode+0x26/0x50
[    4.239241]  ? __x64_sys_newstat+0x16/0x20
[    4.239243]  ? do_syscall_64+0x69/0x90
[    4.239245]  ? syscall_exit_to_user_mode+0x26/0x50
[    4.239246]  ? __x64_sys_read+0x1a/0x20
[    4.239247]  ? do_syscall_64+0x69/0x90
[    4.239248]  ? do_syscall_64+0x69/0x90
[    4.239250]  ? do_syscall_64+0x69/0x90
[    4.239251]  entry_SYSCALL_64_after_hwframe+0x63/0xcd
[    4.239253] RIP: 0033:0x7fd836f213a9
[    4.239255] Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d b7 8a 0d 00 f7 d8 64 89 01 48
[    4.239256] RSP: 002b:00007fff6021e768 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[    4.239258] RAX: ffffffffffffffda RBX: 0000557a51e6ab30 RCX: 00007fd836f213a9
[    4.239259] RDX: 0000000000000000 RSI: 0000557a51e648c0 RDI: 0000000000000000
[    4.239260] RBP: 0000000000040000 R08: 0000000000000000 R09: 0000557a51e6acb0
[    4.239261] R10: 0000000000000000 R11: 0000000000000246 R12: 0000557a51e648c0
[    4.239262] R13: 0000000000000000 R14: 0000557a51e6a8c0 R15: 0000557a51e6ab30
[    4.239263]  </TASK>
[    4.239264] ================================================================================
[    4.239265] ================================================================================
[    4.239267] UBSAN: array-index-out-of-bounds in /var/lib/dkms/i915-sriov-dkms/0.1/build/drivers/gpu/drm/i915/gt/intel_engine_user.c:63:43
[... 103 lines cut ...]
[    4.240679]  __do_sys_finit_module+0xc8/0x140
[    4.240681]  __x64_sys_finit_module+0x1a/0x20
[    4.240682]  do_syscall_64+0x59/0x90
[    4.240683]  ? syscall_exit_to_user_mode+0x26/0x50
[    4.240684]  ? __x64_sys_mmap+0x33/0x50
[    4.240685]  ? do_syscall_64+0x69/0x90
[    4.240687]  ? syscall_exit_to_user_mode+0x26/0x50
[    4.240688]  ? __x64_sys_newstat+0x16/0x20
[    4.240689]  ? do_syscall_64+0x69/0x90
[    4.240691]  ? syscall_exit_to_user_mode+0x26/0x50
[    4.240692]  ? __x64_sys_read+0x1a/0x20
[    4.240693]  ? do_syscall_64+0x69/0x90
[    4.240694]  ? do_syscall_64+0x69/0x90
[    4.240696]  ? do_syscall_64+0x69/0x90
[    4.240697]  entry_SYSCALL_64_after_hwframe+0x63/0xcd
[    4.240698] RIP: 0033:0x7fd836f213a9
[    4.240699] Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d b7 8a 0d 00 f7 d8 64 89 01 48
[    4.240701] RSP: 002b:00007fff6021e768 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[    4.240702] RAX: ffffffffffffffda RBX: 0000557a51e6ab30 RCX: 00007fd836f213a9
[    4.240703] RDX: 0000000000000000 RSI: 0000557a51e648c0 RDI: 0000000000000000
[    4.240704] RBP: 0000000000040000 R08: 0000000000000000 R09: 0000557a51e6acb0
[    4.240705] R10: 0000000000000000 R11: 0000000000000246 R12: 0000557a51e648c0
[    4.240705] R13: 0000000000000000 R14: 0000557a51e6a8c0 R15: 0000557a51e6ab30
[    4.240707]  </TASK>
[    4.240707] ================================================================================

strongtz, 2022-09-29:
I'm not really familiar with the i915 driver, and I only made a dkms package for it. It sounds like kernel header mismatch.

btw, is it a critically error? Seems the driver didn't really fail.

vvhh2002, 2022-09-29:
yes, it isn't a critically error. it seem that SRIOV works. I didn't test on kvm .
I fixed the problem with the code below. it use the code from branch `5.10/yocto`.

`// SPDX-License-Identifier: MIT
/*
 * Copyright © 2019 Intel Corporation
 */

#include <linux/list.h>
#include <linux/list_sort.h>
#include <linux/llist.h>

#include "i915_drv.h"
#include "intel_engine.h"
#include "intel_engine_user.h"
#include "intel_gt.h"
#include "uc/intel_guc_submission.h"

struct intel_engine_cs *
intel_engine_lookup_user(struct drm_i915_private *i915, u8 class, u8 instance)
{
	struct rb_node *p = i915->uabi_engines.rb_node;

	while (p) {
		struct intel_engine_cs *it =
			rb_entry(p, typeof(*it), uabi_node);

		if (class < it->uabi_class)
			p = p->rb_left;
		else if (class > it->uabi_class ||
			 instance > it->uabi_instance)
			p = p->rb_right;
		else if (instance < it->uabi_instance)
			p = p->rb_left;
		else
			return it;
	}

	return NULL;
}

void intel_engine_add_user(struct intel_engine_cs *engine)
{
	llist_add((struct llist_node *)&engine->uabi_node,
		  (struct llist_head *)&engine->i915->uabi_engines);
}

static const u8 uabi_classes[] = {
	[RENDER_CLASS] = I915_ENGINE_CLASS_RENDER,
	[COPY_ENGINE_CLASS] = I915_ENGINE_CLASS_COPY,
	[VIDEO_DECODE_CLASS] = I915_ENGINE_CLASS_VIDEO,
	[VIDEO_ENHANCEMENT_CLASS] = I915_ENGINE_CLASS_VIDEO_ENHANCE,
	[COMPUTE_CLASS] = I915_ENGINE_CLASS_COMPUTE,
};

static int engine_cmp(void *priv,const struct list_head *A,const struct list_head *B)
{
	const struct intel_engine_cs *a =
		container_of((struct rb_node *)A, typeof(*a), uabi_node);
	const struct intel_engine_cs *b =
		container_of((struct rb_node *)B, typeof(*b), uabi_node);

	if (uabi_classes[a->class] < uabi_classes[b->class])
		return -1;
	if (uabi_classes[a->class] > uabi_classes[b->class])
		return 1;

	if (a->instance < b->instance)
		return -1;
	if (a->instance > b->instance)
[... 216 lines cut ...]
				errors++;
			}
		}

		if (drm_WARN(&i915->drm, errors,
			     "Invalid UABI engine mapping found"))
			i915->uabi_engines = RB_ROOT;
	}

	set_scheduler_caps(i915);
}

unsigned int intel_engines_has_context_isolation(struct drm_i915_private *i915)
{
	struct intel_engine_cs *engine;
	unsigned int which;

	which = 0;
	for_each_uabi_engine(engine, i915)
		if (engine->default_state)
			which |= BIT(engine->uabi_class);

	return which;
}
`

closed 2022-09-29 by vvhh2002

skill7899, 2022-10-13:
> // SPDX-License-Identifier: MIT
> /*
>  * Copyright © 2019 Intel Corporation
>  */
>
> #include <linux/list.h>
> #include <linux/list_sort.h>
> #include <linux/llist.h>
>
> #include "i915_drv.h"
> #include [...]

@vvhh2002 how to fix it？