strongtz/i915-sriov-dkms #4: It looks like not support linux kernel 5.14.

IriKaQ, 2022-09-30:
Please fix the following errors if possible.

When I install it into almalinux 9.0, and use `dkms autoinstall` to compile it, and some error is report:

```
DKMS make.log for i915-sriov-dkms-5.15.49 for kernel 5.14.0-70.13.1.el9_0.x86_64 (x86_64)
Sat Oct  1 08:11:12 AM CST 2022
make: Entering directory '/usr/src/kernels/5.14.0-70.13.1.el9_0.x86_64'
  CC [M]  /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/i915_driver.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/i915_drm_client.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/i915_config.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/i915_getparam.o
In file included from /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/gt/uc/intel_uc.h:9,
                 from /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/gt/intel_gt_types.h:19,
                 from /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/gt/intel_engine.h:18,
                 from /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/i915_drv.h:58,
                 from /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/i915_config.c:6:
/var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/gt/uc/intel_guc.h:10:10: fatal error: linux/iosys-map.h: No such file or directory
   10 | #include <linux/iosys-map.h>
      |          ^~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/gt/uc/intel_uc.h:9,
                 from /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/gt/intel_gt_types.h:19,
                 from /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/gt/intel_engine.h:18,
                 from /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/i915_drv.h:58,
                 from /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/i915_getparam.c:9:
/var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/gt/uc/intel_guc.h:10:10: fatal error: linux/iosys-map.h: No such file or directory
   10 | #include <linux/iosys-map.h>
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
compilation terminated.
make[1]: *** [scripts/Makefile.build:271: /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/i915_config.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [scripts/Makefile.build:271: /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/i915_getparam.o] Error 1
In file included from /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/gt/uc/intel_uc.h:9,
                 from /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/gt/intel_gt_types.h:19,
                 from /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/gt/intel_engine.h:18,
                 from /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/i915_drv.h:58,
                 from /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/gt/intel_context.h:14,
                 from /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/gem/i915_gem_context.h:12,
                 from /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/i915_drm_client.c:14:
/var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/gt/uc/intel_guc.h:10:10: fatal error: linux/iosys-map.h: No such file or directory
   10 | #include <linux/iosys-map.h>
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:271: /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/i915_drm_client.o] Error 1
In file included from /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/i915_driver.c:52:
/var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/display/intel_display_types.h:34:10: fatal error: drm/display/drm_dp_dual_mode_helper.h: No such file or directory
   34 | #include <drm/display/drm_dp_dual_mode_helper.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:271: /var/lib/dkms/i915-sriov-dkms/5.15.49/build/drivers/gpu/drm/i915/i915_driver.o] Error 1
make: *** [Makefile:1862: /var/lib/dkms/i915-sriov-dkms/5.15.49/build] Error 2
make: Leaving directory '/usr/src/kernels/5.14.0-70.13.1.el9_0.x86_64'
```

strongtz, 2022-10-01:
You should update your kernel to 5.15 or later

IriKaQ, 2022-10-01:
I will try upgrade the kernel or install archlinux。
Thnaks

vvhh2002, 2022-10-01:
It requires Linux kernel with minimum version 5.19

IriKaQ, 2022-10-01:
I install the latest archlinux, and enable 7 VFs successfully。
I will test it that passthrough into a guest, such as windows 10, ubuntu and others。

IriKaQ, 2022-10-03:
I gave up.
Installing the kernel and drivers yourself requires a strong ability, and I don't have it yet.

Thanks

closed 2022-10-03 by IriKaQ