strongtz/i915-sriov-dkms #5: has anyone test passthrough uhd770 ogpu into a guest

vvhh2002, 2022-10-04:
I've tried multiple times and failed

skill7899, 2022-10-14:
[image]
i success

Alion548, 2022-10-14:
> 

Please help me to test if chrome gpu hardware acceleration can work properly, visit: chrome://gpu, thanks a lot!

Alion548, 2022-10-14:
@skill7899 

Please help me to test if chrome gpu hardware acceleration can work properly, visit: chrome://gpu, thanks a lot!

vvhh2002, 2022-10-15:
If the driver can correctly hardware acceleration. You should use rustDesk for testing but not RDP . The right vGPU should be able to virtualize a monitor, as GVT-D does.

skill7899, 2022-10-15:
> @skill7899
> 
> Please help me to test if chrome gpu hardware acceleration can work properly, visit: chrome://gpu, thanks a lot!

。。i find the driver response code 43 error

skill7899, 2022-11-09:
success https://github.com/strongtz/i915-sriov-dkms/issues/7

michael-pptf, 2022-11-11:
> @skill7899
> 
> Please help me to test if chrome gpu hardware acceleration can work properly, visit: chrome://gpu, thanks a lot!

Graphics Feature Status
Canvas: Hardware accelerated
Canvas out-of-process rasterization: Disabled
Direct Rendering Display Compositor: Disabled
Compositing: Hardware accelerated
Multiple Raster Threads: Enabled
OpenGL: Enabled
Rasterization: Hardware accelerated
Raw Draw: Disabled
Video Decode: Hardware accelerated
Video Encode: Hardware accelerated
Vulkan: Disabled
WebGL: Hardware accelerated
WebGL2: Hardware accelerated
WebGPU: Hardware accelerated

closed 2023-05-04 by strongtz