strongtz/i915-sriov-dkms #7: i'can install in pve

skill7899, 2022-10-13:
root@Wei:~# dkms install -m i915-sriov -v dkms

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...(bad exit status: 2)
make -j24 KERNELRELEASE=5.15.60-2-pve -j24 -C /lib/modules/5.15.60-2-pve/build M=/var/lib/dkms/@_PKGBASE@/@PKGVER@/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.15.60-2-pve (x86_64)
Consult /var/lib/dkms/i915-sriov/dkms/build/make.log for more information.
root@Wei:~# cat /var/lib/dkms/i915-sriov/dkms/build/make.log
DKMS make.log for i915-sriov-dkms for kernel 5.15.60-2-pve (x86_64)
Thu 13 Oct 2022 02:08:45 PM CST
make: Entering directory '/usr/src/linux-headers-5.15.60-2-pve'
scripts/Makefile.build:44: /var/lib/dkms/@_PKGBASE@/@PKGVER@/build/Makefile: No such file or directory
make[1]: *** No rule to make target '/var/lib/dkms/@_PKGBASE@/@PKGVER@/build/Makefile'.  Stop.
make: *** [Makefile:1900: /var/lib/dkms/@_PKGBASE@/@PKGVER@/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.15.60-2-pve'

IriKaQ, 2022-10-13:
最低支持的内核版本版本应该是 5.18。

skill7899, 2022-10-13:
> 最低支持的内核版本版本应该是 5.18。
好的，我更新到5.19，可以安装了
https://github.com/strongtz/i915-sriov-dkms/issues/4

skill7899, 2022-10-13:
@IriKaQ 安装成功，重启后开启不成功，已经
`
0.000000] Command line: initrd=\EFI\proxmox\5.19.7-2-pve\initrd.img-5.19.7-2-pve root=ZFS=rpool/ROOT/pve-1 boot=zfs intel_iommu=on i915.enable_guc=7
[    4.108753] Setting dangerous option enable_guc - tainting kernel
[    4.109161] i915 0000:00:02.0: i915_virtualization_probe: entry
[    4.109188] i915 0000:00:02.0: i915_sriov_probe: entry
[    4.109190] i915 0000:00:02.0: i915_sriov_probe: gen12_pci_capability_is_vf in
[    4.109195] i915 0000:00:02.0: i915_sriov_probe: dev_is_pf in
[    4.109197] i915 0000:00:02.0: pf_verify_readiness: entry
[    4.109198] i915 0000:00:02.0: pf_verify_readiness: pf_reduce_totalvfs
[    4.109200] i915 0000:00:02.0: i915_sriov_probe: I915_IOV_MODE_SRIOV_PF
[    4.109201] i915 0000:00:02.0: i915_sriov_probe returns: 3
[    4.109203] i915 0000:00:02.0: Running in SR-IOV PF mode
[    4.109744] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    4.109771] i915 0000:00:02.0: vgaarb: deactivate vga console
[    4.109809] i915 0000:00:02.0: [drm] Transparent Hugepage mode 'huge=within_size'
[    4.110446] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    4.111691] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
[    4.111980] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[    4.112232] i915 0000:00:02.0: Direct firmware load for i915/tgl_guc_70.1.1.bin failed with error -2
[    4.112236] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_70.1.1.bin: fetch failed with error -2
[    4.112238] i915 0000:00:02.0: [drm] GuC firmware(s) can be downloaded from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915
[    4.205475] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_70.1.1.bin version 0.0
[    4.205493] i915 0000:00:02.0: GuC initialization failed -2
[    4.205494] i915 0000:00:02.0: Enabling uc failed (-5)
[    4.205495] i915 0000:00:02.0: Failed to initialize GPU, declaring it wedged!

root@Wei:~# echo 2 > /sys/devices/pci0000\:00/0000\:00\:02.0/sriov_numvfs 
-bash: echo: write error: Numerical result out of range`

IriKaQ, 2022-10-13:
看起来是你的系统中不存在相应的 firmware，你到log中指定的地址下载对应 firmware (`i915/tgl_guc_70.1.1.bin`)，然后放入 `/lib/firmware/` 目录中。

refer: https://www.kernel.org/doc/html/latest/driver-api/firmware/fw_search_path.html

IriKaQ, 2022-10-13:
Download URL: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/i915/tgl_guc_70.1.1.bin

skill7899, 2022-10-13:
> Download URL: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/i915/tgl_guc_70.1.1.bin
> 
> ```
> mkdir -p /lib/firmware/i915
> wget -P /lib/firmware/i915 https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/i915/tgl_guc_70.1.1.bin
> ```

十分感谢成功开启

closed 2022-10-13 by skill7899

IriKaQ, 2022-10-13:
帮忙测试下能否直通 i915 vf 到 windows 虚拟机中正常驱动。

skill7899, 2022-10-13:
> 帮忙测试下能否直通 i915 vf 到 windows 虚拟机中正常驱动。

好呢，明天就干，机子也是别人的。在下载win11镜像。

skill7899, 2022-10-14:
> 帮忙测试下能否直通 i915 vf 到 windows 虚拟机中正常驱动。

[image]
没有问题

riverscn, 2022-10-14:
> > 帮忙测试下能否直通 i915 vf 到 windows 虚拟机中正常驱动。
> 
> [image] 没有问题

驱动是驱动了，你试试能否正常调用显卡功能呢？

skill7899, 2022-10-14:
> 

code 43，没有再拆腾了。linux我看有人已经成功驱动，win的43，老问题

riverscn, 2022-10-14:
> @riverscn
> code 43，没有再拆腾了。linux我看有人已经成功驱动，win的43，老问题

Linux我也没成功，驱动载入失败。
mainline或这个包都没有成功……
等一个成功经验……

skill7899, 2022-10-26:
> mainline
mainline是那个分支。。我没有找到

riverscn, 2022-10-26:
> @riverscn mainline是那个分支。。我没有找到

我成功了。用Windows Update自带驱动。

[image]

skill7899, 2022-10-27:
> > @riverscn 那个分支的sriov？pve的直通配置是是啥？那个windows？11还是10？
> 
> 这个repo，自己通过dkms安装，支持5.19和以后的内核。 PVE的虚拟机用OVMF固件，硬件里添加一个显卡的PCIe VF地址就行了，勾选“primary gpu"。 Windows 10或者11应该都没问题。

你试一下，复制一份，修改了一下显卡为第二个sriov看能不能启

riverscn, 2022-10-27:
> > > @riverscn 那个分支的sriov？pve的直通配置是是啥？那个windows？11还是10？
> > 
> > 
> > 这个repo，自己通过dkms安装，支持5.19和以后的内核。 PVE的虚拟机用OVMF固件，硬件里添加一个显卡的PCIe VF地址就行了，勾选“primary gpu"。 Windows 10或者11应该都没问题。
> 
> 你试一下，复制一份，修改了一下显卡为第二个sriov看能不能启

无论哪个都行，只要你不用PF的地址。

skill7899, 2022-10-27:
> > > > @riverscn 那个分支的sriov？pve的直通配置是是啥？那个windows？11还是10？
> > > 
> > > 
> > > 这个repo，自己通过dkms安装，支持5.19和以后的内核。 PVE的虚拟机用OVMF固件，硬件里添加一个显卡的PCIe VF地址就行了，勾选“primary gpu"。 Windows 10或者11应该都没问题。
> > 
> > 
> > 你试一下，复制一份，修改了一下显卡为第二个sriov看能不能启
> 
> 无论哪个都行，只要你不用PF的地址。

就是同时启动两个windows

skill7899, 2022-10-27:
@riverscn 
我试了，沟上“primary gpu"后，3389远程过去，直接黑屏。啥都没有。我觉得应该跟这个没有什么关系，你不沟上，应该也是驱动正常的。

riverscn, 2022-10-27:
> @riverscn
> 我试了，沟上“primary gpu"后，3389远程过去，直接黑屏。啥都没有。我觉得应该跟这个没有什么关系，你不沟上，应该也是驱动正常的。确认一下，你是不是直通了PF。
> [image]

1. 有关系。没关系我干嘛告诉你要勾上？
2. 黑屏的情况我一开始也遇到过。安装了合适的驱动以后就没问题了。windows update自己就能判断安装合适的驱动，也可以手动指定和我一样的驱动版本。

skill7899, 2022-10-27:
> > @riverscn
> > 我试了，沟上“primary gpu"后，3389远程过去，直接黑屏。啥都没有。我觉得应该跟这个没有什么关系，你不沟上，应该也是驱动正常的。确认一下，你是不是直通了PF。
> > [image]
> 
> 1. 有关系。没关系我干嘛告诉你要勾上？
> 2. 黑屏的情况我一开始也遇到过。安装了合适的驱动以后就没问题了。windows update自己就能判断安装合适的驱动，也可以手动指定和我一样的驱动版本。

我已经指定了一样的驱动版本，所以就想问一下，是不是windows版本问题

riverscn, 2022-10-27:
> > > @riverscn
> > > 我试了，沟上“primary gpu"后，3389远程过去，直接黑屏。啥都没有。我觉得应该跟这个没有什么关系，你不沟上，应该也是驱动正常的。确认一下，你是不是直通了PF。
> > > [image]
> > 
> > 
> > 
> > 1. 有关系。没关系我干嘛告诉你要勾上？
> > 2. 黑屏的情况我一开始也遇到过。安装了合适的驱动以后就没问题了。windows update自己就能判断安装合适的驱动，也可以手动指定和我一样的驱动版本。
> 
> 我已经指定了一样的驱动版本，所以就想问一下，是不是windows版本问题

试试勾上PCIe。然后就是你是用UEFI引导的吗？

skill7899, 2022-10-27:
> > > > @riverscn
> > > > 我试了，沟上“primary gpu"后，3389远程过去，直接黑屏。啥都没有。我觉得应该跟这个没有什么关系，你不沟上，应该也是驱动正常的。确认一下，你是不是直通了PF。
> > > > [image]
> > > 
> > > 
> > > 
> > > 1. 有关系。没关系我干嘛告诉你要勾上？
> > > 2. 黑屏的情况我一开始也遇到过。安装了合适的驱动以后就没问题了。windows update自己就能判断安装合适的驱动，也可以手动指定和我一样的驱动版本。
> > 
> > 
> > 我已经指定了一样的驱动版本，所以就想问一下，是不是windows版本问题
> 
> 试试勾上PCIe。然后就是你是用UEFI引导的吗？

可以了，谢谢，成功驱动。。不容易。应该是我没有删除干净驱动的版本。

Alion548, 2022-11-09:
> > @riverscn 那个分支的sriov？pve的直通配置是是啥？那个windows？11还是10？
> 
> 这个repo，自己通过dkms安装，支持5.19和以后的内核。 PVE的虚拟机用OVMF固件，硬件里添加一个显卡的PCIe VF地址就行了，勾选“primary gpu"。 Windows 10或者11应该都没问题。

麻烦可以看下虚拟机配置xml文件吗？

skill7899, 2022-11-09:
> > > @riverscn 那个分支的sriov？pve的直通配置是是啥？那个windows？11还是10？
> > 
> > 
> > 这个repo，自己通过dkms安装，支持5.19和以后的内核。 PVE的虚拟机用OVMF固件，硬件里添加一个显卡的PCIe VF地址就行了，勾选“primary gpu"。 Windows 10或者11应该都没问题。
> 
> 麻烦可以看下虚拟机配置xml文件吗？
我只有图片了，机器还给别人了。
[image]

Alion548, 2022-12-02:
> > > @riverscn那个分支的sriov？pve的直通配置是啥？那个windows？11还是10？
> > 
> > 
> > 这个rec​​oto，自己dkms安装，5.19和和内核。。。。的的的ovmf固件。
> 
> 麻烦可以看下虚拟机配置xml文件吗？

``` bash
cat ./GPU-Assign.xml

<hostdev mode='subsystem' type='pci' managed='yes'>
 <driver name='vfio'/>
 <source>
  <address domain='0x0000' bus='0x00' slot='0x02' function='0x1'/>
 </source>
</hostdev>
```