strongtz/i915-sriov-dkms #8: Which Windows guest driver to use

wnark, 2022-10-14:
Do you directly use the Windows driver from Intel's official website? Which driver are you using?

ammgws, 2022-10-21:
@wnark Did you figure out what to do? I'm assuming you're encountering the Code 43 issue inside the Windows VM?

michael-pptf, 2022-11-11:
i was able to use Intel official gfx driver, gfx_win_101.3790_101.2114

Yoskaldyr, 2022-11-29:
> i was able to use Intel official gfx driver, gfx_win_101.3790_101.2114

@michael-pptf What gen 11 or 12, do you have?

Because I saw a lot of success stories for 12 gen and no one for 11 gen :(

Intel has strange version numbering for own DCH Drivers. One package has 2 version of drivers - one for legacy iGPUs and one for the new one (11-13 gen). So gfx_win_101.3790_101.2114.exe driver has 101.3790 version for the new GPUS and 101.2114 for the legacy ones.

michael-pptf, 2022-11-29:
@Yoskaldyr Yea, mine is 12th gen. So yours is 11th gen and it doesn't work under Windows by getting code 43?

zx900930, 2022-12-09:
Can confirm gfx_win_101.3790_101.2114 is working for win10 22h2 intel 12gen iGPU sriov.
If you still get code 43 after fresh install the driver. 
try: 
1. Set the cpu type to `host`.
2. Use another ms original windows img

strongtz, 2022-12-10:
Did anyone manage to get it working in Windows 11? :/

Yoskaldyr, 2022-12-11:
> So yours is 11th gen and it doesn't work under Windows by getting code 43?

yeah. on 11gen I've got 43 error code and gnome shell restarts after virtual machine shutdown

Yoskaldyr, 2022-12-11:
> @Yoskaldyr Yea, mine is 12th gen. So yours is 11th gen and it doesn't work under Windows by getting code 43?

@michael-pptf Did you use Looking glass? Is every hardware decoding working inside virtual machine?

tristan-k, 2023-01-23:
> 1. Set the cpu type to `host`.

Thanks. This was the missing piece on a Intel NUC 12 (i5-1250P) to get the iGPU running inside a Windows10 VM.

pioalpha, 2023-04-13:
I updated the Ubuntu 22.04 kernel to version 6.1.23, installed strongtz's dkms, followed the tutorial https://github.com/intel/kubevirt-gfx-sriov and the SR-IOV of my i5-12400F (UHD 730) worked by installing windows 10 and the intel driver 30.101.3111 indicated in the images. I couldn't get it to work with the latest intel driver or windows 11.

michael-pptf, 2023-04-13:
> I updated the Ubuntu 22.04 kernel to version 6.1.23, installed strongtz's dkms, followed the tutorial https://github.com/intel/kubevirt-gfx-sriov and the SR-IOV of my i5-12400F (UHD 730) worked by [...]

12400F doesn't have an iGPU though. Is it 12400?

pioalpha, 2023-04-13:
> 12400F doesn't have an iGPU though. Is it 12400?

Sorry, corrected to 12400.

pg2web, 2023-05-29:
Some issues resolved, including Code 43:

CPU: 14-core (6-mt/8-st) 13th Gen Intel Core i7-13700H (-MST AMCP-), **GPU Iris XE**

Except issues with host suspend (laptop doesn't come back from sleep and needs poweroff) everything works great on Fedora 38 with some tricks:

1. To get rid of Code 43 error in Windows gues you MUST set vendor_id for your VM. Any id worked for me. for example (XML edit mode in virt-manager):

    <apic/>
    <hyperv mode="custom">
      <relaxed state="on"/>
      <vapic state="on"/>
      <spinlocks state="on" retries="8191"/>
      **<vendor_id state="on" value="IrisXE"/>**

2. Wayland gnome session crashes when you power off VM with passthrough vGPU, but no issues with Gnome on Xorg. The root cause - mutter doesn't handle vGPU hotadd monitoring /dev/dri/ well so when virt-manager detach vGPU and reattach it back to host it triggers gnome session crash. The workaround is simple - just change <hostdev> parameter of assigned vGPU device from managed="yes" to "no". Example for attached 0000:00:02.7:

<hostdev mode="subsystem" type="pci" **managed="no"**>
  <source>
    <address domain="0x0000" bus="0x00" slot="0x02" function="0x7"/>
  </source>
  <address type="pci" domain="0x0000" bus="0x04" slot="0x00" function="0x0"/>
</hostdev>

 Also you need manually (or via qemu pre-boot hook) detach vGPU from host, just once:
virsh nodedev-detach pci_0000_00_02_7

This way it will disappear from /dev/dri and virt-manager will not do nodedev-reattach, confusing Wayland.

Hope it will save you some hairs )

cyril-cheng, 2023-07-06:
emmmm，how to set it in pve

sniirful, 2023-07-22:
I'm on Fedora 38 with an i5-1135G7, i915 driver installed successfully with DKMS (had to disable secure boot or it wouldn't work), VF passed to the VM, which recognizes the Intel Iris XE Graphics, but all kinds of drivers that I try give error 43, both WHQL and BETA. I've tried both Windows 11 and Windows 10 22H2. Also tried the vendor_id as IrisXE and GenuineIntel, on all the configurations.

The only useful info I can gather is that in the Events, it says it loads everything, also igfxn, but BasicDisplay. "Device not started (BasicDisplay)". The driver name seems to be "display.inf", with codes 0x1F and 0xC01E0438

pagkly, 2023-08-17:
Intel i5 1135G7 HP Probook 430 G8 Arch Linux with Windows 10 Guest having error code 43 as well. Setting vendor id did not resolve the issue.

masiv1001, 2023-09-25:
> Intel i5 1135G7 Iris Xe on HP Probook 430 G8 with Arch Linux with Windows 10 Guest having error code 43 as well. Setting vendor id did not resolve the issue.

Try other driver versions, I tried with 3222 and worked (https://downloadmirror.intel.com/738230/gfx_win_101.3222_101.2111.exe), I couldn't get it working past that point though (have the same processor, Dell Latitude 5420)
[image]
[image]

sniirful, 2023-09-27:
Instead of using dkms I have just tried the [Intel mainline tracking kernel v6.4](https://github.com/intel/mainline-tracking/releases/tag/mainline-tracking-v6.4-rt6-preempt-rt-230921T045946Z) and I've got weird results. My specs:
- Intel Core i5-1135G7
- vendor_id set as “GenuineIntel”
- Windows 10 guest

With the [latest](https://downloadmirror.intel.com/788789/gfx_win_101.4826.exe) drivers I couldn't get it work at all. Always error 43, no matter what I did.

With the [3222 driver](https://downloadmirror.intel.com/738230/gfx_win_101.3222_101.2111.exe) linked by @masiv1001 apparently it worked with no more error 43. It did however make my entire computer lag so much that it was barely usable. Also, I had no way of making it work for graphics acceleration. Tried both Looking Glass and RDP but it looked as if the GPU/driver stopped working as soon as I booted the VM without the QXL display.

resiliencer, 2023-10-30:
Guys, as far as I have understood over the past couple of months, first fighting with iGPU full passthrough, then trying to overcome iGPU SR-IOV, one of the most common problems is simply incorrect initialization of the iGPU, which in the case of SR-IOV can be cured seemingly quite easily (in difference from full passthrough). To do this, you need to find the BIOS firmware of your motherboard and pull out IntelGopDriver.efi from there and feed it as a romfile in the virtual machine configuration. At the moment, I can launch virtual machines with Win10 or Win11 from two different Linux hosts without any problems, no matter what driver versions (which were available at that time). I tried to present the main idea here: https://github.com/resiliencer/i915-sriov-dkms

Coder84619, 2023-11-26:
@resiliencer Well with Proxmox 8.1, kernel 6.5, and the strongtz DKMS, all is working perfectly fine without any tweaks using an i5-1240P and i5-N305.

okhsunrog, 2023-11-30:
> Guys, as far as I have understood over the past couple of months, first fighting with iGPU full passthrough, then trying to overcome iGPU SR-IOV, one of the most common problems is simply incorrect [...]

Can you plese explain how do I add my IntelGopDriver.efi to libvird / VirtManager? I can't find any romfile option. i5-1135G7 here, windows 10 guest, archlinux host, 6.1.63-1-lts kernel. error 43 with almost all drivers. no error 43 with 3222 driver, but the system gets reaally slow, even host. vendor_id is set. the only thing I think I'm missing in the romfile, I finally figured out how to extract it from biosUpdate.exe but can't find a way to add it to my configuration in Virt Manager. Could you provide a guide, please?

okhsunrog, 2023-11-30:
Did some more digging and now the main question I have is: did any of you guys managed to get windows guests with 11th gen CPUs? As far as I see the only happy stories here are from 12th&13th gen owners

baishu12, 2024-07-31:
> Did some more digging and now the main question I have is: did any of you guys managed to get windows guests with 11th gen CPUs? As far as I see the only happy stories here are from 12th&13th gen owners

I am using i5-1155G7 With the [3222 driver](https://downloadmirror.intel.com/738230/gfx_win_101.3222_101.2111.exe) linked by @masiv1001 on pve.lt works fine.I'm using vbios.And set guest type to linux in pve.

xros, 2025-02-09:
The Windows 11 VM had problems with Intel graphics driver version 32 on N100. After I restored it to version 31, it worked again.

PVE 8.3.3, kernel version 6.8.12-8-pve