strongtz/i915-sriov-dkms #10: support kernel 6.0?

riverscn, 2022-10-17:
is kernel 6.0 supported?

x0wllaar, 2022-10-17:
I have just sent a PR #11 with a patch that allows the driver to compile and load (at least in a VM as a smoke test). I do not have an intel GPU right now (probably will arrive by the end of this week), but if you want, you can test the patch today. My fork, https://github.com/x0wllaar/i915-sriov-dkms has the patch and the PKGBUILD to build it to test, you can do:

```
mkdir i915-sriov-k6
cd i915-sriov-k6
git clone https://github.com/x0wllaar/i915-sriov-dkms
cp ./i915-sriov-dkms/PKGBUILD ./i915-sriov-dkms/i915-sriov-dkms-git.install ./build/
cd build/
makepkg
sudo pacman -U i915-sriov-dkms-git-5.15.49.kernel6-1-x86_64.pkg.tar.zst 
```

Again, it seems to install and load on kernel 6.0.2-arch1-1, but I didn't test it beyond that, so I 'm not responsible if your computer blows up on insmod

riverscn, 2022-10-18:
No, I tested.
It doesn't work.

riverscn, 2022-10-18:
Merging SR-IOV related code to 6.0's drm/i915 driver would be very cool.

x0wllaar, 2022-10-18:
Tbh I plan on at least looking into that, but for 6.1 because it will be LTS and because I will have more free time during the holidays/break around Christmas

closed 2022-10-21 by strongtz