strongtz/i915-sriov-dkms #12: Can't build on debian (user error likely)

skug67, 2022-10-21:
I'm trying to build this on a Debian system in advance of receiving my new machine w/an Alder Lake processor.  I've got kernel version 5.19.0-2 installed, saved the source to /usr/src/i915-sriov-5.15.49 edited dkms.conf to use package name of "i915-sriov" and package version of "5.15.49" and am running "dkms build i915-sriov/5.15.49"

The build errors with the attached log.  Any advice/suggestions would be very welcome.

[i915-sriov.log](https://github.com/strongtz/i915-sriov-dkms/files/9838290/i915-sriov.log)

skug67, 2022-10-21:
Just for kicks tried building against a debian 6.0 kernel and got a very different set of errors....
[i915-sriov-kernel6.log](https://github.com/strongtz/i915-sriov-dkms/files/9838453/i915-sriov-kernel6.log)

x0wllaar, 2022-10-21:
That's strange. Can you try building on 6.0.2?

skug67, 2022-10-21:
Hmmm..... I think the debian kernel package 6.0.0-1 that I was using actual tracks to the upstream 6.0.2 version.  I can try to build a kernel 6.0.2 kernel from scratch and then use this on top of that, but that would definitely be a longer-term process.

x0wllaar, 2022-10-22:
I just built it using the 6.0.3 kernel from the mainline ppa:

```
$uname -a
Linux gregory-laptop1 6.0.3-060003-generic #202210211149 SMP PREEMPT_DYNAMIC Fri Oct 21 11:53:42 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

Can you try installing this kernel? The debs are available at https://kernel.ubuntu.com/~kernel-ppa/mainline/v6.0.3/

skug67, 2022-10-22:
Unfortunately, Debian and Ubuntu apparently have a difference of opinion about the use of zstd in debs so installing those packages isnt' straightforward.  Since I'm doing this in a dedicated virtual machine, I'm pretty sure I can finagle it but it might take a little work....

skug67, 2022-10-22:
Success.  Able to install the Ubuntu kernel debs by just subbing in the Ubuntu version of dpkg.  Once I did that the DKMS modules built fine against that kernel and I was able to boot into the new kernel as well.  How it actually interacts with the hardware will have to wait until the system is delivered.

Thanks a million!!!!

(Any intel as to why it won't build against the Debian-packaged kernel would be appreciated -- to the extent possible, I'd prefer to avoid running a mixed-distro setup.  But totally understand if that's not a priority for this project.)

skug67, 2022-10-29:
New machine arrived.  Unfortunately, trying to boot with the DKMS module enabled produces the following console messages ending with a hung boot:

```
i915 0000:00:02.0: i915_virtualization_probe: entry
i915 0000:00:02.0: i915_sriov_probe: entry
i915 0000:00:02.0: i915_sriov_probe: gen12_pci_capability_is_vf in
i915 0000:00:02.0: i915_sriov_probe: dev_is_pf in
i915 0000:00:02.0: pf_verify_readiness: entry
i915 0000:00:02.0: GuC virtualization disabled
i915 0000:00:02.0: i915_sriov_probe: SR-IOV probe failed
i915 0000:00:02.0: i915_sriov_probe returns: 1
fb0: switching to i915 from EFI VGA
```

Zylanx, 2022-11-13:
Also fails with NixOS's kernel packages.

Not sure how many patches are applied, but I can't imagine it is too many? Most of the patching happens to the built derivatives I think?

Zylanx, 2022-11-14:
Worked it out.

It's because NixOS, and Debian kernels both do not have the CONFIG_DRM_I915_PXP option set. While the custom Makefile ignores those settings and blindly sets the extra cflags to include that.

NixOS also doesn't have the Intel PMIC config option set which also causes another missing symbol.

closed 2023-05-04 by strongtz