strongtz/i915-sriov-dkms #13: "device initialization failed" when booting a debian vm

ltsdw, 2022-11-03:
I don't know much about virtualization, even less about sr-iov, since intel 11th gen processors doesn't have support to gvt-g I looked up for an alternative and decided to try this sr-iov thing.

Probably this isn't the place to report this but maybe someone knows something.

[image]
[image]

[debian11.xml.txt](https://github.com/strongtz/i915-sriov-dkms/files/9933410/debian11.xml.txt)

strongtz, 2022-11-04:
Does the VM have dkms module installed?

ltsdw, 2022-11-04:
> Does the VM have dkms module installed?

not that I'm aware of, only dkms module that I have is the i915-sriov.

all I did was follow the basic qemu vertualization wiki page from arch [QEMU](https://wiki.archlinux.org/title/QEMU), installed the qemu-ui-egl-headless and adapted a xml that I  found on reddit.

didn't install anything on the guest.

strongtz, 2022-11-04:
you need to install this module in VM as well

ltsdw, 2022-11-04:
Thank you, I tried compiling the module using dkms on debian, but the building is failing because of some functions not implemented in the kernel headers.

I'll close this issue as I'm not familiar with virtualization, dkms and the problem apparently isn't related to this project.

closed 2022-11-04 by ltsdw