strongtz/i915-sriov-dkms #14: reboot persistence & set video output to a VF?

truvatech, 2022-11-28:
Hello, got this working on a new mobo 13th gen i9 after a BIOS firmware update and using this command rather than echo suggested in the readme:
`echo 3 | sudo tee /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs`

My only question, is how can I make this persist after a reboot? I have tried with 
`echo "devices/pci0000:00/0000:00:02.0/sriov_numvfs = 3" > /etc/sysfs.conf`
and making sure sysfsutils is installed, as found on [a chinese forum about someone who managed to get this working](https://www-right-com-cn.translate.goog/forum/thread-8258222-1-1.html?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=en&_x_tr_pto=wapp), which lead me here.

But that does not seem to work for me.

The other question I had, is how can we manage access to the video output from virtual functions? Is that even possible?

I would ideally like to split this into virtual functions and have the video output directly from a VM set with one of these virtual functions, but on boot.

Any idea?

Thank you.

I am using Fedora Server (SELinux enabled)

IriKaQ, 2022-11-28:
Try udev rules.

truvatech, 2022-12-01:
Thank you for the idea. I am not familiar with udev rules, so I will have to understand what those are and how they work first before trying that out.

For now I am not virtualizing it, and just passing it straight through, as I could not figure out how to output to my monitor from a virtual function, but one thing I wanted to add to my issue in case this helps anyone is the fact that I had to edit dkms.conf in order for this to compile
replace
`PACKAGE_NAME="@_PKGBASE@"
PACKAGE_VERSION="@PKGVER@"`
with
`PACKAGE_NAME="i915-sriov"
PACKAGE_VERSION="dkms"`

Move the contents of this repo to /usr/src
and that should allow you to build with dkms install -m i915-sriov -v dkms

Alion548, 2022-12-02:
I use systemd to implement.

``` text
[Unit]
Description=Intel GPU SR-IOV
After=syslog.target systemd-modules-load.service

[Service]
Type=oneshot
ExecStart=/usr/bin/bash -c '/usr/bin/echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs'

[Install]
WantedBy=multi-user.target
```

strongtz, 2022-12-08:
I have no idea if the VFs even support display framebuffer.

truvatech, 2023-04-25:
Thank you, I did not think to even consider whether or not VFs would have access to the physical outputs