strongtz/i915-sriov-dkms #16: Array out of bound error on boot

gehring, 2022-12-07:
I was able to build and install on pop os, with kernel 6.0.6-76060006-generic, but although everything seems to boot fine and the i915 driver seems to report being able to create VFs, I get a permission denied error when trying to create some using:

`&>  sudo echo "devices/pci0000:00/0000:00:02.0/sriov_numvfs = 2" > /etc/sysfs.conf`

That might be unrelated to the issue I'm reporting, but thought I'd mention it since it's part of the context. Otherwise, I see the following 3 out of bound error when starting. It doesn't seem to prevent the driver from running as far as I can tell. Any help would be much appreciated!

`&> dmesg | grep i915`
<details>

```
[   19.251638] i915 0000:00:02.0: i915_virtualization_probe: entry
[   19.251661] i915 0000:00:02.0: i915_sriov_probe: entry
[   19.251663] i915 0000:00:02.0: i915_sriov_probe: gen12_pci_capability_is_vf in
[   19.251668] i915 0000:00:02.0: i915_sriov_probe: dev_is_pf in
[   19.251669] i915 0000:00:02.0: pf_verify_readiness: entry
[   19.251670] i915 0000:00:02.0: pf_verify_readiness: pf_reduce_totalvfs
[   19.251671] i915 0000:00:02.0: i915_sriov_probe: I915_IOV_MODE_SRIOV_PF
[   19.251672] i915 0000:00:02.0: i915_sriov_probe returns: 3
[   19.251673] i915 0000:00:02.0: Running in SR-IOV PF mode
[   19.252232] i915 0000:00:02.0: [drm] VT-d active for gfx access
[   19.252240] fb0: switching to i915 from EFI VGA
[   19.252457] i915 0000:00:02.0: vgaarb: deactivate vga console
[   19.252533] i915 0000:00:02.0: [drm] Transparent Hugepage mode 'huge=within_size'
[   19.253141] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[   19.254822] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[   19.256146] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc_ver2_16.bin (v2.16)
[   19.390285] i915 0000:00:02.0: [drm] GuC firmware i915/adlp_guc_70.1.1.bin version 70.1
[   19.390292] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc_7.9.3.bin version 7.9
[   19.410479] i915 0000:00:02.0: [drm] HuC authenticated
[   19.411126] i915 0000:00:02.0: [drm] GuC submission enabled
[   19.411130] i915 0000:00:02.0: [drm] GuC SLPC enabled
[   19.411645] i915 0000:00:02.0: [drm] GuC RC: enabled
[   19.416707] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[   19.416856] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[   20.546165] UBSAN: array-index-out-of-bounds in /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gt/intel_engine_user.c:61:43
[   20.546190]  engine_cmp+0xc0/0x120 [i915]
[   20.546270]  ? intel_engine_reset_pinned_contexts+0x70/0x70 [i915]
[   20.546313]  intel_engines_driver_register+0x7e/0x6a0 [i915]
[   20.546356]  i915_gem_driver_register+0x1b/0x30 [i915]
[   20.546417]  i915_driver_probe.cold+0x7e5/0xa87 [i915]
[   20.546493]  i915_pci_probe+0xe2/0x1e0 [i915]
[   20.546564]  i915_pci_register_driver+0x23/0x30 [i915]
[   20.546606]  i915_init+0x3b/0xf2 [i915]
[   20.546694] UBSAN: array-index-out-of-bounds in /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gt/intel_engine_user.c:63:43
[   20.546707]  engine_cmp+0xf8/0x120 [i915]
[   20.546758]  ? intel_engine_reset_pinned_contexts+0x70/0x70 [i915]
[   20.546801]  intel_engines_driver_register+0x7e/0x6a0 [i915]
[   20.546852]  i915_gem_driver_register+0x1b/0x30 [i915]
[   20.546917]  i915_driver_probe.cold+0x7e5/0xa87 [i915]
[   20.546997]  i915_pci_probe+0xe2/0x1e0 [i915]
[   20.547070]  i915_pci_register_driver+0x23/0x30 [i915]
[   20.547114]  i915_init+0x3b/0xf2 [i915]
[   20.547195] UBSAN: array-index-out-of-bounds in /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gt/intel_engine_user.c:214:36
[   20.547206]  intel_engines_driver_register+0x5b9/0x6a0 [i915]
[   20.547262]  i915_gem_driver_register+0x1b/0x30 [i915]
[   20.547332]  i915_driver_probe.cold+0x7e5/0xa87 [i915]
[   20.547404]  i915_pci_probe+0xe2/0x1e0 [i915]
[   20.547474]  i915_pci_register_driver+0x23/0x30 [i915]
[   20.547518]  i915_init+0x3b/0xf2 [i915]
[   20.553334] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[   20.580999] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   20.581114] i915 0000:00:02.0: 7 VFs could be associated with this PF
[   20.584546] fbcon: i915drmfb (fb0) is primary device
[   20.584609] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
```
</details>

holkono, 2022-12-10:
I'm also on pop os with same kernel. I believe I somewhat got it working.

```
&>sudo kernelstub -p
&>Kernel Boot Options:.intel_iommu=on i915.enable_psr=0 i915.enable_fbc=0 i915.enable_guc=7
```
Don't mind the i915 psr and fbc command, it was to combat a horrible frame-buffer issue I experienced on my new HP Elitebook 840 G9 14" notebook.

Systemd service to echo a number somewhat early;
```
$ cat /etc/systemd/system/i915-sriov.service 
[Unit]
Description=Intel GPU SR-IOV
After=syslog.target systemd-modules-load.service

[Service]
Type=oneshot
ExecStart=/usr/bin/bash -c '/usr/bin/echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs'

[Install]
WantedBy=multi-user.target

```

Results;
```
$ sudo lshw -c display
  *-display:0               
       description: VGA compatible controller
       product: Intel Corporation
       vendor: Intel Corporation
       physical id: 2
       bus info: pci@0000:00:02.0
       logical name: /dev/fb0
       version: 0c
       width: 64 bits
       clock: 33MHz
       capabilities: pciexpress msi pm vga_controller bus_master cap_list rom fb
       configuration: depth=32 driver=i915 latency=0 mode=1920x1200 resolution=1920,1200 visual=truecolor xres=1920 yres=1200
       resources: iomemory:600-5ff iomemory:400-3ff irq:203 memory:603c000000-603cffffff memory:4000000000-400fffffff ioport:3000(size=64) memory:c0000-dffff memory:4010000000-4016ffffff memory:4020000000-40ffffffff
  *-display:1
       description: VGA compatible controller
       product: Intel Corporation
       vendor: Intel Corporation
       physical id: 2.1
       bus info: pci@0000:00:02.1
       version: 0c
       width: 64 bits
       clock: 33MHz
       capabilities: pciexpress msi vga_controller bus_master cap_list
       configuration: driver=i915 latency=0
       resources: irq:205 memory:4010000000-4010ffffff memory:4020000000-403fffffff
  *-display:2
       description: VGA compatible controller
       product: Intel Corporation
       vendor: Intel Corporation
       physical id: 2.2
       bus info: pci@0000:00:02.2
       version: 0c
       width: 64 bits
       clock: 33MHz
       capabilities: pciexpress msi vga_controller cap_list
       configuration: driver=vfio-pci latency=0
       resources: irq:0 memory:4011000000-4011ffffff memory:4040000000-405fffffff
```

Hope this helps.

gehring, 2022-12-12:
@holkono Thank you! This definitely helped! Not sure if it was the systemd service that I needed or if it was moving `i915.enable_guc=7` to the kernel options (I had it set in `modeprobe.d`) but it now works. I still get those out-of-bound errors but they don't seem to hurt anything.

closed 2023-05-04 by strongtz