strongtz/i915-sriov-dkms #18: Kernel 6.1 compilation failed

Alion548, 2022-12-13:
``` bash
/var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gem/i915_gem_ttm.c: In function '__i915_gem_ttm_object_init ':
/var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gem/i915_gem_ttm.c: 1233:36: Error: 'ttm_bo_init_reserved' 4th argument type is incompatible
 1233 | bo_type, & i915_sys_placement,
      | ^~~~~~~
      | |
      | enum ttm_bo_type
In file included from./include/drm/ttm/ttm_bo_driver: 42,
                 From /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gem/i915_gem_ttm.c: 8:
./include/drm/ttm/ttm_bo_api: 321:71: Note: Type'struct ttm_placement * 'is required, but the argument type is'enum ttm_bo_type'
  321 | enum ttm_bo_type type, struct ttm_placement * placement,
      | ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
/var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gem/i915_gem_ttm.c: 1233:45: Warning: Assign pointer to integer when passing 5th parameter of 'ttm_bo_init_reserved ', no type conversion [-Wint-conversion]
 1233 | bo_type, & i915_sys_placement,
      | ^~~~~~~~~~~~~~~~~~~
      | |
      | struct ttm_placement *
./include/drm/ttm/ttm_bo_api: 322:35: Note: Requires type 'uint32_t '{or'unsigned int'}, but argument type is'struct ttm_placement * '
  322 | uint32_t alignment, struct ttm_operation_ctx * ctx,
      | ~~~~~~~~~^~~~~~~~~
/var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gem/i915_gem_ttm.c: 1234:46: Warning: Assign integer to pointer when passing 6th argument of 'ttm_bo_init_reserved ', no type conversion [-Wint-conversion]
 1234 | page_size > > PAGE_SHIFT,
./include/drm/ttm/ttm_bo_api: 322:72: Note: Type'struct ttm_operation_ctx * 'is required, but the argument type is' resource_size_t '{or'long long unsigned int'}
  322 | uint32_t alignment, struct ttm_operation_ctx * ctx,
      | ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gem/i915_gem_ttm.c: 1235:36: Error: Converting between incompatible pointer types when passing 7th argument to 'ttm_bo_init_reserved '[-Werror = incompatible-pointer-types]
 1235 | & ctx, null, null, i915_ttm_bo_destroy);
      | ^~~~
      | |
      | struct ttm_operation_ctx *
./include/drm/ttm/ttm_bo_api: 323:43: Note: Type'struct sg_table * 'is required, but the argument type is'struct ttm_operation_ctx *'
  323 | struct sg_table * sg, struct dma_resv * resv,
      | ~~~~~~~~~~~~~~~~~^~
/var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gem/i915_gem_ttm.c: 1232:15: Error: Too many arguments for function 'ttm_bo_init_reserved '
 1232 | ret = ttm_bo_init_reserved (& i915- > bdev, i915_gem_to_ttm (obj), size,
      | ^~~~~~~~~~~~~~~~~~~~
./include/drm/ttm/ttm_bo_api: 320:5: Note: Declare here
  320 | int ttm_bo_init_reserved (struct ttm_device * bdev, struct ttm_buffer_object * bo,
      | ^~~~~~~~~~~~~~~~~~~~
Cc1: Some warnings are treated as errors
Make [3 ]: *** [ scripts/Makefile.build:250:/var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gem/i915_gem_ttm] Error 1
Make [3 ]: *** Waiting for unfinished tasks....
/var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gem/i915_gem_userptr.c: In function 'probe_range ':
/var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gem/i915_gem_userptr.c: 434:54: Error: 'struct vm_area_struct' has no member named vm_next
  434 | for (vma = find_vma (mm, addr); vma; vma = vma- > vm_next) {
      | ^~
Make [3 ]: *** [ scripts/Makefile.build:250:/var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gem/i915_gem_userptr] Error 1
/var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_gem_gtt.c: In function 'random_offset ':
/var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_gem_gtt.c: 142:32: Error: implicit declaration of function 'get_random_int '; did you mean 'get_random_long'? [-Werror = implicit-function-declaration]
  142 | addr = get_random_int ();
      | ^~~~~~~~~~~~~~
      | get_random_long
Cc1: Some warnings are treated as errors
Make [3 ]: *** [ scripts/Makefile.build:250:/var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_gem_gtt] Error 1
/var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gem/i915_gem_execbuffer.c: In function 'gen8_dispatch_bsd_engine ':
/var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gem/i915_gem_execbuffer.c: 2422:25: Error: implicit declaration of function 'get_random_int '; did you mean 'get_random_long'? [-Werror = implicit-function-declaration]
 2422 | get_random_int () % num_vcs_engines (dev_priv);
      | ^~~~~~~~~~~~~~
      | get_random_long
Cc1: Some warnings are treated as errors
```

zhtengw, 2022-12-15:
The [new release](https://github.com/intel/linux-intel-lts/releases/tag/lts-v5.15.71-adl-linux-221121T044440Z) from linux-intel-lts can build with linux-6.1 with some modification. I may update my branch at some time.

zhtengw, 2022-12-15:
I have updated on [my repo](https://github.com/zhtengw/i915-sriov-dkms/tree/6.1). You could have a try.

Alion548, 2022-12-15:
> I have updated on [my repo](https://github.com/zhtengw/i915-sriov-dkms/tree/6.1). You could have a try.
It works fine!

strongtz, 2022-12-23:
@zhtengw Hi, would you maybe consider making a pull request? Thanks!

zhtengw, 2022-12-23:
> @zhtengw Hi, would you maybe consider making a pull request? Thanks!

Sure, #21 .

Githopp, 2022-12-27:
For Kernel 6.1.1 compiling does not work for me. In the make.log it states that there is no makefile. (Used the aur script)
Can anybody confirm?

wanlce, 2022-12-29:
Arch Linux  
i915-sriov-dkms-git
-----
The i915 kernel module will be available on reboot.
:: 正在运行事务后钩子函数...
(1/2) Arming ConditionNeedsUpdate...
(2/2) Install DKMS modules
==> dkms install --no-depmod i915-sriov-dkms/5.15.71 -k 6.1.1-zen1-1.1-zen
Error! Bad return status for module build on kernel: 6.1.1-zen1-1.1-zen (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/5.15.71/build/make.log for more information.
==> WARNING: `dkms install --no-depmod i915-sriov-dkms/5.15.71 -k 6.1.1-zen1-1.1-zen' exited 10
==> dkms install --no-depmod i915-sriov-dkms/5.15.71 -k 6.2.0-rc1-next-20221226-1-next-git
Error! Bad return status for module build on kernel: 6.2.0-rc1-next-20221226-1-next-git (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/5.15.71/build/make.log for more information.
==> WARNING: `dkms install --no-depmod i915-sriov-dkms/5.15.71 -k 6.2.0-rc1-next-20221226-1-next-git' exited 10
==> dkms install --no-depmod i915-sriov-dkms/5.15.71 -k 6.1.1-arch1-1.1
Error! Bad return status for module build on kernel: 6.1.1-arch1-1.1 (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/5.15.71/build/make.log for more information.
==> WARNING: `dkms install --no-depmod i915-sriov-dkms/5.15.71 -k 6.1.1-arch1-1.1' exited 10

-------
~►tail -n 200 /var/lib/dkms/i915-sriov-dkms/5.15.71/build/make.log                  14:44
DKMS make.log for i915-sriov-dkms-5.15.71 for kernel 6.1.1-zen1-1.1-zen (x86_64)
2022年 12月 29日 星期四 14:40:28 CST
make: 进入目录“/usr/lib/modules/6.1.1-zen1-1.1-zen/build”
scripts/Makefile.build:44: /var/lib/dkms/i915-sriov/6.1/build/Makefile: 没有那个文件或目录
make[1]: *** 没有规则可制作目标“/var/lib/dkms/i915-sriov/6.1/build/Makefile”。 停止。
make: *** [Makefile:1990：/var/lib/dkms/i915-sriov/6.1/build] 错误 2
make: 离开目录“/usr/lib/modules/6.1.1-zen1-1.1-zen/build”

zx900930, 2023-01-16:
If you are using debian, the `CONFIG_DRM_I915_PXP` flag will make the compilation failed because debian kernels doesn't includes those flags by default.

closed 2023-05-04 by strongtz