strongtz/i915-sriov-dkms #19: Cannot modify sriov_numvfs

s1idewhist1e, 2022-12-14:
When I attempt to modify sriov_numvfs to change the number of virtual functions I get an error saying there is no such file or directory.

Attempting to echo into it using `tee`
```bash
> echo 2 | sudo tee -a /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs 
2
tee: '/sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs': No such file or directory
```

Any ideas on why this could be happening and what I can do?

kernkraft235, 2023-02-18:
So I just figured this out. Even though I had dkms installed and new i915 built it wouldnt override the built in driver by default

Run:
dkms status | grep i915 ## You will see dkms in the output if using new driver
dkms autoinstall --force ## overrides default module
rmmod i915 ## unloads old i915 and saves you a reboot
modprobe i915 ## loads current i915 which you just installed

kernkraft235, 2023-02-18:
Also it appears you havent added i915.enable_guc=7 to you boot options

closed 2023-05-04 by strongtz