strongtz/i915-sriov-dkms #22: i7 1260p 支持核显直通吗

hantianxiaoxue, 2022-12-25:
i7 1260p 支持核显直通吗？pve安装win10虚拟机，把核显给win10使用，能实现吗？

michael-pptf, 2023-03-17:
I can't see why not. The iGPU is the same across the board for Alder Lake/Raptor Lake, but you are using an embedded system, and this DKMS module (or anything SR-IOV) requires correct BIOS implementation to work. 

I don't know what kind of device you are running, but just look into the BIOS to see if you can find an option called "SR-IOV". If it isn't there, chances are it doesn't have the BIOS implementation for it, but there is a slim chance that's enabled by default when you enable both VT-x and VT-d.

Give it a try with a spare SSD to install Proxmox on. The worst thing can happen is you have to swap the SSD again.

hantianxiaoxue, 2023-05-10:
> I can't see why not. The iGPU is the same across the board for Alder Lake/Raptor Lake, but you are using an embedded system, and this DKMS module (or anything SR-IOV) requires correct BIOS [...]

谢谢，确定N305处理器是可以用的

closed 2023-05-10 by strongtz