strongtz/i915-sriov-dkms #29: Unable to configure virtual functions

FabioEight, 2023-01-11:
Hello, 
I built and installed the modules following the suggestion reported in #25 in my PVE with kernel 6.1.
Now, I was trying to set up the virtual functions but I have this error:

```
root@pve:~# cat /sys/devices/pci0000\:00/0000\:00\:02.0/sriov_numvfs 
0
root@pve:~# echo 2 > /sys/devices/pci0000\:00/0000\:00\:02.0/sriov_numvfs 
-bash: echo: write error: No such file or directory

root@pve:~# lspci
00:02.0 VGA compatible controller: Intel Corporation Device 46a6 (rev 0c)

root@pve:~# dmesg | grep "00:02"
[    0.687972] pci 0000:00:02.0: [8086:46a6] type 00 class 0x030000
[    0.687980] pci 0000:00:02.0: reg 0x10: [mem 0x603c000000-0x603cffffff 64bit]
[    0.687985] pci 0000:00:02.0: reg 0x18: [mem 0x4000000000-0x400fffffff 64bit pref]
[    0.687988] pci 0000:00:02.0: reg 0x20: [io  0x3000-0x303f]
[    0.688002] pci 0000:00:02.0: BAR 2: assigned to efifb
[    0.688003] pci 0000:00:02.0: DMAR: Skip IOMMU disabling for graphics
[    0.688005] pci 0000:00:02.0: Video device with shadowed ROM at [mem 0x000c0000-0x000dffff]
[    0.688028] pci 0000:00:02.0: reg 0x344: [mem 0x00000000-0x00ffffff 64bit]
[    0.688029] pci 0000:00:02.0: VF(n) BAR0 space: [mem 0x00000000-0x06ffffff 64bit] (contains BAR0 for 7 VFs)
[    0.688033] pci 0000:00:02.0: reg 0x34c: [mem 0x00000000-0x1fffffff 64bit pref]
[    0.688034] pci 0000:00:02.0: VF(n) BAR2 space: [mem 0x00000000-0xdfffffff 64bit pref] (contains BAR2 for 7 VFs)
[    1.159309] pci 0000:00:02.0: vgaarb: setting as boot VGA device
[    1.159309] pci 0000:00:02.0: vgaarb: bridge control possible
[    1.159309] pci 0000:00:02.0: vgaarb: VGA device added: decodes=io+mem,owns=io+mem,locks=none
[    1.166163] system 00:02: [io  0x1854-0x1857] has been reserved
[    1.166817] pnp 00:03: disabling [mem 0xc0000000-0xcfffffff] because it overlaps 0000:00:02.0 BAR 9 [mem 0x00000000-0xdfffffff 64bit pref]
[    1.179500] pci 0000:00:02.0: BAR 9: assigned [mem 0x4020000000-0x40ffffffff 64bit pref]
[    1.179504] pci 0000:00:02.0: BAR 7: assigned [mem 0x4010000000-0x4016ffffff 64bit]
[    1.180384] pci_bus 0000:02: resource 0 [io  0x4000-0x4fff]
[    1.180385] pci_bus 0000:02: resource 1 [mem 0x78000000-0x841fffff]
[    1.180385] pci_bus 0000:02: resource 2 [mem 0x6020000000-0x603bffffff 64bit pref]
[    1.182107] pci 0000:00:02.0: Adding to iommu group 0
[  447.996075] pci 0000:00:02.0: no driver bound to device; cannot configure SR-IOV
[  492.919810] pci 0000:00:02.0: no driver bound to device; cannot configure SR-IOV

root@pve:~# cat /etc/kernel/cmdline 
root=ZFS=rpool/ROOT/pve-1 boot=zfs quiet intel_iommu=on iommu=pt i915.enable_guc=7
```

Any hint please?
Thanks!

vnxme, 2023-01-26:
Hello, @FabioEight! 

Thank you for sharing it. I have the same issue on my i7-1280p. I tried to resolve the BAR 9 overlapping/assignment by adding `acpi_enforce_resources=lax` and `realloc=on` to /etc/default/grub, but it barely changed anything. I also increased/decreased the aperture size in BIOS, but I didn't notice any effect on this issue. 

My log is now as follows:

```
root@h540:~# dmesg -T | grep :00:02
[Thu Jan 26 14:53:38 2023] pci 0000:00:02.0: [8086:46a6] type 00 class 0x030000
[Thu Jan 26 14:53:38 2023] pci 0000:00:02.0: reg 0x10: [mem 0x41000000-0x41ffffff 64bit]
[Thu Jan 26 14:53:38 2023] pci 0000:00:02.0: reg 0x18: [mem 0x50000000-0x5fffffff 64bit pref]
[Thu Jan 26 14:53:38 2023] pci 0000:00:02.0: reg 0x20: [io  0x4000-0x403f]
[Thu Jan 26 14:53:38 2023] pci 0000:00:02.0: DMAR: Skip IOMMU disabling for graphics
[Thu Jan 26 14:53:38 2023] pci 0000:00:02.0: Video device with shadowed ROM at [mem 0x000c0000-0x000dffff]
[Thu Jan 26 14:53:38 2023] pci 0000:00:02.0: reg 0x344: [mem 0x00000000-0x00ffffff 64bit]
[Thu Jan 26 14:53:38 2023] pci 0000:00:02.0: VF(n) BAR0 space: [mem 0x00000000-0x06ffffff 64bit] (contains BAR0 for 7 VFs)
[Thu Jan 26 14:53:38 2023] pci 0000:00:02.0: reg 0x34c: [mem 0x00000000-0x1fffffff 64bit pref]
[Thu Jan 26 14:53:38 2023] pci 0000:00:02.0: VF(n) BAR2 space: [mem 0x00000000-0xdfffffff 64bit pref] (contains BAR2 for 7 VFs)
[Thu Jan 26 14:53:38 2023] pci 0000:00:02.0: vgaarb: setting as boot VGA device
[Thu Jan 26 14:53:38 2023] pci 0000:00:02.0: vgaarb: bridge control possible
[Thu Jan 26 14:53:38 2023] pci 0000:00:02.0: vgaarb: VGA device added: decodes=io+mem,owns=io+mem,locks=none
[Thu Jan 26 14:53:38 2023] pnp 00:06: disabling [mem 0xc0000000-0xcfffffff] because it overlaps 0000:00:02.0 BAR 9 [mem 0x00000000-0xdfffffff 64bit pref]
[Thu Jan 26 14:53:38 2023] pci 0000:00:02.0: BAR 9: no space for [mem size 0xe0000000 64bit pref]
[Thu Jan 26 14:53:38 2023] pci 0000:00:02.0: BAR 9: failed to assign [mem size 0xe0000000 64bit pref]
[Thu Jan 26 14:53:38 2023] pci 0000:00:02.0: BAR 7: assigned [mem 0x43000000-0x49ffffff 64bit]
[Thu Jan 26 14:53:38 2023] pci 0000:00:02.0: BAR 9: no space for [mem size 0xe0000000 64bit pref]
[Thu Jan 26 14:53:38 2023] pci 0000:00:02.0: BAR 9: failed to assign [mem size 0xe0000000 64bit pref]
[Thu Jan 26 14:53:38 2023] pci 0000:00:02.0: Adding to iommu group 0
[Thu Jan 26 14:53:42 2023] i915 0000:00:02.0: [drm] VT-d active for gfx access
[Thu Jan 26 14:53:42 2023] i915 0000:00:02.0: vgaarb: deactivate vga console
[Thu Jan 26 14:53:42 2023] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[Thu Jan 26 14:53:42 2023] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[Thu Jan 26 14:53:42 2023] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc_ver2_16.bin (v2.16)
[Thu Jan 26 14:53:42 2023] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[Thu Jan 26 14:53:43 2023] i915 0000:00:02.0: [drm] *ERROR* Zero GuC log crash dump size!
[Thu Jan 26 14:53:43 2023] i915 0000:00:02.0: [drm] *ERROR* Zero GuC log debug size!
[Thu Jan 26 14:53:43 2023] i915 0000:00:02.0: [drm] GuC error state capture buffer maybe too small: 2097152 < 2557128 (min = 852376)
[Thu Jan 26 14:53:43 2023] i915 0000:00:02.0: [drm] GuC firmware i915/adlp_guc_70.bin version 70.5.1
[Thu Jan 26 14:53:43 2023] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc.bin version 7.9.3
[Thu Jan 26 14:53:43 2023] i915 0000:00:02.0: [drm] HuC authenticated
[Thu Jan 26 14:53:43 2023] i915 0000:00:02.0: [drm] GuC submission enabled
[Thu Jan 26 14:53:43 2023] i915 0000:00:02.0: [drm] GuC SLPC enabled
[Thu Jan 26 14:53:43 2023] i915 0000:00:02.0: [drm] GuC RC: enabled
[Thu Jan 26 14:53:43 2023] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[Thu Jan 26 14:53:43 2023] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[Thu Jan 26 14:53:43 2023] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[Thu Jan 26 14:53:43 2023] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[Thu Jan 26 14:53:43 2023] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[Thu Jan 26 14:53:43 2023] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[Thu Jan 26 14:53:43 2023] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
```

zhtengw, 2023-01-29:
Hi @FabioEight . I notice that `Kernel driver in use: vfio-pci` which means the video card is not using i915 driver. 
Before enabling the VFs of video card, you should stop passing through it to any guest VMs. And remove any `options vfio-pci ids=8086:46a6` in .conf files in `/etc/modprobe.d/`.
```
# lspci -kk -s 0000:00:02
00:02.0 VGA compatible controller: Intel Corporation Device 4692 (rev 0c)
        DeviceName: Onboard - Video
        Kernel driver in use: i915
        Kernel modules: i915
```

Byolock, 2023-02-26:
@vnxme did you found a solution to your issue? My dmesg output looks very similar so I think we could be having the same problem :

```
[Sun Feb 26 18:02:42 2023] pci 0000:00:02.0: [8086:4692] type 00 class 0x030000
[Sun Feb 26 18:02:42 2023] pci 0000:00:02.0: reg 0x10: [mem 0x41000000-0x41ffffff 64bit]
[Sun Feb 26 18:02:42 2023] pci 0000:00:02.0: reg 0x18: [mem 0x50000000-0x5fffffff 64bit pref]
[Sun Feb 26 18:02:42 2023] pci 0000:00:02.0: reg 0x20: [io  0x5000-0x503f]
[Sun Feb 26 18:02:42 2023] pci 0000:00:02.0: DMAR: Skip IOMMU disabling for graphics
[Sun Feb 26 18:02:42 2023] pci 0000:00:02.0: Video device with shadowed ROM at [mem 0x000c0000-0x000dffff]
[Sun Feb 26 18:02:42 2023] pci 0000:00:02.0: reg 0x344: [mem 0x00000000-0x00ffffff 64bit]
[Sun Feb 26 18:02:42 2023] pci 0000:00:02.0: VF(n) BAR0 space: [mem 0x00000000-0x06ffffff 64bit] (contains BAR0 for 7 VFs)
[Sun Feb 26 18:02:42 2023] pci 0000:00:02.0: reg 0x34c: [mem 0x00000000-0x1fffffff 64bit pref]
[Sun Feb 26 18:02:42 2023] pci 0000:00:02.0: VF(n) BAR2 space: [mem 0x00000000-0xdfffffff 64bit pref] (contains BAR2 for 7 VFs)
[Sun Feb 26 18:02:42 2023] pci 0000:00:02.0: vgaarb: setting as boot VGA device
[Sun Feb 26 18:02:42 2023] pci 0000:00:02.0: vgaarb: bridge control possible
[Sun Feb 26 18:02:42 2023] pci 0000:00:02.0: vgaarb: VGA device added: decodes=io+mem,owns=io+mem,locks=none
[Sun Feb 26 18:02:42 2023] pnp 00:04: disabling [mem 0xc0000000-0xcfffffff] because it overlaps 0000:00:02.0 BAR 9 [mem 0x00000000-0xdfffffff 64bit pref]
[Sun Feb 26 18:02:42 2023] pci 0000:00:02.0: BAR 9: no space for [mem size 0xe0000000 64bit pref]
[Sun Feb 26 18:02:42 2023] pci 0000:00:02.0: BAR 9: failed to assign [mem size 0xe0000000 64bit pref]
[Sun Feb 26 18:02:42 2023] pci 0000:00:02.0: BAR 7: assigned [mem 0x43000000-0x49ffffff 64bit]
[Sun Feb 26 18:02:42 2023] pci 0000:00:02.0: BAR 9: no space for [mem size 0xe0000000 64bit pref]
[Sun Feb 26 18:02:42 2023] pci 0000:00:02.0: BAR 9: failed to assign [mem size 0xe0000000 64bit pref]
[Sun Feb 26 18:02:42 2023] pci 0000:00:02.0: BAR 9: no space for [mem size 0xe0000000 64bit pref]
[Sun Feb 26 18:02:42 2023] pci 0000:00:02.0: BAR 9: failed to assign [mem size 0xe0000000 64bit pref]
[Sun Feb 26 18:02:42 2023] pci 0000:00:02.0: Adding to iommu group 0
[Sun Feb 26 18:02:55 2023] i915 0000:00:02.0: [drm] VT-d active for gfx access
[Sun Feb 26 18:02:55 2023] i915 0000:00:02.0: vgaarb: deactivate vga console
[Sun Feb 26 18:02:55 2023] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[Sun Feb 26 18:02:55 2023] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[Sun Feb 26 18:02:55 2023] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[Sun Feb 26 18:02:55 2023] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
[Sun Feb 26 18:02:55 2023] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
[Sun Feb 26 18:02:55 2023] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!
[Sun Feb 26 18:02:55 2023] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
[Sun Feb 26 18:02:55 2023] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!
[Sun Feb 26 18:02:55 2023] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_70.bin version 70.5.1
[Sun Feb 26 18:02:55 2023] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc.bin version 7.9.3
[Sun Feb 26 18:02:55 2023] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
[Sun Feb 26 18:02:55 2023] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!
[Sun Feb 26 18:02:55 2023] i915 0000:00:02.0: [drm] HuC authenticated
[Sun Feb 26 18:02:55 2023] i915 0000:00:02.0: [drm] GuC submission enabled
[Sun Feb 26 18:02:55 2023] i915 0000:00:02.0: [drm] GuC SLPC enabled
[Sun Feb 26 18:02:55 2023] i915 0000:00:02.0: [drm] GuC RC: enabled
[Sun Feb 26 18:02:55 2023] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[Sun Feb 26 18:02:55 2023] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[Sun Feb 26 18:02:55 2023] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[Sun Feb 26 18:02:55 2023] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[Sun Feb 26 18:02:55 2023] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[Sun Feb 26 18:02:55 2023] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
```

Unfourtanetly im out of Ideas at this point.

closed 2023-05-04 by strongtz