strongtz/i915-sriov-dkms #33: mdev_*register_device Functions don't Exist in 6.1

sempervictus, 2023-02-14:
Building in-tree with LTO catches these couple of gems:
```
drivers/gpu/drm/i915/gvt/gvt.c:138:2: error: call to undeclared function 'mdev_unregister_device'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
drivers/gpu/drm/i915/gvt/gvt.c:243:8: error: call to undeclared function 'mdev_register_device'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
```
6.1 changed this to `mdev_*register_driver`

zhtengw, 2023-02-14:
It is because `CONFIG_DRM_I915_GVT_KVMGT` is enabled in your linux config. Since GVT-g and SR-IOV would not exist in one chip on the same time, you can just disable `CONFIG_DRM_I915_GVT_KVMGT`.

sempervictus, 2023-02-14:
Thanks. That `gvt.c` file however does appear to be out of its era in terms of kernel revision :)

zhtengw, 2023-02-15:
> With _that_ config option disabled, still failing to build in-tree (i can't build DKMS with LTO build trimming the symbols, being my guess) with:
> 
> ```
>   GEN     .vmlinux.objs
>   MODPOST Module.symvers
> ERROR: modpost: "intel_pxp_tee_stream_message" [drivers/gpu/drm/i915/i915.ko] undefined!
> WARNING: modpost: modpost: Found 1 writable function pointer.
> To see full details build your kernel with:
> 'make CONFIG_DEBUG_WRITABLE_FUNCTION_POINTERS_VERBOSE=y'
> ```
> 
> which has a header defined in `drivers/gpu/drm/i915/pxp/intel_pxp_tee.h` but has no function body in the code.

`intel_pxp_tee_stream_message` is introduced to linux mainline only since [6.2-rc1](https://github.com/torvalds/linux/commit/9058f9d795ea9ad59fd579249a6d724d78dfeaf8). So [deleted its function body](https://github.com/strongtz/i915-sriov-dkms/commit/cb48a68db44d5a20b715a380d786151d78cc699d#diff-e9b153d3d145eac216e72a04bb94f5b0c5aa489e75144a60264de07e89e07e1c) to build with linux-6.1. If the compiler complain that `intel_pxp_tee_stream_message` undefined, you could just delete the definition in `drivers/gpu/drm/i915/pxp/intel_pxp_tee.h`.

> 
> How is this repo currently constructed? Where is the code coming from and how is it being ported to the 6.1 branch? Might be able to help reconcile some of this...
It shows in the first line of `README.md`.

sempervictus, 2023-02-15:
Thanks for the heads up.
Dont know if we have some sort of divergence, but the `README.md` in my repo says this is from Intel's 5.15 LTS repo (at the .71 release) which is why i'm wondering how the 6.1 version is constructed as there's a lot of delta introduced in 6 and 6.1, and they don't seem to have a 6.1 branch/tag yet.

zhtengw, 2023-02-15:
> Thanks for the heads up. Dont know if we have some sort of divergence, but the `README.md` in my repo says this is from Intel's 5.15 LTS repo (at the .71 release) which is why i'm wondering how the [...]

It is done manually. SR-IOV was first enabled in `lts-v5.15.49-adl-linux-220826T092047Z`, in which many depended functions were introduced to linux mainline since linux-5.19. When linux-6.1 released, I found out that `i915` in `lts-v5.15.49-adl-linux-220826T092047Z` can not be built against linux-6.1 with little modification. But `i915` in `lts-v5.15.71-linux-221108T021454Z` can be, so I merged `i915` from `lts-v5.15.71-linux-221108T021454Z` into this repo. And when linux-6.2 released, codes from `lts-v5.15.86-linux-230131T143409Z` will be merged.

zhtengw, 2023-02-19:
> Can [this branch from the lts repo](https://github.com/intel/linux-intel-lts/tree/lts-v6.1.8-linux-230201T082419Z/drivers/gpu/drm/i915) be used instead? Looks like it is set up to build against 6.1 and has SR-IOV support built in.

It is [done](https://github.com/zhtengw/i915-sriov-dkms/tree/6.1), and a PR #34 is opened.

closed 2023-05-04 by strongtz