strongtz/i915-sriov-dkms #37: dkms install error

oniLand, 2023-02-20:
```
➜  ~ dkms install -m i915-sriov -v dkms

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j12 KERNELRELEASE=6.1.0-1-pve -j12 -C /lib/modules/6.1.0-1-pve/build M=/var/lib/dkms/i915-sriov/dkms/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.0-1-pve (x86_64)
Consult /var/lib/dkms/i915-sriov/dkms/build/make.log for more information.
➜  ~ cat /var/lib/dkms/i915-sriov/dkms/build/make.log                              
DKMS make.log for i915-sriov-dkms for kernel 6.1.0-1-pve (x86_64)
Mon 20 Feb 2023 11:35:15 PM CST
make: Entering directory '/usr/src/linux-headers-6.1.0-1-pve'
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_driver.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_drm_client.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_config.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_getparam.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_hwmon.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_ioctl.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_irq.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_mitigations.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_module.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_params.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_pci.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_scatterlist.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_suspend.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_switcheroo.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_sysfs.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_sysrq.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_utils.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_device_info.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_dram.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_memory_region.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_pch.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_pcode.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_pm.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_region_ttm.o
/var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_memory_region.c: In function ‘__iopagetest’:
/var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_memory_region.c:41:13: error: implicit declaration of function ‘get_random_u32_below’; did you mean ‘get_random_u32_wait’? [-Werror=implicit-function-declaration]
   41 |  int byte = get_random_u32_below(pagesize);
      |             ^~~~~~~~~~~~~~~~~~~~
      |             get_random_u32_wait
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_runtime_pm.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_sbi.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_step.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_uncore.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_wakeref.o
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:258: /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_memory_region.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:1996: /var/lib/dkms/i915-sriov/dkms/build] Error 2
make: Leaving directory '/usr/src/linux-headers-6.1.0-1-pve'
➜  ~ vi /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_memory_region.c                            
➜  ~ 
➜  ~ vi /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_memory_region.c
➜  ~ uname -a                          
Linux pve 6.1.0-1-pve #1 SMP PREEMPT_DYNAMIC PVE 6.1.0-1 (Tue, 13 Dec 2022 15:08:53 +0 x86_64 GNU/Linux
```

oniLand, 2023-02-21:
hello , i change kernel version and rebuild (6.1.0.1 -> 6.1.10.1), but also install error.

```
➜  ~ cat /var/lib/dkms/i915-sriov/dkms/build/make.log                         
DKMS make.log for i915-sriov-dkms for kernel 6.1.10-1-pve (x86_64)
Tue 21 Feb 2023 10:08:06 PM CST
make: Entering directory '/usr/src/linux-headers-6.1.10-1-pve'
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_driver.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_drm_client.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_config.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_getparam.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_hwmon.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_ioctl.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_irq.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_mitigations.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_module.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_params.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_pci.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_scatterlist.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_suspend.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_switcheroo.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_sysfs.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_sysrq.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_utils.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_device_info.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_dram.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_memory_region.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_pch.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_pcode.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_pm.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_region_ttm.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_runtime_pm.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_sbi.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_step.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_uncore.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_wakeref.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/vlv_sideband.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/vlv_suspend.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_memcpy.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_mm.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_sw_fence.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_sw_fence_work.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_syncmap.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_user_extensions.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_ioc32.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_debugfs.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_debugfs_params.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/display/intel_display_debugfs.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/display/intel_pipe_crc.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_pmu.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gt/gen2_engine_cs.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gt/gen6_engine_cs.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gt/gen6_ppgtt.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gt/gen7_renderclear.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gt/gen8_engine_cs.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gt/gen8_ppgtt.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gt/intel_breadcrumbs.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gt/intel_context.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gt/intel_context_sseu.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gt/intel_engine_cs.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gt/intel_engine_heartbeat.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gt/intel_engine_pm.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gt/intel_engine_user.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gt/intel_execlists_submission.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gt/intel_ggtt.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gt/intel_ggtt_fencing.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gt/intel_ggtt_gmch.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/gt/intel_gt.o
[... cut ...]
      |                                          ^
compilation terminated.
make[1]: *** [scripts/Makefile.build:258: /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_trace_points.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:2009: /var/lib/dkms/i915-sriov/dkms/build] Error 2
make: Leaving directory '/usr/src/linux-headers-6.1.10-1-pve'
➜  ~ 
➜  ~ uname -a
Linux pve 6.1.10-1-pve #1 SMP PREEMPT_DYNAMIC PVE 6.1.10-1 (2023-02-07T00:00Z) x86_64 GNU/Linux
```

oniLand, 2023-02-21:
update: fix it.    

`#define TRACE_INCLUDE_PATH  `  has wrong path.
in
```
i915_trace.h
intel_display_trace.h
trace.h
```

stnd, 2023-03-01:
> 

What did you do to fix it? I am having the same issue.

zhtengw, 2023-03-01:
> 
> What did you do to fix it? I am having the same issue.

This issue was fixed by #40 . Paste your kernel version and dkms log will be helpful.

stnd, 2023-03-03:
> 
@zhtengw 

CPU: i5-12500 w UHD770
Kernel 6.1.0-1-pve

Using latest master repository, I was able to install but it does not seem to work as intended. Any advice would be appreciated. 

To install I set PACKAGE_NAME and PACKAGE_VERSION to the following in dkms.conf

```
PACKAGE_NAME="i915-sriov"
PACKAGE_VERSION="dkms"

MAKE[0]="make -j$(nproc) -C ${kernel_source_dir} M=${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build KVER=${kernelver}"
CLEAN="make -j$(nproc) -C ${kernel_source_dir} M=${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build KVER=${kernelver} clean"

BUILT_MODULE_NAME[0]="i915"
DEST_MODULE_LOCATION[0]=/kernel/drivers/gpu/drm/i915

AUTOINSTALL=yes
```

Running dkms status I get
`i915-sriov, dkms, 6.1.0-1-pve, x86_64: installed`

Running: dmesg  | grep -i i915 I get some errors Missing GuC-Err-Cap reglist
```
[    0.000000] Command line: initrd=\EFI\proxmox\6.1.0-1-pve\initrd.img-6.1.0-1-pve root=ZFS=rpool/ROOT/pve-1 boot=zfs quiet intel_iommu=on iommu=pt i915.enable_guc=7
[    0.074758] Kernel command line: initrd=\EFI\proxmox\6.1.0-1-pve\initrd.img-6.1.0-1-pve root=ZFS=rpool/ROOT/pve-1 boot=zfs quiet intel_iommu=on iommu=pt i915.enable_guc=7
[    3.676081] i915 0000:00:02.0: enabling device (0006 -> 0007)
[    3.676824] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    3.676843] i915 0000:00:02.0: vgaarb: deactivate vga console
[    3.676874] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    3.677297] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=mem
[    3.677886] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[    3.678866] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
[    3.770715] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
[    3.770718] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!
[    3.772158] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
[    3.772160] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!
[    3.772549] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_70.bin version 70.5.1
[    3.772550] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc.bin version 7.9.3
[    3.772985] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
[    3.772986] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!
[    3.774788] i915 0000:00:02.0: [drm] HuC authenticated
[    3.775056] i915 0000:00:02.0: [drm] GuC submission enabled
[    3.775057] i915 0000:00:02.0: [drm] GuC SLPC enabled
[    3.775381] i915 0000:00:02.0: [drm] GuC RC: enabled
[    3.776250] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[    3.776325] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    3.777319] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    3.822634] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    3.823244] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    3.823326] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
```

oniLand, 2023-03-03:
> 

same as you.
[image]
cpu is i5-1235u.

stnd, 2023-03-04:
> > 
> 
> same as you. [image] cpu is i5-1235u.

I don't know if we have the same issue. But I was able to get it working by using the 6.1.10-1-pve kernel on proxmox. I really do not know enough about linux and dkms to say why, but at this point i'm happy that i see the virtualized gpu.

```
00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:4690] (rev 0c)
00:02.1 VGA compatible controller [0300]: Intel Corporation Device [8086:4690] (rev 0c)
00:02.2 VGA compatible controller [0300]: Intel Corporation Device [8086:4690] (rev 0c)
00:02.3 VGA compatible controller [0300]: Intel Corporation Device [8086:4690] (rev 0c)
00:02.4 VGA compatible controller [0300]: Intel Corporation Device [8086:4690] (rev 0c)
00:02.5 VGA compatible controller [0300]: Intel Corporation Device [8086:4690] (rev 0c)
00:02.6 VGA compatible controller [0300]: Intel Corporation Device [8086:4690] (rev 0c)
00:02.7 VGA compatible controller [0300]: Intel Corporation Device [8086:4690] (rev 0c)
```

XanthuxSmith, 2023-03-07:
I got very similar error as @oniLand. my CPU is i7-1265U.
On Proxmox 7.3.-1, kernal 6.1.10-1-pve

enpioodada, 2023-03-31:
bios打开Above 4GB MMIO BIOS assignment

mlavio829, 2023-06-28:
Same issue on 11th gen rocketlake

closed 2023-07-01 by oniLand