strongtz/i915-sriov-dkms #42: 11800H + Win 11 super laggy on Fedora 37.

sieskei, 2023-03-02:
Hello,
I successfully install and bind new function iGPU (0000:00:02.1) to vfio-pci.
BUT when I'm running Windows 11 with Spice (Virtio) + new iGPU, **host** and **guest** become super super laggy.

Notes:
1. use -cpu host

What I tried:
1. various intel drivers
2. run without spice (no graphical output) only iGPU

I get error code 43 without HyperV features.

Thanks in advance.
Stay tuned for more info and logs.

sieskei, 2023-03-03:
> [    2.514220] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
[    2.514222] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!
[    2.515353] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
[    2.515354] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!
[    2.515766] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_70.1.1.bin version 70.1.1
[    2.515768] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc_7.9.3.bin version 7.9.3
[    2.516215] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
[    2.516216] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!
[    2.518166] i915 0000:00:02.0: [drm] HuC authenticated
[    2.518713] i915 0000:00:02.0: [drm] GuC submission enabled
[    2.518714] i915 0000:00:02.0: [drm] GuC SLPC enabled
[    2.519212] i915 0000:00:02.0: [drm] GuC RC: enabled
[    2.520052] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized

This is a cut from my **dmesg** logs, can someone help?

raldone01, 2023-03-03:
Maybe you should open the issue on linux-intel-lts too.

pagkly, 2023-08-17:
@sieskei what did you mean by "without Hyper-V features"? How did you get it to work on guest? Mine with 11th Gen Intel throws code 43 error. Thanks

closed 2024-12-29 by strongtz