strongtz/i915-sriov-dkms #45: Framebuffer/virtual display?

neeps, 2023-03-07:
I've been attempting to start gnome and gdk with iGPU passthrough, but X11 complains about missing framebuffer during bootup.
Anyone know whether there its possible to circumvent or workaround this in any way?

sieskei, 2023-03-07:
I think it's not possible at the moment because the virtual card has no output like gvt-g and I couldn't attach a virtual monitor to it in Ubuntu.

neeps, 2023-03-08:
Thank you @sieskei, for confirming that you are unable too. I'll close this issue for now then.

closed 2023-03-08 by neeps

FallingSnow, 2023-03-15:
Maybe you could use this? https://docs.kernel.org/gpu/vkms.html

neeps, 2023-03-16:
@FallingSnow Thank you!
That coupled with file permission a=+rw to /dev/dri/renderD128 solved problems i had with hardware acceleration for remote firefox.

tristan-k, 2023-03-19:
@neeps Can you describe the things you did step by step with the proper commands?

neeps, 2023-03-26:
Not anymore unfortunately, after a couple of reboots both gnome and sway refuse to work with vkms+Intel gpu

neeps, 2023-03-28:
If any of you are interested in pursuing such a setup, my setup is complaining about missing  /usr/lib/dri/vkms_dri.so.
waypipe client + server setup, and from hypervisor host this used to work:

waypipe --video=hw,bpf=120000,h264 --remote-node=/dev/dri/renderD128 ssh <IP> env WAYLAND_DISPLAY=wayland-1 MOZ_ENABLE_WAYLAND=1 MOZ_WAYLAND_DRM_DEVICE=/dev/dri/renderD128 MOZ_DRM_DEVICE=/dev/dri/renderD128 MOZ_DISABLE_RDD_SANDBOX=1 firefox --no-profile --new-instance

sieskei, 2023-03-28:
Hello @neeps,
I'm interested in understanding what you're doing, but everything is very unclear.
Please describe the steps related to VKMS.

I have compiled sr-iov i915 in the host and client, and the virtual video card is available in the guest. What do we do next?

FallingSnow, 2023-03-29:
> If any of you are interested in pursuing such a setup, my setup is complaining about missing /usr/lib/dri/vkms_dri.so. waypipe client + server setup, and from hypervisor host this used to work: [...]

If you are using qemu, you may want to try virtio gpu with opengl and spice enabled. There is a `/usr/lib/dri/virtio_gpu_dri.so`. You don't need vkms at that point. Then you might be able to run your command.