strongtz/i915-sriov-dkms #48: cannot write value for attribute devices/pci0000:00/0000:00:02.0/sriov_numvfs

SometingLight, 2023-03-10:
Hi all,

Been having difficulties with setting this up for weeks, I was using this how-to as a basis: https://www.michaelstinkerings.org/gpu-virtualization-with-intel-12th-gen-igpu-uhd-730/

CPU: Intel i5-13500
Motherboard: Gigabyte B660 Gaming X

Current Kernal and Headers
`root@SometingLight:~# uname -r
6.1.10-1-pve`

Compiled
`root@SometingLight:~# dkms status
i915-sriov, 6.1: added
i915-sriov-dkms, 6.1, 6.1.10-1-pve, x86_64: installed`

However this is where it goes wrong and the iGPU is not split
`root@SometingLight:~# systemctl status sysfsutils.service
● sysfsutils.service - LSB: Set sysfs variables from /etc/sysfs.conf
     Loaded: loaded (/etc/init.d/sysfsutils; generated)
     Active: active (exited) since Wed 2023-03-08 00:50:50 CET; 2 days ago
       Docs: man:systemd-sysv-generator(8)
    Process: 1091 ExecStart=/etc/init.d/sysfsutils start (code=exited, status=0/SUCCESS)
        CPU: 4ms

Mar 08 00:50:50 SometingLight systemd[1]: Starting LSB: Set sysfs variables from /etc/sysfs.conf...
Mar 08 00:50:50 SometingLight sysfsutils[1091]: Setting sysfs variables......
Mar 08 00:50:50 SometingLight sysfsutils[1137]: cannot write value for attribute devices/pci0000:00/0000:00:02.0/sriov_numvfs ...
Mar 08 00:50:50 SometingLight sysfsutils[1141]:  failed!
Mar 08 00:50:50 SometingLight sysfsutils[1091]: done.
Mar 08 00:50:50 SometingLight systemd[1]: Started LSB: Set sysfs variables from /etc/sysfs.conf.`

It is this line that concerns me: _**Mar 08 00:50:50 SometingLight sysfsutils[1137]: cannot write value for attribute devices/pci0000:00/0000:00:02.0/sriov_numvfs ...**_

The echo command also fails:
`root@SometingLight:~# echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
-bash: echo: write error: Numerical result out of range`

I have another GPU, this works perfectly in a Plex VM
`root@SometingLight:~# lspci | grep "VGA"
00:02.0 VGA compatible controller: Intel Corporation AlderLake-S GT1 (rev 0c)
01:00.0 VGA compatible controller: NVIDIA Corporation GP104GL [Quadro P4000] (rev a1)`

guc returns a tainted kernal but this seems normal
`root@SometingLight:~# dmesg | grep guc
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.1.10-1-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=7
[    0.058045] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.1.10-1-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=7
[    8.893775] Setting dangerous option enable_guc - tainting kernel
[    9.045631] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_70.bin version 70.5.1`

Any ideas?

zhtengw, 2023-03-11:
Could you show the contents of `.conf` files under `/etc/modprobe.d/` and the output of `lspci -vs 00:02.0`?

This may be caused by that the iGPU is passing through to any VM. Please make sure that **Kernel driver in use** of  **00:02.0** is `i915` instead of `vfio-pci`.

SometingLight, 2023-03-11:
```
root@SometingLight:~# more /etc/modprobe.d/*.conf
::::::::::::::
/etc/modprobe.d/dkms.conf
::::::::::::::
# modprobe information used for DKMS modules
#
# This is a stub file, should be edited when needed,
# used by default by DKMS.
::::::::::::::
/etc/modprobe.d/pve-blacklist.conf
::::::::::::::
# This file contains a list of modules which are not supported by Proxmox VE 

# nidiafb see bugreport https://bugzilla.proxmox.com/show_bug.cgi?id=701
blacklist nvidiafb
```

```
root@SometingLight:~#  lspci -vs 00:02.0
00:02.0 VGA compatible controller: Intel Corporation AlderLake-S GT1 (rev 0c) (prog-if 00 [VGA controller])
        DeviceName: Onboard - Video
        Subsystem: Gigabyte Technology Co., Ltd Device d000
        Flags: bus master, fast devsel, latency 0, IRQ 129, IOMMU group 0
        Memory at 41000000 (64-bit, non-prefetchable) [size=16M]
        Memory at 50000000 (64-bit, prefetchable) [size=256M]
        I/O ports at 5000 [size=64]
        Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
        Capabilities: [40] Vendor Specific Information: Len=0c <?>
        Capabilities: [70] Express Root Complex Integrated Endpoint, MSI 00
        Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable+ 64bit-
        Capabilities: [d0] Power Management version 2
        Capabilities: [100] Process Address Space ID (PASID)
        Capabilities: [200] Address Translation Service (ATS)
        Capabilities: [300] Page Request Interface (PRI)
        Capabilities: [320] Single Root I/O Virtualization (SR-IOV)
        Kernel driver in use: i915
        Kernel modules: i915
```

zhtengw, 2023-03-11:
All looks good. And I found out that there are no messages like `i915 0000:00:02.0: Running in SR-IOV PF mode` in your `dmesg` output. Did you run `update-initramfs -u` after installing the dkms module?
Try to update initramfs and reboot to test:
```
  update-initramfs -u
  pve-efiboot-tool refresh
```

SometingLight, 2023-03-11:
I did this before  but does not hurt to try again:

```
root@SometingLight:~# update-initramfs -u && pve-efiboot-tool refresh
update-initramfs: Generating /boot/initrd.img-6.1.10-1-pve
Running hook script 'zz-proxmox-boot'..
Re-executing '/etc/kernel/postinst.d/zz-proxmox-boot' in new private mount namespace..
No /etc/kernel/proxmox-boot-uuids found, skipping ESP sync.
Running hook script 'proxmox-auto-removal'..
Running hook script 'zz-proxmox-boot'..
Re-executing '/etc/kernel/postinst.d/zz-proxmox-boot' in new private mount namespace..
No /etc/kernel/proxmox-boot-uuids found, skipping ESP sync.
```

Post reboot, same thing
```
root@SometingLight:~# uptime
 09:25:38 up 1 min,  1 user,  load average: 1.54, 0.65, 0.24
root@SometingLight:~# systemctl status sysfsutils.service
● sysfsutils.service - LSB: Set sysfs variables from /etc/sysfs.conf
     Loaded: loaded (/etc/init.d/sysfsutils; generated)
     Active: active (exited) since Sat 2023-03-11 09:24:15 CET; 1min 27s ago
       Docs: man:systemd-sysv-generator(8)
    Process: 1083 ExecStart=/etc/init.d/sysfsutils start (code=exited, status=0/SUCCESS)
        CPU: 4ms

Mar 11 09:24:15 SometingLight systemd[1]: Starting LSB: Set sysfs variables from /etc/sysfs.conf...
Mar 11 09:24:15 SometingLight sysfsutils[1083]: Setting sysfs variables......
Mar 11 09:24:15 SometingLight sysfsutils[1120]: cannot write value for attribute devices/pci0000:00/0000:00:02.0/sriov_numvfs ...
Mar 11 09:24:15 SometingLight sysfsutils[1121]:  failed!
Mar 11 09:24:15 SometingLight sysfsutils[1083]: done.
Mar 11 09:24:15 SometingLight systemd[1]: Started LSB: Set sysfs variables from /etc/sysfs.conf.
```

SometingLight, 2023-03-11:
I was not aware that was a motherboard-related feature ... will have to look it up, I added the line and ran the 2 refresh commands earlier and still no change.

Should the line be in grub or elsewhere?
```
root@SometingLight:~# cat /etc/default/grub
# If you change this file, run 'update-grub' afterwards to update
# /boot/grub/grub.cfg.
# For full documentation of the options in this file, see:
#   info -f grub -n 'Simple configuration'

GRUB_DEFAULT=0
GRUB_TIMEOUT=5
GRUB_DISTRIBUTOR=`lsb_release -i -s 2> /dev/null || echo Debian`
GRUB_CMDLINE_LINUX_DEFAULT="quiet intel_iommu=on iommu=pt i915.enable_guc=7 pci=realloc,assign-busses"
GRUB_CMDLINE_LINUX=""

# Uncomment to enable BadRAM filtering, modify to suit your needs
# This works with Linux (no patch required) and with any kernel that obtains
# the memory map information from GRUB (GNU Mach, kernel of FreeBSD ...)
#GRUB_BADRAM="0x01234567,0xfefefefe,0x89abcdef,0xefefefef"

# Uncomment to disable graphical terminal (grub-pc only)
#GRUB_TERMINAL=console

# The resolution used on graphical terminal
# note that you can use only modes which your graphic card supports via VBE
# you can see them in real GRUB with the command `vbeinfo'
#GRUB_GFXMODE=640x480

# Uncomment if you don't want GRUB to pass "root=UUID=xxx" parameter to Linux
#GRUB_DISABLE_LINUX_UUID=true

# Uncomment to disable generation of recovery mode menu entries
#GRUB_DISABLE_RECOVERY="true"

# Uncomment to get a beep at grub start
#GRUB_INIT_TUNE="480 440 1"
```

SometingLight, 2023-03-11:
Forgot to update-grub, and executed the usual command but no difference. Could I be putting the command int he wrong file?

zhtengw, 2023-03-12:
You can check the command line options by `dmesg | grep "Kernel command line"` after boot.
If you have done all these things, I do not have any more ideas now.

SometingLight, 2023-03-12:
Looks to be applied but still, sysfs has the same error. I'm totally at a loss here, maybe the 13500 is just limited ...

```
rootSometingLight~# dmesg | grep "Kernel command line"
[    0.056357] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.1.15-1-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt 915.enable_guc=7 pci=realloc,assign-busses
```

FWeinb, 2023-03-13:
Having the same issue with an 13500, here is a `dmesg | grep i915`
```
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.1.15-1-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=7
[    0.052028] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.1.15-1-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=7
[    2.573836] i915 0000:00:02.0: Running in SR-IOV PF mode
[    2.574496] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    2.574560] i915 0000:00:02.0: vgaarb: deactivate vga console
[    2.574593] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    2.575178] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    2.575929] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[    2.576347] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
[    2.665526] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
[    2.665529] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!
[    2.667008] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
[    2.667009] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!
[    2.667414] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_70.bin version 70.5.1
[    2.667415] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc.bin version 7.9.3
[    2.667853] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
[    2.667856] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!
[    2.670181] i915 0000:00:02.0: [drm] HuC authenticated
[    2.670429] i915 0000:00:02.0: [drm] GuC submission enabled
[    2.670430] i915 0000:00:02.0: [drm] GuC SLPC enabled
[    2.670781] i915 0000:00:02.0: [drm] GuC RC: enabled
```

and a `lspci -s 00:02 -vv`
```
00:02.0 VGA compatible controller: Intel Corporation AlderLake-S GT1 (rev 0c) (prog-if 00 [VGA controller])
        Subsystem: ASRock Incorporation Device 4680
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 141
        IOMMU group: 0
        Region 0: Memory at 6000000000 (64-bit, non-prefetchable) [size=16M]
        Region 2: Memory at 4000000000 (64-bit, prefetchable) [size=256M]
        Region 4: I/O ports at 3000 [size=64]
        Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
        Capabilities: [40] Vendor Specific Information: Len=0c <?>
        Capabilities: [70] Express (v2) Root Complex Integrated Endpoint, MSI 00
                DevCap: MaxPayload 128 bytes, PhantFunc 0
                        ExtTag- RBE+ FLReset+
                DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
                        RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop- FLReset-
                        MaxPayload 128 bytes, MaxReadReq 128 bytes
                DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
                DevCap2: Completion Timeout: Not Supported, TimeoutDis- NROPrPrP- LTR-
                         10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS-
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- OBFF Disabled,
                         AtomicOpsCtl: ReqEn-
        Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable+ 64bit-
                Address: fee00018  Data: 0000
                Masking: 00000000  Pending: 00000000
        Capabilities: [d0] Power Management version 2
                Flags: PMEClk- DSI+ D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
                Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [100 v1] Process Address Space ID (PASID)
                PASIDCap: Exec- Priv-, Max PASID Width: 14
                PASIDCtl: Enable- Exec- Priv-
        Capabilities: [200 v1] Address Translation Service (ATS)
                ATSCap: Invalidate Queue Depth: 00
                ATSCtl: Enable+, Smallest Translation Unit: 00
        Capabilities: [300 v1] Page Request Interface (PRI)
                PRICtl: Enable- Reset-
                PRISta: RF- UPRGI- Stopped+
                Page Request Capacity: 00008000, Page Request Allocation: 00000000
        Capabilities: [320 v1] Single Root I/O Virtualization (SR-IOV)
                IOVCap: Migration-, Interrupt Message Number: 000
                IOVCtl: Enable- Migration- Interrupt- MSE- ARIHierarchy-
                IOVSta: Migration-
                Initial VFs: 7, Total VFs: 7, Number of VFs: 0, Function Dependency Link: 00
                VF offset: 1, stride: 1, Device ID: 4680
                Supported Page Size: 00000553, System Page Size: 00000001
                Region 0: Memory at 0000004010000000 (64-bit, non-prefetchable)
                Region 2: Memory at 0000004020000000 (64-bit, prefetchable)
                VF Migration: offset: 00000000, BIR: 0
        Kernel driver in use: i915
        Kernel modules: i915
```

SometingLight, 2023-03-13:
> [Edit] Just got it working had a type in my sysfs.conf line, the following sysfs.conf is working for me:
> 
> ```
> devices/pci0000:00/0000:00:02.0/sriov_numvfs = 7
> ```
> 
> Outdated problem
> Having the same issue with an 13500, here is a `dmesg | grep i915`
> 
> ```
> [    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.1.15-1-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=7
> [    0.052028] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.1.15-1-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=7
> [    2.573836] i915 0000:00:02.0: Running in SR-IOV PF mode
> [    2.574496] i915 0000:00:02.0: [drm] VT-d active for gfx access
> [    2.574560] i915 0000:00:02.0: vgaarb: deactivate vga console
> [    2.574593] i915 0000:00:02.0: [drm] Using Transparent Hugepages
> [    2.575178] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
> [    2.575929] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
> [    2.576347] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
> [    2.665526] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
> [    2.665529] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!
> [    2.667008] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
> [    2.667009] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!
> [    2.667414] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_70.bin version 70.5.1
> [    2.667415] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc.bin version 7.9.3
> [    2.667853] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
> [    2.667856] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!
> [    2.670181] i915 0000:00:02.0: [drm] HuC authenticated
> [    2.670429] i915 0000:00:02.0: [drm] GuC submission enabled
> [    2.670430] i915 0000:00:02.0: [drm] GuC SLPC enabled
> [    2.670781] i915 0000:00:02.0: [drm] GuC RC: enabled
> ```
> 
> and a `lspci -s 00:02 -vv`
> 
> ```
> 00:02.0 VGA compatible controller: Intel Corporation AlderLake-S GT1 (rev 0c) (prog-if 00 [VGA controller])
>         Subsystem: ASRock Incorporation Device 4680
>         Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
>         Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
>         Latency: 0, Cache Line Size: 64 bytes
>         Interrupt: pin A routed to IRQ 141
>         IOMMU group: 0
>         Region 0: Memory at 6000000000 (64-bit, non-prefetchable) [size=16M]
>         Region 2: Memory at 4000000000 (64-bit, prefetchable) [size=256M]
>         Region 4: I/O ports at 3000 [size=64]
>         Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
>         Capabilities: [40] Vendor Specific Information: Len=0c <?>
>         Capabilities: [70] Express (v2) Root Complex Integrated Endpoint, MSI 00
>                 DevCap: MaxPayload 128 bytes, PhantFunc 0
>                         ExtTag- RBE+ FLReset+
>                 DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
>                         RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop- FLReset-
>                         MaxPayload 128 bytes, MaxReadReq 128 bytes
>                 DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
>                 DevCap2: Completion Timeout: Not Supported, TimeoutDis- NROPrPrP- LTR-
>                          10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
>                          EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
>                          FRS-
>                          AtomicOpsCap: 32bit- 64bit- 128bitCAS-
>                 DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- OBFF Disabled,
>                          AtomicOpsCtl: ReqEn-
>         Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable+ 64bit-
>                 Address: fee00018  Data: 0000
>                 Masking: 00000000  Pending: 00000000
>         Capabilities: [d0] Power Management version 2
>                 Flags: PMEClk- DSI+ D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
>                 Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
>         Capabilities: [100 v1] Process Address Space ID (PASID)
>                 PASIDCap: Exec- Priv-, Max PASID Width: 14
[... 88 lines cut ...]
                Flags: PMEClk- DSI+ D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
                Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [100 v1] Process Address Space ID (PASID)
                PASIDCap: Exec- Priv-, Max PASID Width: 14
                PASIDCtl: Enable- Exec- Priv-
        Capabilities: [200 v1] Address Translation Service (ATS)
                ATSCap: Invalidate Queue Depth: 00
                ATSCtl: Enable+, Smallest Translation Unit: 00
        Capabilities: [300 v1] Page Request Interface (PRI)
                PRICtl: Enable- Reset-
                PRISta: RF- UPRGI- Stopped+
                Page Request Capacity: 00008000, Page Request Allocation: 00000000
        Capabilities: [320 v1] Single Root I/O Virtualization (SR-IOV)
                IOVCap: Migration-, Interrupt Message Number: 000
[... cut ...]
                Initial VFs: 7, Total VFs: 7, Number of VFs: 0, Function Dependency Link: 00
                VF offset: 1, stride: 1, Device ID: 4680
                Supported Page Size: 00000553, System Page Size: 00000001
                Region 0: Memory at 0000000044000000 (64-bit, non-prefetchable)
                Region 2: Memory at 0000000000000000 (64-bit, prefetchable)
                VF Migration: offset: 00000000, BIR: 0
        Kernel driver in use: i915
        Kernel modules: i915
```

zhtengw, 2023-03-14:
@SometingLight Other idea, try increasing the memory to MAX of the interal graphics card.
I found the options from the link your shared.
[image]

SometingLight, 2023-03-14:
> @SometingLight Another idea, try increasing the memory to MAX of the interal graphics card. I found the options from the link your shared. [image]

Thanks for the suggestion, I changed the aperture as the other values were maxed. Is a refresh of some sort required or grub update as afterwards I didn't see any changes.

[image]

fanchangyong, 2023-03-16:
Same here, with 12700CPU + MicroStar B660M motherboard.
My BIOS is pretty old(2021), I'll try to update my BIOS and see if it'll help

SometingLight, 2023-03-17:
Best of luck @fanchangyong, may I just need Gigabyte to release a new BIOS but I don't have much hope.

michael-pptf, 2023-03-17:
Food for thought, on the Proxmox host, can you see `/dev/dri` has any devices under it at all?
`ls /dev/dri`

SometingLight, 2023-03-18:
@michael-pptf looks to be available:

```
root@SometingLight:~# ls -altrh /dev/dri
total 0
crw-rw----  1 root render 226, 128 Mar 14 09:55 renderD128
drwxr-xr-x  3 root root        100 Mar 14 09:55 .
drwxr-xr-x  2 root root         80 Mar 14 09:55 by-path
drwxr-xr-x 21 root root       4.9K Mar 14 09:55 ..
crw-rw----  1 root video  226,   0 Mar 17 20:56 card0
```

michael-pptf, 2023-03-18:
> @michael-pptf looks to be available:
>
> ```
> root@SometingLight:~# ls -altrh /dev/dri
> total 0
> crw-rw----  1 root render 226, 128 Mar 14 09:55 renderD128
> drwxr-xr-x  3 root root        100 Mar 14 09:55 [...]

That's so weird, because I was thinking that the driver wasn't initialized correctly.
My potato mini PC with Intel N95 chip can enable iGPU without a hassle, and it doesn't have the SRIOV option in the bios.

Sorry, I am out of ideas.

SometingLight, 2023-03-19:
@michael-pptf are you also running Proxmox?

I'm wondering, could it be that my RTX P4000 is impacting this? Although I'm passing it through to a VM?

michael-pptf, 2023-03-20:
> 

Yes, I have many successful installations on Proxmox ranging from 5.19 to 6.2.

No, I do not think other PCIe devices would affect this, that's what IOMMU is for, to isolate each device in its own group.

Back to square one then, it might be your motherboard that has an ineffecive SRIOV implementation.

SometingLight, 2023-03-20:
> Yes, I have many successful installations on Proxmox ranging from 5.19 to 6.2.
>
> No, I do not think other PCIe devices would affect this, that's what IOMMU is for, to isolate each device in its own [...]

Gigabyte has disappointed me :-(.

closed 2023-05-04 by strongtz