strongtz/i915-sriov-dkms #54: intel_gpu_top does not work

FallingSnow, 2023-03-19:
Can anyone else confirm that `intel_gpu_top` (from packages `intel-gpu-tools` in arch) is not working for them too?
```sh
$ uname -a
Linux divinity2 6.2.6-arch1-1 #1 SMP PREEMPT_DYNAMIC Mon, 13 Mar 2023 17:02:08 +0000 x86_64 GNU/Linux

$ sudo intel_gpu_top
Failed to detect engines! (No such file or directory)
(Kernel 4.16 or newer is required for i915 PMU support.)
[1]    5232 segmentation fault  sudo intel_gpu_top
```

FallingSnow, 2023-03-19:
It seems this may be coming from this driver being too old. `dmesg | grep i915` reveals a line that at the very least the driver cannot see the PMU on this processor: `i915 0000:00:02.2: [drm] PMU not supported for this GPU.`

nyanmisaka, 2023-03-23:
You can simply `strace` the `intel_gpu_top` and you will see the hardcoded `/sys/devices/i915/events` is not available.
```
execve("/usr/bin/intel_gpu_top", ["intel_gpu_top"], 0x7ffca824a8e0 /* 13 vars */) = 0
brk(NULL)                               = 0x55cc22c40000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=14374, ...}) = 0
mmap(NULL, 14374, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f9bc8bb6000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0@>\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1905632, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f9bc8bb4000
mmap(NULL, 1918592, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f9bc89df000
mmap(0x7f9bc8a01000, 1417216, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x22000) = 0x7f9bc8a01000
mmap(0x7f9bc8b5b000, 323584, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x17c000) = 0x7f9bc8b5b000
mmap(0x7f9bc8baa000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1ca000) = 0x7f9bc8baa000
mmap(0x7f9bc8bb0000, 13952, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f9bc8bb0000
close(3)                                = 0
arch_prctl(ARCH_SET_FS, 0x7f9bc8bb5580) = 0
mprotect(0x7f9bc8baa000, 16384, PROT_READ) = 0
mprotect(0x55cc20d6c000, 4096, PROT_READ) = 0
mprotect(0x7f9bc8be4000, 4096, PROT_READ) = 0
munmap(0x7f9bc8bb6000, 14374)           = 0
ioctl(1, TCGETS, {B38400 opost isig icanon echo ...}) = 0
brk(NULL)                               = 0x55cc22c40000
brk(0x55cc22c61000)                     = 0x55cc22c61000
openat(AT_FDCWD, "/sys/devices/i915/events", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = -1 ENOENT (No such file or directory)
write(2, "Failed to detect engines! (No su"..., 111Failed to detect engines! (No such file or directory)
(Kernel 4.16 or newer is required for i915 PMU support.)
) = 111
exit_group(1)                           = ?
+++ exited with 1 +++
```

FallingSnow, 2023-03-23:
It is on my system. Maybe because I'm using the 6.1 linux-intel-lts.
```sh
➜  ~ uname -a
Linux divinity2 6.1.8-1-intel-lts-sriov #1 SMP PREEMPT_DYNAMIC Sun, 19 Mar 2023 02:15:31 +0000 x86_64 GNU/Linux
➜  ~ ls /sys/devices/i915/events
actual-frequency       ccs0-busy.unit      rcs0-busy.unit               vcs0-busy       vcs1-sema.unit
actual-frequency.unit  ccs0-sema           rcs0-sema                    vcs0-busy.unit  vcs1-wait
bcs0-busy              ccs0-sema.unit      rcs0-sema.unit               vcs0-sema       vcs1-wait.unit
bcs0-busy.unit         ccs0-wait           rcs0-wait                    vcs0-sema.unit  vecs0-busy
bcs0-sema              ccs0-wait.unit      rcs0-wait.unit               vcs0-wait       vecs0-busy.unit
bcs0-sema.unit         interrupts          requested-frequency          vcs0-wait.unit  vecs0-sema
bcs0-wait              rc6-residency       requested-frequency.unit     vcs1-busy       vecs0-sema.unit
bcs0-wait.unit         rc6-residency.unit  software-gt-awake-time       vcs1-busy.unit  vecs0-wait
ccs0-busy              rcs0-busy           software-gt-awake-time.unit  vcs1-sema       vecs0-wait.unit
➜  ~ sudo intel_gpu_top
Failed to detect engines! (No such file or directory)
(Kernel 4.16 or newer is required for i915 PMU support.)
[1]    8358 segmentation fault  sudo intel_gpu_top
```

The last few lines of strace
```
newfstatat(AT_FDCWD, "/sys/devices/pci0000:00/0000:00:02.7/drm/renderD135/subsystem", {st_mode=S_IFLNK|0777, st_size=0, ...}, AT_SYMLINK_NOFOLLOW) = 0
readlinkat(AT_FDCWD, "/sys/devices/pci0000:00/0000:00:02.7/drm/renderD135/subsystem", "../../../../../class/drm", 4096) = 24
openat(AT_FDCWD, "/", O_RDONLY|O_CLOEXEC|O_PATH|O_DIRECTORY) = 3
openat(3, ".", O_RDONLY|O_CLOEXEC|O_PATH|O_DIRECTORY) = 4
openat(3, ".", O_RDONLY|O_CLOEXEC|O_PATH|O_DIRECTORY) = 5
newfstatat(5, "", {st_mode=S_IFDIR|0755, st_size=4096, ...}, AT_EMPTY_PATH) = 0
openat(5, "sys", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 6
newfstatat(6, "", {st_mode=S_IFDIR|0555, st_size=0, ...}, AT_EMPTY_PATH) = 0
close(5)                                = 0
openat(6, "devices", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 5
newfstatat(5, "", {st_mode=S_IFDIR|0755, st_size=0, ...}, AT_EMPTY_PATH) = 0
close(6)                                = 0
openat(5, "pci0000:00", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 6
newfstatat(6, "", {st_mode=S_IFDIR|0755, st_size=0, ...}, AT_EMPTY_PATH) = 0
close(5)                                = 0
openat(6, "0000:00:02.7", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 5
newfstatat(5, "", {st_mode=S_IFDIR|0755, st_size=0, ...}, AT_EMPTY_PATH) = 0
close(6)                                = 0
openat(5, "drm", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 6
newfstatat(6, "", {st_mode=S_IFDIR|0755, st_size=0, ...}, AT_EMPTY_PATH) = 0
close(5)                                = 0
close(4)                                = 0
close(3)                                = 0
faccessat2(6, "uevent", F_OK, 0)        = -1 ENOENT (No such file or directory)
close(6)                                = 0
openat(AT_FDCWD, "/", O_RDONLY|O_CLOEXEC|O_PATH|O_DIRECTORY) = 3
openat(3, ".", O_RDONLY|O_CLOEXEC|O_PATH|O_DIRECTORY) = 4
openat(3, ".", O_RDONLY|O_CLOEXEC|O_PATH|O_DIRECTORY) = 5
newfstatat(5, "", {st_mode=S_IFDIR|0755, st_size=4096, ...}, AT_EMPTY_PATH) = 0
openat(5, "sys", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 6
newfstatat(6, "", {st_mode=S_IFDIR|0555, st_size=0, ...}, AT_EMPTY_PATH) = 0
close(5)                                = 0
openat(6, "devices", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 5
newfstatat(5, "", {st_mode=S_IFDIR|0755, st_size=0, ...}, AT_EMPTY_PATH) = 0
close(6)                                = 0
openat(5, "pci0000:00", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 6
newfstatat(6, "", {st_mode=S_IFDIR|0755, st_size=0, ...}, AT_EMPTY_PATH) = 0
close(5)                                = 0
openat(6, "0000:00:02.7", O_RDONLY|O_NOFOLLOW|O_CLOEXEC|O_PATH) = 5
newfstatat(5, "", {st_mode=S_IFDIR|0755, st_size=0, ...}, AT_EMPTY_PATH) = 0
close(6)                                = 0
close(4)                                = 0
close(3)                                = 0
faccessat2(5, "uevent", F_OK, 0)        = 0
fstatfs(5, {f_type=SYSFS_MAGIC, f_bsize=4096, f_blocks=0, f_bfree=0, f_bavail=0, f_files=0, f_ffree=0, f_fsid={val=[0, 0]}, f_namelen=255, f_frsize=4096, f_flags=ST_VALID|ST_NOSUID|ST_NODEV|ST_NOEXEC|ST_RELATIME}) = 0
close(5)                                = 0
readlinkat(AT_FDCWD, "/sys/devices/pci0000:00/0000:00:02.7/subsystem", "../../../bus/pci", 4096) = 16
openat(AT_FDCWD, "/sys/devices/i915_0000_00_02.2/events", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = -1 ENOENT (No such file or directory)
write(2, "Failed to detect engines! (No su"..., 111Failed to detect engines! (No such file or directory)
(Kernel 4.16 or newer is required for i915 PMU support.)
) = 111
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x78} ---
+++ killed by SIGSEGV (core dumped) +++
[1]    8428 segmentation fault  sudo strace intel_gpu_top
```

FallingSnow, 2023-03-23:
```
➜  ~ lspci
00:00.0 Host bridge: Intel Corporation Device a704 (rev 01)
00:02.0 VGA compatible controller: Intel Corporation Raptor Lake-S GT1 [UHD Graphics 770] (rev 04)
00:02.1 VGA compatible controller: Intel Corporation Raptor Lake-S GT1 [UHD Graphics 770] (rev 04)
00:02.2 VGA compatible controller: Intel Corporation Raptor Lake-S GT1 [UHD Graphics 770] (rev 04)
00:02.3 VGA compatible controller: Intel Corporation Raptor Lake-S GT1 [UHD Graphics 770] (rev 04)
00:02.4 VGA compatible controller: Intel Corporation Raptor Lake-S GT1 [UHD Graphics 770] (rev 04)
00:02.5 VGA compatible controller: Intel Corporation Raptor Lake-S GT1 [UHD Graphics 770] (rev 04)
00:02.6 VGA compatible controller: Intel Corporation Raptor Lake-S GT1 [UHD Graphics 770] (rev 04)
00:02.7 VGA compatible controller: Intel Corporation Raptor Lake-S GT1 [UHD Graphics 770] (rev 04)
...
```

closed 2023-05-04 by strongtz

wenke313, 2024-12-29:
无解吗？我的安装了也是这样

bbaa-bbaa, 2024-12-29:
@wenke313 host上可以用`intel_gpu_top -d sys:/sys/devices/pci0000:00/0000:00:02.0`

wenke313, 2024-12-30:
@bbaa-bbaa 谢谢，这样测试可以看到，请问lxc下的docker怎么调用gpu？

hnlyxjb, 2025-01-11:
> @bbaa-bbaa 谢谢，这样测试可以看到，请问lxc下的docker怎么调用gpu？

nano /etc/pve/lxc/[CT_ID].conf 在pve下添加
lxc.cgroup2.devices.allow: c 226:0 rwm
lxc.cgroup2.devices.allow: c 226:128 rwm
lxc.mount.entry: /dev/dri/card0 dev/dri/card0 none bind,optional,create=file
lxc.mount.entry: /dev/dri/renderD128 dev/dri/renderD128 none bind,optional,create=file
lxc.apparmor.profile: unconfined
重启lxc容器，在lxc容器命令查看

Canary233, 2025-03-22:
> Can anyone else confirm that `intel_gpu_top` (from packages `intel-gpu-tools` in arch) is not working for them too?
> 
> $ uname -a
> Linux divinity2 6.2.6-arch1-1 #1 SMP PREEMPT_DYNAMIC Mon, 13 Mar 2023 17:02:08 +0000 x86_64 GNU/Linux
> 
> $ sudo intel_gpu_top
> Failed to detect engines! (No such file or directory)
> (Kernel 4.16 or newer is required for i915 PMU support.)
> [1]    5232 segmentation fault  sudo intel_gpu_top

Use`intel_gpu_top -d sriov`

johntdavis84, 2025-03-22:
> Canary233

Oh, wow. Thank you!

I’ve been using this command the entire time, and getting very odd output because I’m looking directly at the whole iGPU.
`intel_gpu_top -d sys:/sys/devices/pci0000:00/0000:00:02.0`

A lot of the time, that doesn’t show any activity, even when a VM is using its VF to tie up 40+ percent of the iGPU.

However, testing now with `intel_gpu_top -d sriov`, I’m still seeing no activity on any of the engines in `intel_gpu_top`; they’re all at 0 percent, even though can see the frequency, RC6 (?) wattage and irq/s apparently changing as the GPU works.

I recently updated both Proxmox and the guest VM, but neither of them were a kernel update. Still, I might need to reinstall the driver in both the host and the guest.