strongtz/i915-sriov-dkms #55: Strange errors during Proxmox boot, then more errors in the guest VM

pahowes, 2023-03-19:
Environment:

* ASRock Deskmini (B660 chipset)
* Intel i5-13400 CPU
* Proxmox VE 7.3
* Linux Kernel 6.2.6-1-pve

Issue:
Messages appear on the console during boot and later in `dmesg` that take the form:
```
kvmgt: disagrees about version of symbol xyz
Unknown symbol xyz
```
This is the full dump:
```
# dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.2.6-1-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on i915.enable_guc=7
[    0.081176] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.2.6-1-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on i915.enable_guc=7
[    2.772997] i915: loading out-of-tree module taints kernel.
[    2.950886] i915 0000:00:02.0: Running in SR-IOV PF mode
[    2.951536] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    2.951559] i915 0000:00:02.0: vgaarb: deactivate vga console
[    2.951596] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    2.997853] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    2.998363] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[    3.007338] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
[    3.102243] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
[    3.102246] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!
[    3.103576] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
[    3.103577] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!
[    3.103955] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_70.bin version 70.5.1
[    3.103956] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc.bin version 7.9.3
[    3.104388] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
[    3.104389] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!
[    3.106184] i915 0000:00:02.0: [drm] HuC authenticated
[    3.106636] i915 0000:00:02.0: [drm] GuC submission enabled
[    3.106637] i915 0000:00:02.0: [drm] GuC SLPC enabled
[    3.106989] i915 0000:00:02.0: [drm] GuC RC: enabled
[    3.107696] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[    3.107793] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    3.108513] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    3.134143] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    3.134242] i915 0000:00:02.0: 7 VFs could be associated with this PF
[    3.134669] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    3.134734] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    3.139745] kvmgt: disagrees about version of symbol i915_gem_object_set_to_cpu_domain
[    3.139747] kvmgt: Unknown symbol i915_gem_object_set_to_cpu_domain (err -22)
[    3.139785] kvmgt: disagrees about version of symbol i915_gem_object_alloc
[    3.139786] kvmgt: Unknown symbol i915_gem_object_alloc (err -22)
[    3.139791] kvmgt: disagrees about version of symbol i915_gem_object_create_shmem
[    3.139792] kvmgt: Unknown symbol i915_gem_object_create_shmem (err -22)
[    3.139803] kvmgt: disagrees about version of symbol i915_gem_object_pin_map
[    3.139803] kvmgt: Unknown symbol i915_gem_object_pin_map (err -22)
[    3.139810] kvmgt: disagrees about version of symbol i915_gem_gtt_insert
[    3.139810] kvmgt: Unknown symbol i915_gem_gtt_insert (err -22)
[    3.139811] kvmgt: disagrees about version of symbol i915_request_add
[    3.139812] kvmgt: Unknown symbol i915_request_add (err -22)
[    3.139820] kvmgt: disagrees about version of symbol i915_gem_ww_ctx_fini
[    3.139820] kvmgt: Unknown symbol i915_gem_ww_ctx_fini (err -22)
[    3.139850] kvmgt: disagrees about version of symbol i915_gem_ww_ctx_init
[    3.139850] kvmgt: Unknown symbol i915_gem_ww_ctx_init (err -22)
[    3.139852] kvmgt: disagrees about version of symbol i915_reserve_fence
[    3.139852] kvmgt: Unknown symbol i915_reserve_fence (err -22)
[    3.139857] kvmgt: disagrees about version of symbol i915_gem_object_ggtt_pin_ww
[    3.139857] kvmgt: Unknown symbol i915_gem_object_ggtt_pin_ww (err -22)
[    3.139879] kvmgt: disagrees about version of symbol i915_gem_object_init
[    3.139879] kvmgt: Unknown symbol i915_gem_object_init (err -22)
[    3.139914] kvmgt: disagrees about version of symbol _i915_vma_move_to_active
[    3.139914] kvmgt: Unknown symbol _i915_vma_move_to_active (err -22)
[    3.139916] kvmgt: disagrees about version of symbol i915_request_create
[... 220 lines cut ...]
[   22.017187] i915 0000:06:10.0: [drm] *ERROR* Failed to map the ggtt page table
[   22.041929] i915 0000:06:10.0: Device initialization failed (-12)
[   22.041940] i915: probe of 0000:06:10.0 failed with error -12
```

```
# lspci | grep -i vga
00:01.0 VGA compatible controller: Device 1234:1111 (rev 02)
06:10.0 VGA compatible controller: Intel Corporation Device 4682 (rev 0c)
```

Attempting to transcode video with ffmpeg results in the following error:
```
ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -I input.mkv -codec:v h264_vaapi output.mp4
...
[AVHWDeviceContext @ 0x55d88aaaa480] No VA display found for device /dev/dri/renderD128.
Device creation failed: -22.
[hevc @ 0x55d88ab2e140] No device available for decoder: device type vaapi needed for codec hevc.
Stream mapping:
  Stream #0:0 -> #0:0 (hevc (native) -> h264 (h264_vaapi))
  Stream #0:1 -> #0:1 (truehd (native) -> aac (native))
Device setup failed for decoder on input stream #0:0 : Invalid argument
```

I would appreciate any thoughts or further diagnostics to try.

zhtengw, 2023-03-19:
Do you install i915-sriov in your guest Linux?

michael-pptf, 2023-03-20:
I have the same Desk Mini B660 and I can run the VFs in both Windows and Linux guests.

Please check if the dkms driver is installed in the guest machine, then check if your user is allowed to use graphics by `ls -a /dev/dri`. If you are not `root`, chances are you are not allowed to use the graphics card. Add your user to the `render` and `video` group to see if it helps.

pahowes, 2023-03-21:
It's entirely possible that the Ubuntu kernel configuration is a bit different than Debian's. On the other hand, I can see from `modinfo i915` that the correct driver does exist on my system. Weird.

The only reason I'm not going down the LXC route right now is that my Plex server is already setting in a Debian VM. My hope is that I can get this working, then simply reproduce the appropriate changes to that VM to enable hardware transcoding.

michael-pptf, 2023-03-21:
> It's entirely possible that the Ubuntu kernel configuration is a bit different than Debian's. On the other hand, I can see from `modinfo i915` that the correct driver does exist on my system. [...]

Sorry, I am Ubuntu & Jellyfin camp... Maybe try compiling your own kernel and adding the right i915 driver from Linux mainline? Never tried it before and seemed like an awful lot of work to get a quirk working. It might be easier to export the config, reinstall Plex in an LXC (or Ubuntu VM), then importing the config

pahowes, 2023-03-21:
I'm slowly switching over to Jellyfin as well. At the moment I have a bunch of friends connected to my Plex server, some of whom are not very technical. Explaining why they can't use Jellyfin on their SmartTV when the Plex app came built-in is not a conversation I'm looking forward to!

Currently the 6.2.7 kernel is building in my Debian VM. I'll spin up a Ubuntu VM to see if I have different/better results. I tend to stick with Debian because it's usually smaller, more hardened, and has a smaller surface area. It's also amazingly simple to install. Then again I date back to installing Slackware from floppies and the 0.9x kernels, so almost anything is easier!

michael-pptf, 2023-03-21:
> I'm slowly switching over to Jellyfin as well. At the moment I have a bunch of friends connected to my Plex server, some of whom are not very technical. Explaining why they can't use Jellyfin on [...]

Oh, it hit me. In a Linux VM, in order to use compute, you have to set the emulated graphics to 'none'. Before you do that, of course, you should enable serial output in the linux commandline default. May or may not work, but I have my emulated graphics disabled in my Linux guest.

zhtengw, 2023-03-21:
hi, @pahowes. 
You cannot get VF work in Linux Guest without i915-sriov built.
In your situation, it is because that the kernel in Debian repo is configured without CONFIG_INTEL_MEI_PXP and CONFIG_DRM_I915_PXP enabled. You can reconfig the kernel with these enabled and recompiled both kernel and i915-sriov. Or you can just install the pve-kernel to your Debian guest and make i915-sriov compilable.

pahowes, 2023-03-21:
> In your situation, it is because that the kernel in Debian repo is configured without CONFIG_INTEL_MEI_PXP and CONFIG_DRM_I915_PXP enabled. You can reconfig the kernel with these enabled and [...]

That was the key!! Ubuntu has in it's kernel config:
```
CONFIG_INTEL_MEI_PXP=m
CONFIG_DRM_I915_PXP=y
```
Both settings are missing from the Debian kernel. Good find! I'm going to give this another shot, now that I'm aware of these.

michael-pptf, 2023-03-21:
> That was the key!! Ubuntu has in it's kernel config:
>
> ```
> CONFIG_INTEL_MEI_PXP=m
> CONFIG_DRM_I915_PXP=y
> ```
>
> Both settings are missing from the Debian kernel. Good find! I'm going to give this [...]

Maybe submit a PR to the README after you are done with this? Never used Debian before so I am curious to how to do it. Sorry, could not be more helpful to your problem.

pahowes, 2023-03-25:
I've had a great deal of success getting the testing (6.1.x) kernel compiled with the correct modules. My Debian VM recognizes the render device with `vainfo`. Plus, using ffmpeg with the x264_vaapi or hvec_vaapi encoder provides about a 10x speed boost and even let me transcode 4k video in better than real time in the VM.

I hope to have some time over the weekend to finalize my instructions and send a pull request with updates to the README.

pahowes, 2023-04-03:
I had to test it a bit more than I thought. Some packages were rearranged in Debian's `testing` repo which left me scratching my head a bit.

closed 2023-09-30 by strongtz