strongtz/i915-sriov-dkms #56: Hard blocking on host

nikolasgkou, 2023-03-25:
Hello there!

When my VM was running I would notice this: 
[ 5368.956440] Asynchronous wait on fence 0000:00:02.0:gnome-shell[2653]:63318 timed out (hint:intel_atomic_commit_ready [i915])
[ 5368.956749] Asynchronous wait on fence 0000:00:02.0:gnome-shell[2653]:63318 timed out (hint:intel_atomic_commit_ready [i915])

Gnome is opening all the cards by default, and when the VM powers on my host blocks very hard. I can't do any actions every 4-5 seconds, not even mouse movements.

00:00.0 Host bridge [0600]: Intel Corporation Device [8086:4601] (rev 04)
    Subsystem: Lenovo Device [17aa:50ac]
    Kernel driver in use: igen6_edac
    Kernel modules: igen6_edac
00:02.0 VGA compatible controller [0300]: Intel Corporation Alder Lake-UP3 GT2 [UHD Graphics] [8086:4628] (rev 0c)
    Subsystem: Lenovo Device [17aa:50ac]
    Kernel driver in use: i915
    Kernel modules: i915
00:02.1 VGA compatible controller [0300]: Intel Corporation Alder Lake-UP3 GT2 [UHD Graphics] [8086:4628] (rev 0c)
    Subsystem: Lenovo Device [17aa:50ac]
    Kernel driver in use: i915
    Kernel modules: i915
    
The fact i915 is bound to the VFs is causing this issue.
Would this constitute for an enhancement on i915 or is it a gnome thing?
I'm very unfamiliar with everything here. Please let me know to provide more relevant data.

michael-pptf, 2023-03-25:
Are you using the iGPU as your main display output in your host?

If so, that would not work. The host will use the physical function (PF) and it will not correctly allocate resource if you also have a virtual function (VF) on.

nikolasgkou, 2023-03-26:
I see, yes, I do use PF as the main display output in the host.
It sometimes works : 
[   69.881612] virbr0: port 1(vnet0) entered blocking state
[   69.881622] virbr0: port 1(vnet0) entered listening state
[   70.212810] mmap kvmfr0: 7f293fdff000-7f2943dff000 with size 67108864 offset 0
[   70.830243] i915 0000:00:02.0: VF1 FLR
[   70.943285] i915 0000:00:02.0: VF1 FLR
[   71.907486] virbr0: port 1(vnet0) entered learning state
[   73.956453] virbr0: port 1(vnet0) entered forwarding state
[   73.956467] virbr0: topology change detected, propagating
[   99.357189] x86/split lock detection: #AC: CPU 2/KVM/5086 took a split_lock trap at address: 0x74989d60
[  107.788446] x86/split lock detection: #AC: CPU 1/KVM/5085 took a split_lock trap at address: 0x74989d60
[  110.967847] x86/split lock detection: #AC: CPU 0/KVM/5084 took a split_lock trap at address: 0x74989d2d
[  266.247017] x86/split lock detection: #AC: CPU 3/KVM/5087 took a split_lock trap at address: 0x74989d2d

The current state of the system is nearly on 0 load. I guess it's not conflicting yet? 
Do you know if it's possible to use the VF for the host? I wonder how the display outputs would work in that case. They seem to be bound to the PF.

closed 2023-09-30 by strongtz