strongtz/i915-sriov-dkms #67: [drm] *ERROR* GT0: IOV

riverscn, 2023-04-27:
```
[   16.684514] i915 0000:00:02.7: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.0 (0000000000000000)
[   16.684528] i915 0000:00:02.7: [drm] *ERROR* GT0: IOV: Found interface version 0.1.0.0
```

zhtengw, 2023-04-27:
你好，这个错误在所有测试的内核上驱动VF时都有，目前没发现对功能的影响在哪里，还需测试观察一下

DrymarchonShaun, 2023-04-28:
seeing the same thing on `6.2.12-arch1-1`, the shell does some weird stuff right after i run `echo "7" > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs`  and if i attempt to `cat` that file, my shell hangs.

```[    1.124101] i915: loading out-of-tree module taints kernel.
[    1.125148] i915: module verification failed: signature and/or required key missing - tainting kernel
[    1.310180] i915 0000:00:02.0: Running in SR-IOV PF mode
[    1.310691] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    1.310805] i915 0000:00:02.0: vgaarb: deactivate vga console
[    1.310850] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    1.363028] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    1.371771] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc.bin (v2.19)
[    1.415956] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/adlp_guc_70.bin version 70.5.1
[    1.415960] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[    1.431852] i915 0000:00:02.0: [drm] HuC authenticated
[    1.433019] i915 0000:00:02.0: [drm] GT0: GuC submission enabled
[    1.433020] i915 0000:00:02.0: [drm] GT0: GuC SLPC enabled
[    1.433564] i915 0000:00:02.0: [drm] GuC RC: enabled
[    1.434206] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    3.112489] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    3.113285] i915 0000:00:02.0: 7 VFs could be associated with this PF
[    3.125805] fbcon: i915drmfb (fb0) is primary device
[    3.125819] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[   19.560235] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[   19.561005] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[   19.745330] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[  560.497281] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=io+mem
[  560.497348] i915 0000:00:02.1: enabling device (0000 -> 0002)
[  560.497379] i915 0000:00:02.1: Running in SR-IOV VF mode
[  560.497894] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.0 (0000000000000000)
[  560.498053] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.0.0
[  560.499413] i915 0000:00:02.1: [drm] VT-d active for gfx access
[  560.499438] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[  560.499876] RIP: 0010:wa_list_apply+0x32/0x190 [i915]
[  560.500019]  intel_gt_init_hw+0x84/0x270 [i915 1689466a3263fa96e1e0e44798b3efdfd858fe85]
[  560.500099]  intel_gt_resume+0xbe/0x2f0 [i915 1689466a3263fa96e1e0e44798b3efdfd858fe85]
[  560.500195]  intel_gt_init+0x17d/0x360 [i915 1689466a3263fa96e1e0e44798b3efdfd858fe85]
[  560.500294]  i915_gem_init+0x198/0x200 [i915 1689466a3263fa96e1e0e44798b3efdfd858fe85]
[  560.500409]  i915_driver_probe+0x720/0xdb0 [i915 1689466a3263fa96e1e0e44798b3efdfd858fe85]
[  560.500544]  i915_sriov_pf_enable_vfs+0x130/0x260 [i915 1689466a3263fa96e1e0e44798b3efdfd858fe85]
[  560.500852]  mbcache jbd2 dm_crypt cbc encrypted_keys trusted asn1_encoder tee dm_mod crct10dif_pclmul sdhci_pci crc32_pclmul cqhci crc32c_intel polyval_clmulni polyval_generic gf128mul sdhci ghash_clmulni_intel sha512_ssse3 nvme aesni_intel xhci_pci nvme_core spi_intel_pci crypto_simd mmc_core cryptd spi_intel xhci_pci_renesas nvme_common serio_raw atkbd libps2 vivaldi_fmap i8042 serio i915(OE) drm_buddy video wmi ttm intel_gtt drm_display_helper cec
[  560.500877] RIP: 0010:wa_list_apply+0x32/0x190 [i915]```

rjblake, 2023-05-01:
> 你好，这个错误在所有测试的内核上驱动VF时都有，目前没发现对功能的影响在哪里，还需测试观察一下

Got the same her, but so far all seems to be working normally. Will keep a check on things and report back if I see any issues.

zhtengw, 2023-05-02:
Since lts-v6.1.12-linux-230415T124447Z, GUC_VF_VERSION_ABI is [set to 1.1](https://github.com/intel/linux-intel-lts/commit/d34428c9acff78062ad8f2a9e781e441713c3a61#diff-32a023414b62c5da60f487281a6e542e415a0b3509dc7ebf9548fb060e12a5e4). But the version value get from register is always 1.0. I just revert the ABI version to 1.0 to prevent this error.

rjblake, 2023-05-02:
@zhtengw - what is the best way to rebuild/do a clean build? Once built, I have only been able to rebuild after removing otherwise it returns a `Module i915-sriov-dkms/6.1 already built for kernel 6.2.11-1-pve/4`

ShCby, 2023-05-02:
@rjblake Just simple `dkms remove -m i915-sriov-dkms -v 6.1` and then go to /usr/src to delete `i915-sriov-dkms-6.1`

closed 2023-05-02 by riverscn