strongtz/i915-sriov-dkms #69: pve主机启动后过一段时间自动重启

xiaowuap, 2023-04-30:
按照教程安装在pve7.4-3，Linux 6.2.9-1-pve内核的机子上，启动后十几分钟pve主机就自动重启了（还没直通sriov出来的假核显，然后dmesg看内核日志似乎没报错）
虚拟数填写的是7
i312100+prime h670plus d4
大佬牛逼！（大喊）

zhtengw, 2023-04-30:
我这里是跟你一样的CPU和pve版本，目前没遇到过由于这个核显导致的内核崩溃或重启的现象。你可以查看一下/var/log/syslog，重启前后的时间系统做了什么事情，也查看一下dmesg。如果跟i915 sriov模块没关，更建议你在pve论坛寻求帮助。

xiaowuap, 2023-04-30:
能分享一下你的grub配置和 /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs，/etc/sysfs.conf嘛大佬😂 @zhtengw

ShCby, 2023-04-30:
> 

应该就是默认+README里面的那些过程编辑之后的样子，我也没有遇到pve重启的问题

zhtengw, 2023-05-01:
> 能分享一下你的grub配置和 /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs，/etc/sysfs.conf嘛大佬😂 @zhtengw

以下供你参考：

i915-sriov Version
```
commit cc684d3172ee49455afc74fc02f8bb2aa75943ed (HEAD -> master, origin/master, origin/HEAD)
Merge: 42db9d2 012e880
Author: Sophon <wuxilin123@gmail.com>
Date:   Wed Apr 26 07:57:32 2023 +0800

    Merge pull request #65 from zhtengw/master

    Update README for new i915 params usage
```

/etc/default/grub
```
GRUB_DEFAULT=0
GRUB_TIMEOUT=5
GRUB_DISTRIBUTOR=`lsb_release -i -s 2> /dev/null || echo Debian`
GRUB_CMDLINE_LINUX_DEFAULT="quiet intel_iommu=on"
GRUB_CMDLINE_LINUX=""
```

sriov_numvfs
```
# cat /etc/sysfs.conf
devices/pci0000:00/0000:00:02.0/sriov_numvfs = 4
# cat /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
4
```

/etc/modprobe.d/
```
# cat /etc/modprobe.d/i915.conf
options i915 enable_guc=3 max_vfs=7
```

ShCby, 2023-05-01:
> 以下供你参考：
>
> i915-sriov Version
>
> ```
> commit cc684d3172ee49455afc74fc02f8bb2aa75943ed (HEAD -> master, origin/master, origin/HEAD)
> Merge: 42db9d2 012e880
> Author: Sophon <wuxilin123@gmail.com>
> Date:   Wed [...]

大佬有关注新的DKMS模块无法在虚拟机里面使用的问题吗？是DKMS模块本身的问题还是i915 driver的问题？

ShCby, 2023-05-01:
> > 大佬有关注新的DKMS模块无法在虚拟机里面使用的问题吗？是DKMS模块本身的问题还是i915 driver的问题？
> 
> 你好，有关注到很多个issue，但是我的机型上没有复现（我的虚拟机是Windows 10和Gentoo），所以暂时也没有什么想法，要找时间装其他发行版测试看看才知道。

大佬方便加一下telegram/discord讨论一下吗？

closed 2023-05-10 by strongtz