strongtz/i915-sriov-dkms #70: which kernel version of ubuntu 2204 LTS vm guest would work?

gzbenson, 2023-05-01:
Host : 8 x 12th Gen Intel(R) Core(TM) i3-12100 (1 Socket)
Kernel Version Linux 6.2.9-1-pve #1 SMP PREEMPT_DYNAMIC PVE 6.2.9-1
PVE Manager Version pve-manager/7.4-3

I upgraded the pve kernel to 6.2.9 and installed the i915-sriov-dkms on the host , 12th vGPU(sriov) VFs function is working .

I'd like to fresh install ubuntu vm guest and try VA-API video acceleration in VM ,  which kernel version of ubuntu 22.04 LTS(defaut 5.19.0-41 gereric) ) vm guest would work ?

I already tried the i915-sriov-dkms  on ubuntu 22.04 (5.19-41) & ubuntu 23.04 (6.2.0 kernel ) , it's not working 

which kernel verison should i use on ubuntu 22.04 ?

zhtengw, 2023-05-01:
For ubuntu 22.04 (5.19-41), you can try the [5.19 branch](https://github.com/zhtengw/i915-sriov-dkms/tree/5.19) in [my repo](https://github.com/zhtengw/i915-sriov-dkms).

For ubuntu 23.04 (6.2.0 kernel ), if this master branch not work, try the [6.1 branch](https://github.com/zhtengw/i915-sriov-dkms/tree/6.1).

gzbenson, 2023-05-01:
> For ubuntu 22.04 (5.19-41), you can try the [5.19 branch](https://github.com/zhtengw/i915-sriov-dkms/tree/5.19) in [my repo](https://github.com/zhtengw/i915-sriov-dkms).
>
> For ubuntu 23.04 (6.2.0 [...]

thanks ， ubuntu 23.04 ( 6.2.0) , tried the 6.1 branch , it's working .   

 ls /dev/dri
by-path  card0  renderD128
root@ubuntu2304apps:~# vainfo
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
error: can't connect to X server!
libva info: VA-API version 1.17.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_17
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.17 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 23.1.2 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSliceLP
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSliceLP
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointEncSliceLP
      VAProfileVP9Profile1            : VAEntrypointVLD
      VAProfileVP9Profile1            : VAEntrypointEncSliceLP
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointEncSliceLP
      VAProfileVP9Profile3            : VAEntrypointVLD
      VAProfileVP9Profile3            : VAEntrypointEncSliceLP
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileHEVCMain422_10         : VAEntrypointVLD
      VAProfileHEVCMain422_12         : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         : VAEntrypointVLD
      VAProfileHEVCSccMain            : VAEntrypointVLD
      VAProfileHEVCSccMain            : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain10          : VAEntrypointVLD
      VAProfileHEVCSccMain10          : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444         : VAEntrypointVLD
      VAProfileHEVCSccMain444         : VAEntrypointEncSliceLP
      VAProfileAV1Profile0            : VAEntrypointVLD
      VAProfileHEVCSccMain444_10      : VAEntrypointVLD
      VAProfileHEVCSccMain444_10      : VAEntrypointEncSliceLP

closed 2023-05-01 by gzbenson