strongtz/i915-sriov-dkms #76: linux 6.4.0-rc1+ drm/i915_mei_hdcp_interface.h: No such file or directory

paigeadelethompson, 2023-05-16:
Is there a particular branch of the linux kernel you would have this build with? Also do you have any plans of submitting the changes to the linux src tree? 

```
DKMS make.log for i915-sriov-dkms-6.1 for kernel 6.4.0-rc1+ (x86_64)
Tue 16 May 2023 04:38:20 AM PDT
make: Entering directory '/usr/src/linux'
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_driver.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_drm_client.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_config.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_getparam.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_hwmon.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_ioctl.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_irq.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_mitigations.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_module.o
In file included from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_driver.c:51:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_display_types.h:46:10: fatal error: drm/i915_mei_hdcp_interface.h: No such file or directory
   46 | #include <drm/i915_mei_hdcp_interface.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:252: /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_driver.o] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_display_trace.h:20,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_irq.c:38:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_display_types.h:46:10: fatal error: drm/i915_mei_hdcp_interface.h: No such file or directory
   46 | #include <drm/i915_mei_hdcp_interface.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:252: /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_irq.o] Error 1
make: *** [Makefile:2026: /var/lib/dkms/i915-sriov-dkms/6.1/build] Error 2
make: Leaving directory '/usr/src/linux'
```

zhtengw, 2023-05-16:
You can build this module with linux-6.1 to linux-6.3. Linux-6.4-rc is not tested now.

> Also do you have any plans of submitting the changes to the linux src tree?

This is the job what Intel is doing.

paigeadelethompson, 2023-05-17:
@zhtengw Sorry for the mixup, I realized later after I posted this that the kernel version was specified in the title of the repo I just overlooked it. 

The job that Intel is doing? I can't tell if you're kidding or if that's an actual reference; as near as I can tell Intel isn't doing anything except selling a feature for a product that they haven't actually followed through on now for two generations in a row. I really must be missing something big because as near as I can tell you're the only one who has made any progress.

paigeadelethompson, 2023-05-18:
@zhtengw @strongtz had a chance to build this and test it-- built fine on 6.4.0-rc2+ however: 

```
[   38.157095] i915: Unknown symbol intel_pxp_irq_enable (err -2)
[   38.157737] i915: Unknown symbol intel_pxp_debugfs_register (err -2)
[   38.158463] i915: Unknown symbol intel_pxp_resume_complete (err -2)
[   38.159052] i915: Unknown symbol intel_pxp_terminate (err -2)
[   38.159208] i915: Unknown symbol intel_pxp_runtime_suspend (err -2)
[   38.159542] i915: Unknown symbol intel_pxp_session_management_init (err -2)
[   38.160734] i915: Unknown symbol intel_pxp_irq_handler (err -2)
[   38.161671] i915: Unknown symbol intel_pxp_suspend (err -2)
[   38.161721] i915: Unknown symbol intel_pxp_gsccs_init (err -2)
[   38.162539] i915: Unknown symbol intel_pxp_gsccs_fini (err -2)
[   38.162599] i915: Unknown symbol intel_pxp_suspend_prepare (err -2)
[   38.162861] i915: Unknown symbol intel_pxp_irq_disable (err -2)
```

zhtengw, 2023-05-18:
> @zhtengw @strongtz had a chance to build this and test it-- built fine on 6.4.0-rc2+ however:
>
> ```
> [   38.157095] i915: Unknown symbol intel_pxp_irq_enable (err -2)
> [   38.157737] i915: Unknown [...]

You should enable CONFIG_DRM_I915_PXP in your kernel config.

paigeadelethompson, 2023-05-18:
I will try that I had a feeling it might be something like that,

were you referring to this comment by any chance (the work that Intel is doing): https://github.com/intel/linux-intel-lts/issues/33#issuecomment-1176996341

paigeadelethompson, 2023-05-18:
Looks good, fascinating that the PXP feature works on this since it's 11th gen; the release introduction on LWN indicates it's an option available on 12 gen intel CPUs:

```
[  268.155800] i915: loading out-of-tree module taints kernel.
[  268.279490] Setting dangerous option enable_guc - tainting kernel
[  268.279498] Setting dangerous option enable_fbc - tainting kernel
[  268.279499] Setting dangerous option enable_guc - tainting kernel
[  268.279500] Setting dangerous option enable_fbc - tainting kernel
[  268.280175] i915 0000:00:02.0: Running in SR-IOV PF mode
[  268.280814] i915 0000:00:02.0: [drm] VT-d active for gfx access
[  268.280999] Console: switching to colour dummy device 80x25
[  268.281039] i915 0000:00:02.0: vgaarb: deactivate vga console
[  268.281096] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[  268.283498] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[  268.284589] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_ops [i915])
[  268.297063] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/tgl_dmc_ver2_12.bin (v2.12)
[  268.318587] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.1.1.bin version 70.1.1
[  268.318596] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc_7.9.3.bin version 7.9.3
[  268.322532] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads!
[  268.322913] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[  268.322916] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[  268.323350] i915 0000:00:02.0: [drm] GuC RC: enabled
[  268.324041] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[  268.324152] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[  269.474711] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[  269.480856] ACPI: video: Video Device [GFX0] (multi-head: yes  rom: no  post: no)
[  269.482166] input: Video Bus as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/LNXVIDEO:00/input/input9
[  269.482441] i915 0000:00:02.0: 7 VFs could be associated with this PF
[  269.489544] fbcon: i915drmfb (fb0) is primary device
[  269.503619] Console: switching to colour frame buffer device 216x67
[  269.537557] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
```

paigeadelethompson, 2023-05-19:
pass through works, I actually don't know very much about this, do you know whether or not it was intended to be a headless solution or is it theoretically possible to attach a display to one of the VFs? 

When I first heard about i915/sriov I hoped that they would be able to design it so you could practically have something like a nested display for a VF inside of your primary display sort of in the same way that you can with sdl/virtio-gpu with virt-manager/viewer to get near-native video acceleration for guests. That admittedly sounds a lot more difficult but it doesn't seem unimaginable. i915 is definitely a much better solution than virtio-gpu at least in terms of watching a video on a guest and just compatibility in general (its taken years for virtio-gpu to come around on windows and I'm still not sure  if it's really there, yet.)

closed 2023-05-22 by paigeadelethompson