strongtz/i915-sriov-dkms #78: ALDER-P 13th generation i7-13700H (Fedora 38) - doesn't come back from sleep

pg2web, 2023-05-27:
DKMS-built module works but after sleep Dell Precision 3581 laptop doesn't wake up, have to press and hold power button to poweroff. Rolling back to original 6.2.15-300.fc38.x86_64 kernel i915 module returns normal sleep behavior.

CPU: 14-core (6-mt/8-st) 13th Gen Intel Core i7-13700H (-MST AMCP-)
speed/min/max: 2638/400/4800:5000:3700 MHz
Kernel: 6.2.15-300.fc38.x86_64 x86_64 Up: 15m Mem: 3644.1/31766.5 MiB (11.5%)

00:02.0 VGA compatible controller: Intel Corporation Raptor Lake-P [Iris Xe Graphics] (rev 04)

pg2web, 2023-06-29:
I opened the issue with intel-lts https://github.com/intel/linux-intel-lts/issues/44 , so we can keep it active there since it's not related to DKMS.

invenibo, 2023-07-29:
Seems intel might have fixed the problem [in this commit](https://github.com/intel/linux-intel-lts/commit/fecdf03f88d08557ec939b280c69e32ca90690c1).

pg2web, 2023-07-31:
> Seems intel might have fixed the problem [in this commit](https://github.com/intel/linux-intel-lts/commit/fecdf03f88d08557ec939b280c69e32ca90690c1).

I guess this is the same patch[ I already tested some time ago](https://github.com/strongtz/i915-sriov-dkms/issues/84#issuecomment-1599780752) - didn't fix it, at least for my CPU.

yjzzjy4, 2024-03-26:
> I guess this is the same patch[ I already tested some time ago](https://github.com/strongtz/i915-sriov-dkms/issues/84#issuecomment-1599780752) - didn't fix it, at least for my CPU.

Building linux-intel-lts at tag: `lts-v6.6.21-linux-240320T015445Z` resolves the suspend issue. Can confirm this on my 13900H CPU, hopefully it would be helpful to you.

closed 2024-12-29 by strongtz