strongtz/i915-sriov-dkms #79: 11500t,Unable to open sriov

sjkhsl, 2023-06-05:
grub file content
```
# If you change this file, run 'update-grub' afterwards to update
# /boot/grub/grub.cfg.
# For full documentation of the options in this file, see:
#   info -f grub -n 'Simple configuration'

GRUB_DEFAULT=0
GRUB_TIMEOUT=5
GRUB_DISTRIBUTOR=`lsb_release -i -s 2> /dev/null || echo Debian`
GRUB_CMDLINE_LINUX_DEFAULT="quiet intel_iommu=on iommu=pt pcie_acs_override=downstream i915.enable_guc=3 i915.max_vfs=3"
GRUB_CMDLINE_LINUX=""

# Uncomment to enable BadRAM filtering, modify to suit your needs
# This works with Linux (no patch required) and with any kernel that obtains
# the memory map information from GRUB (GNU Mach, kernel of FreeBSD ...)
#GRUB_BADRAM="0x01234567,0xfefefefe,0x89abcdef,0xefefefef"

# Uncomment to disable graphical terminal (grub-pc only)
#GRUB_TERMINAL=console

# The resolution used on graphical terminal
# note that you can use only modes which your graphic card supports via VBE
# you can see them in real GRUB with the command `vbeinfo'
#GRUB_GFXMODE=640x480

# Uncomment if you don't want GRUB to pass "root=UUID=xxx" parameter to Linux
#GRUB_DISABLE_LINUX_UUID=true

# Uncomment to disable generation of recovery mode menu entries
#GRUB_DISABLE_RECOVERY="true"

# Uncomment to get a beep at grub start
#GRUB_INIT_TUNE="480 440 1"
```

dkms status
```
root@pve:~# dkms status
i915-sriov-dkms, 6.1, 6.1.15-1-pve, x86_64: installed
```

/etc/sysfs.conf
```
root@pve:~# cat /etc/sysfs.conf
devices/pci0000:00/0000:00:02.0/sriov_numvfs = 7
```

dmesg | grep i915
```
root@pve:~# dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.1.15-1-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt pcie_acs_override=downstream i915.enable_guc=3 i915.max_vfs=3
[    0.090225] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.1.15-1-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt pcie_acs_override=downstream i915.enable_guc=3 i915.max_vfs=3
[    5.383331] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    5.383409] i915 0000:00:02.0: vgaarb: deactivate vga console
[    5.383455] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    5.383869] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    5.384484] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[    5.389877] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/rkl_dmc_ver2_03.bin (v2.3)
[    5.392898] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.1.1.bin version 70.1.1
[    5.392902] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc_7.9.3.bin version 7.9.3
[    5.396472] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads!
[    5.396772] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    5.396774] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    5.397044] i915 0000:00:02.0: [drm] GuC RC: enabled
[    5.397395] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[    5.397519] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    5.433645] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    5.434889] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    5.471234] fbcon: i915drmfb (fb0) is primary device
[    5.565707] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
```

lspci | grep VGA
```
root@pve:~# lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Device 4c8a (rev 04)
```

zhtengw, 2023-06-05:
i5-11500t is a rocket lake cpu, which not supports the SR-IOV technology.

sjkhsl, 2023-06-05:
> i5-11500t is a rocket lake cpu, which not supports the SR-IOV technology.

Thanks, I thought the 11th generation could do it!

closed 2023-09-30 by strongtz