strongtz/i915-sriov-dkms #81: Windows 11 guest recognize the sr-iov gpu. But how do I get virtual display acceleraion

odror, 2023-06-11:
CPU: Intel i9-13900k with IGPU UHD 770.
I have Arch linux with the latest SR-IOV dkms module. I am able to pass any of the intel GPU Sr-IOV devices. I am NOT getting error 43, unless I pass the entire gpu. (device 00:02.0 ). If I pass any of the SR-IOV GPUs. I am not getting that error.

So How to I get to use this device. 
I want to do 1 of the following:
1. Use it to accelerate  a virtual display
2. Be able to pass at least one or all the display ports (Do I also need to also pass one of the thunderbolts for that purpose). How Do I pass the HDMI port. (I have 3 potential ports 2 thunderbolts and 1 HDMI)

[image]
[image]

Xupack88, 2023-06-15:
Don't know if this of any help, but check under "Part V: Windows VM Usage":

[](https://www.michaelstinkerings.org/gpu-virtualization-with-intel-12th-gen-igpu-uhd-730/amp/)

odror, 2023-06-15:
I appreciate your response, but I have noticed that they are using parsec to access the vGPU. I would like to access the vGPU with one of the existing ports (I have 3 ports attached to the iGPU)

I found 
[https://github.com/intel[Display-Virtualization-for-Windows-OS](https://github.com/intel/Display-Virtualization-for Windows-OS)
](https://github.com/intel/Display-Virtualization-for-Windows-OS)

They use qemu connector to access the vGPU. It is not well documented, but I was not able to use the connectors.
I was able to install the driver.

cyril-cheng, 2023-07-03:
my CPU is i7-1165G7  11th , GPU is Iris Xe,  Its SR-IOV graphics always displays 43 in wodows10（11） vm

pg2web, 2023-07-03:
> my CPU is i7-1165G7 11th , GPU is Iris Xe, Its SR-IOV graphics always displays 43 in wodows10（11） vm

Did you check [another thread](https://github.com/strongtz/i915-sriov-dkms/issues/8#issuecomment-1567465036) - only having vendor_id I was able to get rid Code43.

resiliencer, 2023-11-12:
> my CPU is i7-1165G7 11th , GPU is Iris Xe, Its SR-IOV graphics always displays 43 in wodows10（11） vm

Did you try to extract IntelGopDriver.efi from your BIOS and pass it to romfile parameter of vfio-device in qemu/libvirt? In my case, this solved the problem with error n.43. And the video adapter now initializes properly.

closed 2024-12-29 by strongtz