strongtz/i915-sriov-dkms #83: DKMS install failes on 6.2.11-2-pve

lizznl, 2023-06-18:
After following several guides the DKMS install failes;

Kernel 6.2.11-2-pve
Error: unclear

dkms install --force -m i915-sriov-dkms -v 6.1

Creating symlink /var/lib/dkms/i915-sriov-dkms/6.1/source ->
                 /usr/src/i915-sriov-dkms-6.1

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j16 KERNELRELEASE=6.2.11-2-pve -j16 -C /lib/modules/6.2.11-2-pve/build M=/var/lib/dkms/i915-sriov-dkms/6.1/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.2.11-2-pve (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/6.1/build/make.log for more information.
root@pve03:/usr/src# cat dkms install --force -m i915-sriov-dkms -v 6.1 | grep error
cat: unrecognized option '--force'
Try 'cat --help' for more information.
root@pve03:/usr/src# cat dkms install --force -m i915-sriov-dkms -v 6.1^C
root@pve03:/usr/src# cat /var/lib/dkms/i915-sriov-dkms/6.1/build/make.log | grep error
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_scatterlist.c:161:22: error: ‘struct ttm_resource’ has no member named ‘num_pages’
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_scatterlist.c:180:28: error: ‘struct ttm_resource’ has no member named ‘num_pages’
root@pve03:/usr/src# cat /var/lib/dkms/i915-sriov-dkms/6.1/build/make.log
DKMS make.log for i915-sriov-dkms-6.1 for kernel 6.2.11-2-pve (x86_64)
Sun 18 Jun 2023 09:28:41 AM CEST
make: Entering directory '/usr/src/linux-headers-6.2.11-2-pve'
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_driver.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_drm_client.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_config.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_getparam.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_hwmon.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_ioctl.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_irq.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_mitigations.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_module.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_params.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_pci.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_scatterlist.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_suspend.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_switcheroo.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_sysfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_sysrq.o
In file included from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_context_types.h:17,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gem/i915_gem_context_types.h:20,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gem/i915_gem_context.h:10,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_module.c:9:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_utils.h:115: warning: "overflows_type" redefined
  115 | #define overflows_type(x, T) \
      |
In file included from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_utils.h:29,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_context_types.h:17,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gem/i915_gem_context_types.h:20,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gem/i915_gem_context.h:10,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_module.c:9:
./include/linux/overflow.h:157: note: this is the location of the previous definition
  157 | #define overflows_type(n, T)     \
      |
In file included from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_context_types.h:17,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gem/i915_gem_context_types.h:20,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gem/i915_gem_context.h:10,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_drm_client.c:14:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_utils.h:115: warning: "overflows_type" redefined
  115 | #define overflows_type(x, T) \
      |
In file included from ./include/linux/slab.h:16,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_drm_client.c:7:
[... 184 lines cut ...]
                 from ./include/linux/ftrace.h:13,
                 from ./include/linux/kprobes.h:28,
                 from ./include/linux/kgdb.h:19,
                 from ./include/linux/fb.h:6,
                 from ./include/linux/vga_switcheroo.h:34,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_switcheroo.c:6:
./include/linux/overflow.h:157: note: this is the location of the previous definition
  157 | #define overflows_type(n, T)     \
      |
In file included from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_runtime_pm.h:13,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_display_power.h:9,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_bw.h:12,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_driver.c:50:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_utils.h:115: warning: "overflows_type" redefined
  115 | #define overflows_type(x, T) \
      |
In file included from ./include/linux/slab.h:16,
                 from ./include/linux/resource_ext.h:11,
                 from ./include/linux/acpi.h:14,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_driver.c:30:
./include/linux/overflow.h:157: note: this is the location of the previous definition
  157 | #define overflows_type(n, T)     \
      |
make: *** [Makefile:2026: /var/lib/dkms/i915-sriov-dkms/6.1/build] Error 2
make: Leaving directory '/usr/src/linux-headers-6.2.11-2-pve'

lizznl, 2023-06-18:
When running dmesg;

dmesg | grep i915
[    3.765889] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    3.765940] i915 0000:00:02.0: vgaarb: deactivate vga console
[    3.766017] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    3.766509] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    3.768152] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[    3.771233] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc_ver2_16.bin (v2.16)
[    3.866330] i915 0000:00:02.0: [drm] GuC firmware i915/adlp_guc_70.bin version 70.5.1
[    3.866340] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc.bin version 7.9.3
[    3.880892] i915 0000:00:02.0: [drm] HuC authenticated
[    3.881546] i915 0000:00:02.0: [drm] GuC submission enabled
[    3.881547] i915 0000:00:02.0: [drm] GuC SLPC enabled
[    3.882010] i915 0000:00:02.0: [drm] GuC RC: enabled
[    3.882900] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[    3.882993] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    3.884420] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    3.885722] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    3.893049] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    3.893576] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    3.893998] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes

lizznl, 2023-06-18:
Maybe I am doing something wrong with the instructions on:
- https://forum.level1techs.com/t/i915-driver-with-sr-iov-on-13th-gen-i7-13700/197825
- https://www.michaelstinkerings.org/gpu-virtualization-with-intel-12th-gen-igpu-uhd-730/
- https://github.com/strongtz

Using latest raptor lake on Intel NUC 1340P

ksqeib, 2023-06-28:
Same issue on 6.2 and 8.0.
`DKMS make.log for i915-sriov-dkms-6.1 for kernel 6.2.16-3-pve (x86_64)
Wed Jun 28 05:31:34 PM CST 2023
make: Entering directory '/usr/src/linux-headers-6.2.16-3-pve'
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_driver.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_drm_client.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_config.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_getparam.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_hwmon.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_ioctl.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_irq.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_mitigations.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_module.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_params.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_pci.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_scatterlist.o
In file included from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_context_types.h:17,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gem/i915_gem_context_types.h:20,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gem/i915_gem_context.h:10,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_module.c:9:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_utils.h:115: warning: "overflows_type" redefined
  115 | #define overflows_type(x, T) \
      | 
In file included from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_utils.h:29:
./include/linux/overflow.h:157: note: this is the location of the previous definition
  157 | #define overflows_type(n, T)                                    \
      | 
In file included from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_context_types.h:17,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gem/i915_gem_context_types.h:20,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gem/i915_gem_context.h:10,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_drm_client.c:14:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_utils.h:115: warning: "overflows_type" redefined
  115 | #define overflows_type(x, T) \
      | 
In file included from ./include/linux/slab.h:16,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_drm_client.c:7:
./include/linux/overflow.h:157: note: this is the location of the previous definition
  157 | #define overflows_type(n, T)                                    \
      | 
In file included from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_gem.h:32,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_scatterlist.h:14,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_scatterlist.c:7:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_utils.h:115: warning: "overflows_type" redefined
  115 | #define overflows_type(x, T) \
      | 
In file included from ./include/linux/mm.h:26,
                 from ./include/linux/scatterlist.h:8,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_scatterlist.h:11:
./include/linux/overflow.h:157: note: this is the location of the previous definition
  157 | #define overflows_type(n, T)                                    \
      | 
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_scatterlist.c: In function ‘i915_rsgt_from_buddy_resource’:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_scatterlist.c:161:29: error: ‘struct ttm_resource’ has no member named ‘num_pages’
  161 |         const u64 size = res->num_pages << PAGE_SHIFT;
      |                             ^~
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_scatterlist.c:180:35: error: ‘struct ttm_resource’ has no member named ‘num_pages’
  180 |         if (sg_alloc_table(st, res->num_pages, GFP_KERNEL)) {
      |                                   ^~
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_scatterlist.o] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_runtime_pm.h:13,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_display_power.h:9,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_drv.h:42,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_config.c:6:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_utils.h:115: warning: "overflows_type" redefined
  115 | #define overflows_type(x, T) \
      | 
In file included from ./include/linux/device.h:29,
                 from ./include/linux/pm_qos.h:17,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_drv.h:35:
[... 86 lines cut ...]
[... cut ...]
                 from ./include/linux/resource_ext.h:11,
                 from ./include/linux/acpi.h:14,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_driver.c:30:
./include/linux/overflow.h:157: note: this is the location of the previous definition
  157 | #define overflows_type(n, T)                                    \
      | 
make: *** [Makefile:2026: /var/lib/dkms/i915-sriov-dkms/6.1/build] Error 2
make: Leaving directory '/usr/src/linux-headers-6.2.16-3-pve'`

mike6041, 2023-07-07:
I was having the same issue, adding --force to the dkms install command fixed it for me.

bombonato, 2023-07-14:
Thanks @mike6041, the "--force" option works for me. Working on PVE 8.0.3 and kernel 6.2.16-4.

By the way, I found a detailed post about this (https://www.derekseaman.com/2023/06/proxmox-ve-8-windows-11-vgpu-vt-d-passthrough-with-intel-alder-lake.html), only suggest executing a "update-initramfs -u -k all" (i had some issues when upgrading kernel, best to update all installed kernels)

closed 2023-09-30 by strongtz