strongtz/i915-sriov-dkms #84: Compiling against gentoo 6.1.28 kernel fails

adapt-L, 2023-06-20:
I'm not using DKMS, but compiling against my 6.1.28 kernel with `make -C /usr/src/linux M=${PWD}` fails with:

```
i915-sriov-dkms/drivers/gpu/drm/i915/display/intel_opregion.c: In function 'intel_opregion_get_edid':
i915-sriov-dkms/drivers/gpu/drm/i915/display/intel_opregion.c:1129:32: error: passing argument 1 of 'drm_edid_is_valid' discards 'const' qualifier from pointer target type [-Werror=discarded-qualifiers]
 1129 |         if (!drm_edid_is_valid(drm_edid_raw(drm_edid))) {
      |                                ^~~~~~~~~~~~~~~~~~~~~~
In file included from i915-sriov-dkms/drivers/gpu/drm/i915/display/intel_opregion.c:34:
./include/drm/drm_edid.h:591:37: note: expected 'struct edid *' but argument is of type 'const struct edid *'
  591 | bool drm_edid_is_valid(struct edid *edid);
      |                        ~~~~~~~~~~~~~^~~~
cc1: all warnings being treated as errors
```

I suppose that means there's a conflicting definition in my kernel source's include/drm/drm_edid.h?

Also, does this module implement the suspend/resume patch here: https://github.com/intel/linux-intel-lts/commit/d8672fa7af8e0d6919a3efd3b699d44f14ec92c9 ? If not maybe I can help move that patch over.

adapt-L, 2023-06-20:
I figure it's worth it to support this specific kernel version because it's the latest stable one on intel-lts and gentoo-sources (https://packages.gentoo.org/packages/sys-kernel/gentoo-sources)

pg2web, 2023-06-21:
> Also, does this module implement the suspend/resume patch here: [intel/linux-intel-lts@d8672fa](https://github.com/intel/linux-intel-lts/commit/d8672fa7af8e0d6919a3efd3b699d44f14ec92c9) ? If not maybe I can help move that patch over.

Hmm... I've just checked and it looks like DKMS repo doesn't have this patch (at least drivers/gpu/drm/i915/gt/intel_gt_pm.c  has nothing about **#include "iov/intel_iov.h"**). I'll try to apply this patch and see if it will fix my suspend freeze, nothing I tried with i915 settings helped).

pg2web, 2023-06-21:
update: applied mentioned [intel/linux-intel-lts@d8672fa](https://github.com/intel/linux-intel-lts/commit/d8672fa7af8e0d6919a3efd3b699d44f14ec92c9)  patch to DKMS module for my lates Fedora38 6.3.8 kernel - unfortunately didn't help, still needs poweroff after sleep mode, even if SR-IOV feature is not initialized... may be it's just my Precision 3581 with 13th gen CPU and Iris XE.

zhtengw, 2023-06-21:
> I'm not using DKMS, but compiling against my 6.1.28 kernel with `make -C /usr/src/linux M=${PWD}` fails with:
>
> ```
> i915-sriov-dkms/drivers/gpu/drm/i915/display/intel_opregion.c: In function [...]

You have enable `-Werror=discarded-qualifiers` CFLAGS somewher> I'm not using DKMS, but compiling against my 6.1.28 kernel with `make -C /usr/src/linux M=${PWD}` fails with:
> ```
> i915-sriov-dkms/drivers/gpu/drm/i915/display/intel_opregion.c: In function 'intel_opregion_get_edid':
> i915-sriov-dkms/drivers/gpu/drm/i915/display/intel_opregion.c:1129:32: error: passing [...]

You have enabled `-Werror=discarded-qualifiers` CFLAGS somewhere in your system. 
Please try with `make EXTRA_CFLAGS+=-Wdiscarded-qualifiers -C /usr/src/linux M=${PWD}`

zhtengw, 2023-06-21:
> Also, does this module implement the suspend/resume patch here: https://github.com/intel/linux-intel-lts/commit/d8672fa7af8e0d6919a3efd3b699d44f14ec92c9 ? If not maybe I can help move that patch over.

OK, I will synchronize the upstream patches in a few days.

I may also add `EXTRA_CFLAGS+=-Wdiscarded-qualifiers` into Makefile in next update.

closed 2023-09-30 by strongtz