strongtz/i915-sriov-dkms #85: DKMS install failes on 6.2.16-2-pve

skill7899, 2023-06-23:
Fri Jun 23 05:42:54 PM CST 2023
make: Entering directory '/usr/src/linux-headers-6.2.16-3-pve'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Debian 12.2.0-14) 12.2.0
  You are using:           gcc (Debian 10.2.1-6) 10.2.1 20210110
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_driver.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_drm_client.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_config.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_getparam.o
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_hwmon.o
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_ioctl.o
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_irq.o
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_mitigations.o
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_module.o
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_params.o
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_pci.o
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_scatterlist.o
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_suspend.o
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_switcheroo.o
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_sysfs.o
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_utils.o
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_device_info.o
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
gcc: error: unrecognized command-line option ‘-fzero-call-used-regs=used-gpr’
  CC [M]  /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_memory_region.o
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_driver.o] Error 1
make[1]: *** Waiting for unfinished jobs....
gcc: error: unrecognized command-line option ‘-fzero-call-used-regs=used-gpr’
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_drm_client.o] Error 1
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
gcc: error: unrecognized command-line option ‘-fzero-call-used-regs=used-gpr’
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_getparam.o] Error 1
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
gcc: error: unrecognized command-line option ‘-fzero-call-used-regs=used-gpr’
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_config.o] Error 1
gcc: error: unrecognized command-line option ‘-fzero-call-used-regs=used-gpr’
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_ioctl.o] Error 1
gcc: error: unrecognized command-line option ‘-fzero-call-used-regs=used-gpr’
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_irq.o] Error 1
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
gcc: error: unrecognized command-line option ‘-fzero-call-used-regs=used-gpr’
gcc: error: unrecognized command-line option ‘-fzero-call-used-regs=used-gpr’
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_hwmon.o] Error 1
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_mitigations.o] Error 1
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
gcc: error: unrecognized command-line option ‘-fzero-call-used-regs=used-gpr’
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_module.o] Error 1
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
gcc: error: unrecognized command-line option ‘-fzero-call-used-regs=used-gpr’
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_params.o] Error 1
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
gcc: error: unrecognized command-line option ‘-fzero-call-used-regs=used-gpr’
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_scatterlist.o] Error 1
gcc: error: unrecognized command-line option ‘-fzero-call-used-regs=used-gpr’
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_pci.o] Error 1
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
gcc: error: unrecognized command-line option ‘-fzero-call-used-regs=used-gpr’
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_suspend.o] Error 1
gcc: error: unrecognized command-line option ‘-fzero-call-used-regs=used-gpr’
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_sysfs.o] Error 1
gcc: error: unrecognized command-line option ‘-fzero-call-used-regs=used-gpr’
gcc: error: unrecognized command-line option ‘-fzero-call-used-regs=used-gpr’
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_utils.o] Error 1
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_device_info.o] Error 1
gcc: error: unrecognized command-line option ‘-fzero-call-used-regs=used-gpr’
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/i915_switcheroo.o] Error 1
gcc: error: unrecognized command-line option ‘-fzero-call-used-regs=used-gpr’
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov/dkms/build/drivers/gpu/drm/i915/intel_memory_region.o] Error 1
make: *** [Makefile:2026: /var/lib/dkms/i915-sriov/dkms/build] Error 2
make: Leaving directory '/usr/src/linux-headers-6.2.16-3-pve'

love768952693, 2023-06-24:
The 6.2.16 kernel cannot be successfully compiled, I just tested it.
We can only wait for the fix. PVE8 defaults to this version of the kernel. The compromise is to downgrade the kernel, or wait for the author to fix it.

unknown-0054, 2023-06-24:
[image]
i successed

love768952693, 2023-06-24:
> 

哥们，可以把你PVE8上的源和企业源贴出来不，我感觉我源有问题。

`root@n305:/usr/src/i915-sriov-dkms-6.2# dkms status
i915-sriov-dkms/6.2, 6.2.16-3-pve, x86_64: installed (WARNING! Diff between built and installed module!)
root@n305:/usr/src/i915-sriov-dkms-6.2# 
`

love768952693, 2023-06-24:
> [image] i successed

哥们可以把你的 源和企业源 贴出来不，我感觉我源有问题。

`root@n305:/usr/src/i915-sriov-dkms-6.2# dkms status
i915-sriov-dkms/6.2, 6.2.16-3-pve, x86_64: installed (WARNING! Diff between built and installed module!)
root@n305:/usr/src/i915-sriov-dkms-6.2# 
`

love768952693, 2023-06-24:
> [image] i successed

我也搞定了，哈哈

bootli, 2023-06-25:
你咋搞定的

unknown-0054, 2023-06-25:
> 你咋搞定的

加上 --force 因为 6.2内核已经自带12/13代的核显驱动了

zx900930, 2023-06-25:
[image]

You may need the `--force` flag to install dkms on kernal 6.2.16.3-pve
`dkms install -m i915-sriov-dkms -v 6.1 -k 6.2.16-3-pve --force`

bootli, 2023-06-25:
Sign command: /lib/modules/6.2.16-3-pve/build/scripts/sign-file
Binary /lib/modules/6.2.16-3-pve/build/scripts/sign-file not found, modules won't be signed
Error! Your kernel headers for kernel 6.2.16-3-pve cannot be found at /lib/modules/6.2.16-3-pve/build or /lib/modules/6.2.16-3-pve/source.
Please install the linux-headers-6.2.16-3-pve package or use the --kernelsourcedir option to tell DKMS where it's located.

skill7899, 2023-06-26:
> [image]
> 
> You may need the `--force` flag to install dkms on kernal 6.2.16.3-pve `dkms install -m i915-sriov-dkms -v 6.1 -k 6.2.16-3-pve --force`

thanks , i know why，i use dkms install it in pve 7.4.x，the gcc version is not match

skill7899, 2023-06-26:
> 十三代I5 13600 提示这个 Sign command: /lib/modules/6.2.16-3-pve/build/scripts/sign-file Binary /lib/modules/6.2.16-3-pve/build/scripts/sign-file not found, modules won't be signed Error! Your kernel headers for kernel 6.2.16-3-pve cannot be found at /lib/modules/6.2.16-3-pve/build or /lib/modules/6.2.16-3-pve/source. Please install the linux-headers-6.2.16-3-pve package or use the --kernelsourcedir option to tell DKMS where it's located.

apt install pve-header-6.2.16-3-pve

closed 2023-06-27 by skill7899

wubushanyan, 2023-10-31:
> > 十三代I5 13600 提示这个 Sign command: /lib/modules/6.2.16-3-pve/build/scripts/sign-file Binary /lib/modules/6.2.16-3-pve/build/scripts/sign-file not found, modules won't be signed Error! Your kernel headers for kernel 6.2.16-3-pve cannot be found at /lib/modules/6.2.16-3-pve/build or /lib/modules/6.2.16-3-pve/source. Please install the linux-headers-6.2.16-3-pve package or use the --kernelsourcedir option to tell DKMS where it's located.
> 
> apt install pve-headers-6.2.16-3-pve

cant find pve-headers-6.2.16-3-pve how to fix it , which source u use
```
root@bingo:/usr/src/i915-sriov-dkms-6.2# apt install pve-headers-6.2.16-3-pve
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package pve-headers-6.2.16-3-pve
E: Couldn't find any package by glob 'pve-headers-6.2.16-3-pve'
```