strongtz/i915-sriov-dkms #86: I installed pve8.0, but failed to build it on the 6.2.16-3-pve kernel.

Alienswj, 2023-06-23:
The console output is as follows:

```
root@pve:/usr/src/i915-sriov-dkms-6.2# dkms install -m i915-sriov-dkms -v 6.2   
Sign command: /lib/modules/6.2.16-3-pve/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
/var/lib/dkms/i915-sriov-dkms/6.2/source/dkms.conf: line 3: $'\r': command not found
/var/lib/dkms/i915-sriov-dkms/6.2/source/dkms.conf: line 6: $'\r': command not found
/var/lib/dkms/i915-sriov-dkms/6.2/source/dkms.conf: line 9: $'\r': command not found

Building module:
Cleaning build area...(bad exit status: 2)
make -j4 KERNELRELEASE=6.2.16-3-pve -j4 -C /lib/modules/6.2.16-3-pve/build M=/va...(bad exit status: 2)ems
Error! Bad return status for module build on kernel: 6.2.16-3-pve (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/6.2/build/make.log for more information.
```

Then I opened the log file pointed to, and the content is as follows：
```
# cat /var/lib/dkms/i915-sriov-dkms/6.2/build/make.log
DKMS make.log for i915-sriov-dkms-6.2 for kernel 6.2.16-3-pve (x86_64)
Fri Jun 23 06:38:48 PM CST 2023
make: Entering directory '/usr/src/linux-headers-6.2.16-3-pve'
Makefile:149: *** building multiple external modules is not supported.  Stop.
make: Leaving directory '/usr/src/linux-headers-6.2.16-3-pve'
```

Alienswj, 2023-06-24:
😂This is my mistake, because Windows and Linux do not match the symbols for line breaks, so replace Windows '/r/n'  with Linux '/n'，it works well

closed 2023-06-24 by Alienswj

love768952693, 2023-06-24:
> 😂This is my mistake, because Windows and Linux do not match the symbols for line breaks, so replace Windows '/r/n' with Linux '/n'，it works well

The 6.2.16 kernel cannot be successfully compiled, I just tested it.

Alienswj, 2023-06-24:
> > 😂This is my mistake, because Windows and Linux do not match the symbols for line breaks, so replace Windows '/r/n' with Linux '/n'，it works well
> 
> The 6.2.16 kernel cannot be successfully compiled, I just tested it.

都是中国人就说中文吧，我这边试了是6.2.16-3可以dkms编译成功，而且试了直通给win server2022，环境是n100芯片，pve8.0，你可以把步骤贴出来，我可以帮忙看看

love768952693, 2023-06-24:
> 都是中国人就说中文吧，我这边试了是6.2.16-3可以dkms编译成功，而且试了直通给win server2022，环境是n100芯片，pve8.0，你可以把步骤贴出来，我可以帮忙看看

我去，这你都发现了，我感觉是PVE8源的问题。

`i915.ko:
Running module version sanity check.
Module version  for i915.ko
exactly matches what is already found in kernel 6.2.16-3-pve.
DKMS will not replace this module.
You may override by specifying --force.
depmod...
root@n305:/usr/src/i915-sriov-dkms-6.2# dkms status
i915-sriov-dkms/6.2, 6.2.16-3-pve, x86_64: installed (WARNING! Diff between built and installed module!)
root@n305:/usr/src/i915-sriov-dkms-6.2# 
`

love768952693, 2023-06-24:
> 都是中国人就说中文吧，我这边试了是6.2.16-3可以dkms编译成功，而且试了直通给win server2022，环境是n100芯片，pve8.0，你可以把步骤贴出来，我可以帮忙看看

我也搞定了。哈哈

Alienswj, 2023-06-25:
> 我也搞定了。哈哈

应该是dkms install 忘了加 --force了吧

bootli, 2023-06-25:
> > > > 😂这是我的错误，因为 Windows 和 Linux 不匹配换行符号，所以将 Windows '/r/n' 替换为 Linux '/n'，效果很好
> > > 
> > > 
> > > 6.2.16内核无法编译成功，我刚刚测试了一下。
> > 
> > 
> > 都是中国人就说中文吧，我金字塔试了是6.2.16-3可以dkms编译成功，并且试了直通给win server2022，环境是n100芯片，pve8.0，你可以把步骤贴出来，我可以帮忙看看
> 
> 我也搞定了。哈哈

请问你是怎么弄好的

bootli, 2023-06-25:
> 都是中国人就说中文吧，我这边试了是6.2.16-3可以dkms编译成功，而且试了直通给win server2022，环境是n100芯片，pve8.0，你可以把步骤贴出来，我可以帮忙看看

你好 你是怎么编译成功的 我的提示Sign command: /lib/modules/6.2.16-3-pve/build/scripts/sign-file
Binary /lib/modules/6.2.16-3-pve/build/scripts/sign-file not found, modules won't be signed
Error! Your kernel headers for kernel 6.2.16-3-pve cannot be found at /lib/modules/6.2.16-3-pve/build or /lib/modules/6.2.16-3-pve/source.
Please install the linux-headers-6.2.16-3-pve package or use the --kernelsourcedir option to tell DKMS where it's located.

Alienswj, 2023-06-25:
> 你好 你是怎么编译成功的 我的提示Sign command: /lib/modules/6.2.16-3-pve/build/scripts/sign-file Binary /lib/modules/6.2.16-3-pve/build/scripts/sign-file not found, modules won't be signed Error! Your kernel [...]

你安装的就是pve8.0哇？还是后来升级上去的？pve首页看看确定是6.2.16-3-pve内核吗？报错是dkms没找到内核

bootli, 2023-06-25:
> 你安装的就是pve8.0哇？还是后来升级上去的？pve首页看看确定是6.2.16-3-pve内核吗？报错是dkms没找到内核

升级上去 8.0也这个报错 重新pve8.0安装也是这样报错...付了  不知道你是咋安装的

bootli, 2023-06-25:
> 你安装的就是pve8.0哇？还是后来升级上去的？pve首页看看确定是6.2.16-3-pve内核吗？报错是dkms没找到内核

你好你是怎么安装成功的 能发代码来吗 谢谢了

bootli, 2023-06-25:
> 你好你是怎么安装成功的 能发代码来吗 谢谢了

我知道了我是十三代cpu不行 我用另一台9代CPU可以的

wangbaobao11, 2023-06-26:
Sign command: /lib/modules/6.2.16-3-pve/build/scripts/sign-file
Binary /lib/modules/6.2.16-3-pve/build/scripts/sign-file not found, modules won't be signed
Creating symlink /var/lib/dkms/i915-sriov-dkms/6.2/source -> /usr/src/i915-sriov-dkms-6.2
Error! Your kernel headers for kernel 6.2.16-3-pve cannot be found at /lib/modules/6.2.16-3-pve/build or /lib/modules/6.2.16-3-pve/source.
Please install the linux-headers-6.2.16-3-pve package or use the --kernelsourcedir option to tell DKMS where it's located.
大佬们怎么解决啊

wangbaobao11, 2023-06-26:
> > > > 😂这是我的错误，因为 Windows 和 Linux 不匹配换行符号，所以将 Windows '/r/n' 替换为 Linux '/n'，效果很好
> > > 
> > > 
> > > 6.2.16内核无法编译成功，我刚刚测试了一下。
> > 
> > 
> > 都是中国人就说中文吧，我金字塔试了是6.2.16-3可以dkms编译成功，并且试了直通给win server2022，环境是n100芯片，pve8.0，你可以把步骤贴出来，我可以帮忙看看
> 
> 你好你是怎么编译成功的 我的提示Sign command: /lib/modules/6.2.16-3-pve/build/scripts/sign-file Binary /lib/modules/6.2.16-3-pve/build/找不到脚本/签名文件，模块不会被签名 错误！在 /lib/modules/6.2.16-3-pve/build 或 /lib/modules/6.2.16-3-pve/source 中找不到内核 6.2.16-3-pve 的内核头。 请安装 linux-headers-6.2.16-3-pve 软件包或使用 --kernelsourcedir 选项告诉 DKMS 它的位置。

大佬 你解决了吗？？

bootli, 2023-06-26:
> 大佬 你解决了吗？？

你也是这个问题?

wangbaobao11, 2023-06-26:
> 

是啊  我是 7.4升级上来的

wangbaobao11, 2023-06-26:
> 你也是这个问题?

是 啊  我是7.4升级上来的

bootli, 2023-06-26:
> 是 啊 我是7.4升级上来的

十三代CPU？

wangbaobao11, 2023-06-26:
> 十三代CPU？

8505

bootli, 2023-06-26:
> 

正式版?

bootli, 2023-06-26:
> 8505

应该不是正版  我的CPU不是正式版

wangbaobao11, 2023-06-27:
> > > > > 😂这是我的错误，因为 Windows 和 Linux 不匹配换行符号，所以将 Windows '/r/n' 替换为 Linux '/n'，效果很好
> > > > 
> > > > 
> > > > 6.2.16内核无法编译成功，我刚刚测试了一下。
> > > 
> > > 
> > > 都是中国人就说中文吧，我金字塔试了是6.2.16-3可以dkms编译成功，并且试了直通给win server2022，环境是n100芯片，pve8.0，你可以把步骤贴出来，我可以帮忙看看
> > 
> > 
> > 我也搞定了。哈哈
> 
> 应该是 dkms install 忘了加 --force 吧

i915.ko:
Running module version sanity check.
Module version  for i915.ko
exactly matches what is already found in kernel 6.2.16-3-pve.
DKMS will not replace this module.
You may override by specifying --force.
depmod...

大佬 这个--force. 怎么加啊？？   dkms install -m i915-sriov-dkms -v 6.2

wangbaobao11, 2023-06-27:
> 应该不是正版 我的CPU不是正式版

apt install pve-headers-6.2.16-3-pve

Alienswj, 2023-06-27:
> i915.ko:
> Running module version sanity check.
> Module version  for i915.ko
> exactly matches what is already found in kernel 6.2.16-3-pve.
> DKMS will not replace this module.
> You may override by [...]

dkms install --force -m i915-sriov-dkms -v 6.2

wangbaobao11, 2023-06-27:
> dkms install --force -m i915-sriov-dkms -v 6.2

谢大佬！！

wubushanyan, 2023-10-31:
大佬我n100 安装的 pve8 也无法编译
添加了force参数仍然无法安装，尝试安装pve-headers-6.2.16-3-pve 但是无法安装 说是找不到这个source
```
root@bingo:/usr/src/i915-sriov-dkms-6.2# dkms install --force -m i915-sriov-dkms -v 6.2
Error! Your kernel headers for kernel 6.2.16-3-pve cannot be found.
Please install the linux-headers-6.2.16-3-pve package,
or use the --kernelsourcedir option to tell DKMS where it's located
root@bingo:/usr/src/i915-sriov-dkms-6.2# 
root@bingo:/usr/src/i915-sriov-dkms-6.2# apt install pve-headers-6.2.16-3-pve
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package pve-headers-6.2.16-3-pve
E: Couldn't find any package by glob 'pve-headers-6.2.16-3-pve'
```

YGKKmi, 2024-01-03:
Building module:
Cleaning build area...
make -j8 KERNELRELEASE=6.2.16-3-pve -j8 -C /lib/modules/6.2.16-3-pve/build M=/var/lib/dkms/i915-sriov-dkms/6.2/build KVER=6.2.16-3-pve...................(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.2.16-3-pve (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/6.2/build/make.log for more information.

qiujiu!

lvxj11, 2024-04-10:
> Sign command: /lib/modules/6.2.16-3-pve/build/scripts/sign-file Binary /lib/modules/6.2.16-3-pve/build/scripts/sign-file not found, modules won't be signed Creating symlink [...]

虽然有点晚，但刚解决了这个问题就来回答一下。
这是因为更新了内核比如说从6.2.16-3升级到6.2.16-5。但因为没有重启，系统使用的还是旧内核但头文件安装的是最新版。
解决方法是重启pve让它使用最新版的内核，并且重来一遍。

zhaoliangwan, 2024-04-11:
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_dp_mst_find_vcpi_slots_for_bpp’:
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:86:35: error: too many arguments to function ‘drm_dp_calc_pbn_mode’
   86 |                 crtc_state->pbn = drm_dp_calc_pbn_mode(adjusted_mode->crtc_clock,
      |                                   ^~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_display_types.h:36,
                 from /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:40:
./include/drm/display/drm_dp_mst_helper.h:835:5: note: declared here
  835 | int drm_dp_calc_pbn_mode(int clock, int bpp);
      |     ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_dp_mst_mode_valid_ctx’:
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:906:17: error: too many arguments to function ‘drm_dp_calc_pbn_mode’
  906 |                 drm_dp_calc_pbn_mode(mode->clock, min_bpp, false) > port->full_pbn) {
      |                 ^~~~~~~~~~~~~~~~~~~~
./include/drm/display/drm_dp_mst_helper.h:835:5: note: declared here
  835 | int drm_dp_calc_pbn_mode(int clock, int bpp);
      |     ^~~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:251: /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/src/linux-headers-6.5.13-5-pve/Makefile:2039: /var/lib/dkms/i915-sriov-dkms/6.5/build] Error 2
make: *** [Makefile:234: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.5.13-5-pve'

请问大佬们这种报错该怎么解决呢，太多参数放到这个drm_dp_calc_pbn_mode函数里了？

ohmyharry, 2024-09-02:
`root@pve:/usr/src/i915-sriov-dkms-6.2.16-3# cat /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/make.log
DKMS make.log for i915-sriov-dkms-6.2.16-3 for kernel 6.2.16-3-pve (x86_64)
Tue Sep  3 01:28:22 AM CST 2024
make: Entering directory '/usr/src/linux-headers-6.2.16-3-pve'
KERNELRELEASE=6.2.16-3-pve
KERNELVERSION=6.2.16
EXTRAVERSION_MAJOR=3
EXTRAVERSION_MINOR=0
EXTRAVERSION_NAME=pve
EXTRAVERSION_DEFINE=EXTRAVERSION_PVE
LSBRELEASE=12
LSBRELEASE_MAJOR=12
LSBRELEASE_MINOR=0
LSBRELEASE_NAME=Debian
LSBRELEASE_DEFINE=RELEASE_DEBIAN
GUCFIRMWARE_MAJOR=1
GUCFIRMWARE_MINOR=9
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_driver.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_drm_client.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_config.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_getparam.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_hwmon.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_ioctl.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_irq.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_mitigations.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_module.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_params.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_pci.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_scatterlist.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_suspend.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_switcheroo.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_sysfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_utils.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/intel_device_info.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/intel_memory_region.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/intel_pcode.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/intel_pm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/intel_region_ttm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/intel_runtime_pm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/intel_sbi.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/intel_step.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/intel_uncore.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/intel_wakeref.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/vlv_sideband.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/vlv_suspend.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/soc/intel_dram.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/soc/intel_gmch.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/soc/intel_pch.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_memcpy.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_mm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_sw_fence.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_sw_fence_work.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_syncmap.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_user_extensions.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_ioc32.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_debugfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_debugfs_params.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/display/intel_display_debugfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/display/intel_pipe_crc.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/i915_pmu.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/gt/gen2_engine_cs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/gt/gen6_engine_cs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/gt/gen6_ppgtt.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/gt/gen7_renderclear.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/gt/gen8_engine_cs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/gt/gen8_ppgtt.o
/var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/gt/gen8_ppgtt.c:103:12: warning: ‘gen12_pte_encode_shim’ defined but not used [-Wunused-function]
  103 | static u64 gen12_pte_encode_shim(dma_addr_t addr,
      |            ^~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/gt/intel_breadcrumbs.o
[... 201 lines cut ...]
[... cut ...]
  572 | int drm_add_edid_modes(struct drm_connector *connector, struct edid *edid);
      |                                                         ~~~~~~~~~~~~~^~~~
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build/drivers/gpu/drm/i915/display/intel_dp.o] Error 1
make: *** [Makefile:2026: /var/lib/dkms/i915-sriov-dkms/6.2.16-3/build] Error 2
make: Leaving directory '/usr/src/linux-headers-6.2.16-3-pve'
`

报错日志..不知道如何解决