strongtz/i915-sriov-dkms #87: Plex Ubuntu 22.04 LXC HDR tone mapping does not work with Proxmox 8

Coder84619, 2023-06-27:
I have a Proxmox 8 i5-1240P host, which I've configured for 7 VFs. I have a Windows 11 Pro VM using a VF and it works perfectly fine and is rock solid. However, I have a Plex LXC (built with tteck's Plex script), where HDR tone mapping produces constant errors in the Plex logs and the video stream is corrupted. However, HW transcoding works perfectly fine. 

If in the LXC config I use the following config then HDR tone mapping (and HW transcoding) works:

`lxc.cgroup2.devices.allow: c 226:0 rwm
lxc.cgroup2.devices.allow: c 226:128 rwm
lxc.mount.entry: /dev/dri/card0 dev/dri/card0 none bind,optional,create=file
lxc.mount.entry: /dev/dri/renderD128 dev/dri/renderD128 none bind,optional,create=file`

However, if I use any of the VFs, then HDR tone mapping fails but HW transcoding still works:

`lxc.cgroup2.devices.allow: c 226:4 rwm
lxc.cgroup2.devices.allow: c 226:132 rwm
lxc.mount.entry: /dev/dri/card4 dev/dri/card0 none bind,optional,create=file
lxc.mount.entry: /dev/dri/renderD132 dev/dri/renderD128 none bind,optional,create=file`

Inside the Plex LXC vainfo:

`vainfo: VA-API version: 1.17 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 23.1.1 ()
vainfo: Supported profile and entrypoints`

According to Chuck (Plex employee) on the PMS forums, HW transcoding uses Iris XE instructions on the CPU, while HDR tone mapping uses the GPU directly. They are different APIs. So that could explain why one works but not the other.

Coder84619, 2023-06-27:
I've built a Plex LXC with both Ubuntu 22.04 and Debian bookworm with the same results. Given that Windows 11 is rock solid, I suspect it's a Linux driver issue causing the problem.

Coder84619, 2023-06-29:
I tested Plex Media Server on the same Proxmox 8 host using Windows 11 Pro, and both HDR tone mapping and hardware transcoding WORK just fine. So the HDR tone mapping issue appears to be isolated to Linux Plex LXCs.

devedse, 2023-09-11:
@Coder84619 
I have this issue when running Plex inside docker inside an lxc container. The strange thing is that I have this regardless of whether I had even installed this whole dkms VF thing. So I'm wondering how to get tone mapping to work at all.

Did you install any specific drivers on the proxmox 8 host for the Intel CPU?
I have the n100 CPU.

mike6041, 2023-10-12:
I am also experiencing this issue, did you ever find a solution?

Coder84619, 2023-11-25:
@strongtz Plex HDR tone mapping is still broken on Proxmox 8.1 with a Plex LXC on Ubuntu 22.04. There's an old post on the Plex forums about it: https://forums.plex.tv/t/hdr-tone-mapping-issue-on-alder-lake-vgpu-on-linux-proxmox-8-lxc/845630/21

Any idea on how to fix?

closed 2024-12-29 by strongtz