strongtz/i915-sriov-dkms #89: 12th Gen Intel i5-1240P,Unable to open sriov

LokiSharp, 2023-07-05:
`
root@PVE-NUC:/usr/src/i915-sriov-dkms-6.2# neofetch 
         .://:`              `://:.            root@PVE-NUC 
       `hMMMMMMd/          /dMMMMMMh`          ------------ 
        `sMMMMMMMd:      :mMMMMMMMs`           OS: Proxmox VE 8.0.3 x86_64 
`-/+oo+/:`.yMMMMMMMh-  -hMMMMMMMy.`:/+oo+/-`   Host: NUC12WSKi5 M46708-302 
`:oooooooo/`-hMMMMMMMyyMMMMMMMh-`/oooooooo:`   Kernel: 6.2.16-3-pve 
  `/oooooooo:`:mMMMMMMMMMMMMm:`:oooooooo/`     Uptime: 34 mins 
    ./ooooooo+- +NMMMMMMMMN+ -+ooooooo/.       Packages: 924 (dpkg) 
      .+ooooooo+-`oNMMMMNo`-+ooooooo+.         Shell: bash 5.2.15 
        -+ooooooo/.`sMMs`./ooooooo+-           Terminal: /dev/pts/0 
          :oooooooo/`..`/oooooooo:             CPU: 12th Gen Intel i5-1240P (16) @ 4.400GHz 
          :oooooooo/`..`/oooooooo:             GPU: Intel Alder Lake-P 
        -+ooooooo/.`sMMs`./ooooooo+-           Memory: 1592MiB / 63810MiB 
      .+ooooooo+-`oNMMMMNo`-+ooooooo+.
    ./ooooooo+- +NMMMMMMMMN+ -+ooooooo/.                               
  `/oooooooo:`:mMMMMMMMMMMMMm:`:oooooooo/`                             
`:oooooooo/`-hMMMMMMMyyMMMMMMMh-`/oooooooo:`
`-/+oo+/:`.yMMMMMMMh-  -hMMMMMMMy.`:/+oo+/-`
        `sMMMMMMMm:      :dMMMMMMMs`
       `hMMMMMMd/          /dMMMMMMh`
         `://:`              `://:`

root@PVE-NUC:/usr/src/i915-sriov-dkms-6.2# dkms status
i915-sriov-dkms/6.2, 6.2.16-3-pve, x86_64: installed
root@PVE-NUC:/usr/src/i915-sriov-dkms-6.2# cat /etc/sysfs.conf
devices/pci0000:00/0000:00:02.0/sriov_numvfs = 7
root@PVE-NUC:/usr/src/i915-sriov-dkms-6.2# dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.2.16-3-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on i915.enable_guc=3 i915.max_vfs=7
[    0.081585] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.2.16-3-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on i915.enable_guc=3 i915.max_vfs=7
[    7.751160] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    7.751185] i915 0000:00:02.0: vgaarb: deactivate vga console
[    7.751220] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    7.751623] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    7.752676] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[    7.758899] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc_ver2_16.bin (v2.16)
[    7.869314] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
[    7.869318] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!
[    7.871223] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
[    7.871227] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!
[    7.871687] i915 0000:00:02.0: [drm] GuC firmware i915/adlp_guc_70.bin version 70.5.1
[    7.871688] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc.bin version 7.9.3
[    7.872144] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
[    7.872148] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!
[    7.887321] i915 0000:00:02.0: [drm] HuC authenticated
[    7.887696] i915 0000:00:02.0: [drm] GuC submission enabled
[    7.887697] i915 0000:00:02.0: [drm] GuC SLPC enabled
[    7.888547] i915 0000:00:02.0: [drm] GuC RC: enabled
[    7.889514] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[    7.889666] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    7.890683] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    7.921470] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    7.928858] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    7.929369] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    7.929714] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
root@PVE-NUC:/usr/src/i915-sriov-dkms-6.2# lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
`

cyril-cheng, 2023-07-06:
[    7.869314] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Class(1):Compute(4)!
[    7.869318] i915 0000:00:02.0: [drm] Missing GuC-Err-Cap reglist Instance(2):Compute(4)!

Coder84619, 2023-07-09:
Might be the BIOS or motherboard. I have a Beelink with i5-1240P and vGPU works perfectly well with a Windows 11 Pro VM.

LokiSharp, 2023-07-11:
```
root@PVE-NUC:~# dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.2.16-3-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7[    0.081347] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.2.16-3-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7
[    7.387023] i915 0000:00:02.0: Running in SR-IOV PF mode
[    7.387495] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    7.387516] i915 0000:00:02.0: vgaarb: deactivate vga console
[    7.387550] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    7.387956] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    7.389015] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[    7.396103] i915 0000:00:02.0: Direct firmware load for i915/adlp_dmc.bin failed with error -2
[    7.397851] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc_ver2_16.bin (v2.16)
[    7.401683] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/adlp_guc_70.bin version 70.5.1
[    7.401687] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[    7.416677] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads!
[    7.417159] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    7.417160] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    7.417643] i915 0000:00:02.0: [drm] GuC RC: enabled
[    7.418651] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[    7.418768] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    7.419790] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    7.421411] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    7.421631] i915 0000:00:02.0: 7 VFs could be associated with this PF
[    7.429889] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    7.430396] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    7.430901] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    8.168139] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=io+mem
[    8.168226] i915 0000:00:02.1: enabling device (0000 -> 0002)
[    8.168245] i915 0000:00:02.1: Running in SR-IOV VF mode
[    8.168879] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.0.0
[    8.170976] i915 0000:00:02.1: [drm] VT-d active for gfx access
[    8.171009] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[    8.171516] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.0.0
[    8.171714] i915 0000:00:02.1: GuC firmware PRELOADED version 1.0 submission:SR-IOV VF
[    8.171716] i915 0000:00:02.1: HuC firmware PRELOADED
[    8.173515] i915 0000:00:02.1: [drm] Protected Xe Path (PXP) protected content support initialized
[    8.173520] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
[    8.173584] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.1 on minor 1
[    8.173815] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=io+mem
[    8.173822] i915 0000:00:02.1: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[    8.173863] i915 0000:00:02.2: enabling device (0000 -> 0002)
[    8.173875] i915 0000:00:02.2: Running in SR-IOV VF mode
[    8.174433] i915 0000:00:02.2: [drm] GT0: GUC: interface version 0.1.0.0
[    8.175245] i915 0000:00:02.2: [drm] VT-d active for gfx access
[    8.175268] i915 0000:00:02.2: [drm] Using Transparent Hugepages
[    8.176160] i915 0000:00:02.2: [drm] GT0: GUC: interface version 0.1.0.0
[    8.176694] i915 0000:00:02.2: GuC firmware PRELOADED version 1.0 submission:SR-IOV VF
[    8.176698] i915 0000:00:02.2: HuC firmware PRELOADED
[    8.178884] i915 0000:00:02.2: [drm] Protected Xe Path (PXP) protected content support initialized
[    8.178890] i915 0000:00:02.2: [drm] PMU not supported for this GPU.
[    8.178966] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.2 on minor 2
[    8.179259] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=io+mem
[    8.179266] i915 0000:00:02.1: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[    8.179273] i915 0000:00:02.2: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[    8.179339] i915 0000:00:02.3: enabling device (0000 -> 0002)
[    8.179356] i915 0000:00:02.3: Running in SR-IOV VF mode
[    8.179463] i915 0000:00:02.3: [drm] GT0: GUC: interface version 0.1.0.0
[    8.179772] i915 0000:00:02.3: [drm] VT-d active for gfx access
[    8.179793] i915 0000:00:02.3: [drm] Using Transparent Hugepages
[    8.180214] i915 0000:00:02.3: [drm] GT0: GUC: interface version 0.1.0.0
[    8.180366] i915 0000:00:02.3: GuC firmware PRELOADED version 1.0 submission:SR-IOV VF
[    8.180367] i915 0000:00:02.3: HuC firmware PRELOADED
[    8.182014] i915 0000:00:02.3: [drm] Protected Xe Path (PXP) protected content support initialized
[    8.182019] i915 0000:00:02.3: [drm] PMU not supported for this GPU.
[    8.182075] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.3 on minor 3
[    8.182294] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=io+mem
[    8.182301] i915 0000:00:02.1: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[... cut ...]
[   25.619288] i915 0000:00:02.6: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[   25.619295] i915 0000:00:02.7: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[   25.747408] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=io+mem
[   25.747419] i915 0000:00:02.7: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[   25.939272] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=none,decodes=io+mem:owns=io+mem
[   27.129986] i915 0000:00:02.0: Disabled 7 VFs
```

amosyuen, 2023-08-28:
@LokiSharp How did you resolve this? I'm running into problems and have similar error log as you.

> ```
> root@PVE-NUC:~# dmesg | grep i915
> [    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.2.16-3-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=3 [...]

rbray89, 2023-10-12:
@LokiSharp Same issue here. How did you resolve this?

LokiSharp, 2023-10-14:
@amosyuen @rbray89
Cancel the automatic startup of all virtual machines, then unbind all the previous pass-through graphics cards, and then rebind them after restarting.