strongtz/i915-sriov-dkms #90: 12600+pve8.0+win11,Configuration succeeds, but encoder is not available

lucienrenato, 2023-07-07:
Configured according the page https://www.derekseaman.com/2023/06/proxmox-ve-8-windows-11-vgpu-vt-d-passthrough-with-intel-alder-lake.html
i tested on windows11, the decoder works well, but when i start to export video,the softwore crashed. The software log says that the hardware encoder cannot be used.
i also used parsec to control my win11,but parsec cannot use the encoder too. The images of win11 never come to my computer
how to solve this problem?
 And I was wondering if it was possible to trick the vGPU to produce a fake monitor,the  parsec need to capture the monitor。

koda-git, 2023-07-17:
I have a similar problem with running the dkms module from PVE 8.0

It seems like the gpu can decode videos fine but cannot encode videos using quick sync in Handbrake and Parsec

HandBrake 1.6.1 (2023012300)

OS: Microsoft Windows NT 10.0.19044.0
CPU: Intel(R) N95
Ram: 6106 MB, 
GPU Information:
  Intel(R) UHD Graphics - 31.0.101.4575
  Microsoft Remote Display Adapter - 10.0.19041.662 (RDP)
Screen: 1920x1080

[18:45:29] qsv: is available on this system

  "Video": {
    "Encoder": "qsv_h264",
    "Level": "auto",
    "TwoPass": false,
    "Turbo": false,
    "ColorMatrixCode": 0,
    "Options": "lowpower=1",
    "Preset": "balanced",
    "Profile": "auto",
    "Quality": 22,
    "QSV": {
      "Decode": true
    },
    "HardwareDecode": 0
  }
[18:45:29] Intel Quick Sync Video support: yes, gpu list: 0
[18:45:30] qsv_enc_init: using 'hardware (1) via D3D11' implementation, API: 2.9
Error code -17,	hb_qsv_wait_on_sync	613
encqsv: hb_qsv_wait_on_sync failed (-17)
encqsvWork: qsv_enc_work failed -17

[18:45:30] reader: done. 1 scr changes
[18:45:30] work: average encoding speed for job is 0.000000 fps
[18:45:30] comb detect: heavy 0 | light 0 | uncombed 19 | total 19
[18:45:30] decomb: deinterlaced 0 | blended 0 | unfiltered 15 | total 15
[18:45:30] vfr: 11 frames output, 0 dropped and 0 duped for CFR/PFR
[18:45:30] vfr: lost time: 0 (0 frames)

koda-git, 2023-07-19:
Quick update! 
Encoding seems to be working fine on Ubuntu 23.04 with kernel 6.2.0!
I believe that the issue seems to be with the Windows driver

closed 2023-09-30 by strongtz

withlqs, 2024-03-07:
> Quick update! Encoding seems to be working fine on Ubuntu 23.04 with kernel 6.2.0! I believe that the issue seems to be with the Windows driver

Have you solved this problem? I met the same problem.

koda-git, 2024-03-08:
> > Quick update! Encoding seems to be working fine on Ubuntu 23.04 with kernel 6.2.0! I believe that the issue seems to be with the Windows driver
> 
> Have you solved this problem? I met the same problem.

Hmm, make sure that everything is running fine beforehand.
I’ve gotten the video encoding working (sunshine) on Windows 10 LTSC  and transcoding working in a debian container (jellyfin)