strongtz/i915-sriov-dkms #91: [BUG] kernel panic on pve 8 with kernel 6.2.16

BreakVoid, 2023-07-10:
Phenomenons:
1. Cannot connect to PVE Console(as well as ssh to Host and VMs). Sometimes VM(Ubuntu Server 23.04) fails first, and the shutdown, restart, stop, and reset commands sent from the console are all failed. And after several minutes, the host failed too. The monitor directly connected to the host would print some logs indicating a kernel panic occurs. Only a few times can be recorded by `journalctl`. 
2. After I removed the i915-sriov-dkms module, the kernel panic never occurs. And the host can continuously run for 24 hours. And when I reinstalled the module, the issue came back again.
Therefore, I think there should be some bugs in this module. There are some logs and information following. I hope it can help to solve this issue.

Hardware:
Model: CWWK(Changwang)
CPU: Intel N100
Memory: Crucial DDR5 4800 16GB
SSD: Crucial P3 500GB
Network: 4 * Intel I226-V

Installation Steps:
1. The `PACKAGE_VERSION` is set to 6.2, and I think it should be ok for kernel 6.2.16.
2. `i915.max_vfs` is set to 7 and `sriov_numvfs` in `sysfs.conf` is set to 2, since I do not need so many virtual devices.
3. An additional grub parameter `iommu=pt` is set.

Journalctl logs:
```
...
Jul 09 02:42:32 n100-pve kernel: BUG: unable to handle page fault for address: 0000000003491674
Jul 09 02:42:32 n100-pve kernel: #PF: supervisor read access in kernel mode
Jul 09 02:42:32 n100-pve kernel: #PF: error_code(0x0000) - not-present page
Jul 09 02:42:32 n100-pve kernel: PGD 0 P4D 0
Jul 09 02:42:32 n100-pve kernel: Oops: 0000 [#1] PREEMPT SMP NOPTI
Jul 09 02:42:32 n100-pve kernel: CPU: 3 PID: 1005 Comm: CPU 0/KVM Tainted: P     U     O       6.2.16-3-pve #1
Jul 09 02:42:32 n100-pve kernel: Hardware name: CWWK CW-AD4L-N V1/CW-AD4L-N V1, BIOS 5.27 04/04/2023
Jul 09 02:42:32 n100-pve kernel: RIP: 0010:__apic_accept_irq+0x115/0x2c0 [kvm]
Jul 09 02:42:32 n100-pve kernel: Code: 02 00 00 44 89 e8 44 89 e9 49 8b b4 24 a8 00 00 00 c1 f8 05 83 e1 1f c1 e0 04 05>
Jul 09 02:42:32 n100-pve kernel: RSP: 0018:ffffa9a340d2bc30 EFLAGS: 00010206
Jul 09 02:42:32 n100-pve kernel: RAX: 00000000000001f0 RBX: 0000000000000000 RCX: 000000000000000c
Jul 09 02:42:32 n100-pve kernel: RDX: 0000000000000000 RSI: ffff935e490d9000 RDI: ffff935e41ab4d00
Jul 09 02:42:32 n100-pve kernel: RBP: ffffa9a340d2bc70 R08: 0000000000000000 R09: 0000000000000000
Jul 09 02:42:32 n100-pve kernel: R10: 0000000000000001 R11: 0000000000000000 R12: ffff935e41ab4d00
Jul 09 02:42:32 n100-pve kernel: R13: 00000000000000ec R14: ffff935e48dd4800 R15: 0000000000000000
Jul 09 02:42:32 n100-pve kernel: FS:  00007f76c8d3f6c0(0000) GS:ffff9361afb80000(0000) knlGS:0000000000000000
Jul 09 02:42:32 n100-pve kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jul 09 02:42:32 n100-pve kernel: CR2: 0000000003491674 CR3: 0000000111322000 CR4: 0000000000752ee0
Jul 09 02:42:32 n100-pve kernel: PKRU: 55555554
Jul 09 02:42:32 n100-pve kernel: Call Trace:
Jul 09 02:42:32 n100-pve kernel:  <TASK>
Jul 09 02:42:32 n100-pve kernel:  ? kvm_apic_has_interrupt+0x5c/0x120 [kvm]
Jul 09 02:42:32 n100-pve kernel:  kvm_apic_local_deliver+0x7d/0xa0 [kvm]
Jul 09 02:42:32 n100-pve kernel:  kvm_inject_apic_timer_irqs+0x35/0x80 [kvm]
Jul 09 02:42:32 n100-pve kernel:  kvm_inject_pending_timer_irqs+0x17/0x60 [kvm]
Jul 09 02:42:32 n100-pve kernel:  kvm_arch_vcpu_ioctl_run+0x655/0x1740 [kvm]
Jul 09 02:42:32 n100-pve kernel:  ? vmx_vcpu_put+0x134/0x220 [kvm_intel]
Jul 09 02:42:32 n100-pve kernel:  ? default_wake_function+0x1a/0x40
Jul 09 02:42:32 n100-pve kernel:  ? pollwake+0x72/0xa0
Jul 09 02:42:32 n100-pve kernel:  kvm_vcpu_ioctl+0x297/0x7c0 [kvm]
Jul 09 02:42:32 n100-pve kernel:  ? __wake_up_common+0x71/0x150
Jul 09 02:42:32 n100-pve kernel:  ? __wake_up_locked_key+0x1b/0x40
Jul 09 02:42:32 n100-pve kernel:  ? __fget_light+0xa5/0x120
Jul 09 02:42:32 n100-pve kernel:  __x64_sys_ioctl+0x9d/0xe0
Jul 09 02:42:32 n100-pve kernel:  do_syscall_64+0x58/0x90
Jul 09 02:42:32 n100-pve kernel:  ? __fget_light+0xa5/0x120
Jul 09 02:42:32 n100-pve kernel:  ? ksys_write+0xe6/0x100
Jul 09 02:42:32 n100-pve kernel:  ? exit_to_user_mode_prepare+0x39/0x190
Jul 09 02:42:32 n100-pve kernel:  ? syscall_exit_to_user_mode+0x29/0x50
Jul 09 02:42:32 n100-pve kernel:  ? do_syscall_64+0x67/0x90
Jul 09 02:42:32 n100-pve kernel:  ? do_syscall_64+0x67/0x90
Jul 09 02:42:32 n100-pve kernel:  entry_SYSCALL_64_after_hwframe+0x72/0xdc
Jul 09 02:42:32 n100-pve kernel: RIP: 0033:0x7f76cce1dafb
Jul 09 02:42:32 n100-pve kernel: Code: 00 48 89 44 24 18 31 c0 48 8d 44 24 60 c7 04 24 10 00 00 00 48 89 44 24 08 48 8d>
Jul 09 02:42:32 n100-pve kernel: RSP: 002b:00007f76c8d39ff0 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
Jul 09 02:42:32 n100-pve kernel: RAX: ffffffffffffffda RBX: 0000561799847b40 RCX: 00007f76cce1dafb
Jul 09 02:42:32 n100-pve kernel: RDX: 0000000000000000 RSI: 000000000000ae80 RDI: 000000000000001d
Jul 09 02:42:32 n100-pve kernel: RBP: 000000000000ae80 R08: 0000561796ce2e00 R09: 0000000000000000
Jul 09 02:42:32 n100-pve kernel: R10: 0000000000000000 R11: 0000000000000246 R12: 0000000000000000
Jul 09 02:42:32 n100-pve kernel: R13: 0000000000000002 R14: 0000000000000001 R15: 0000000000000000
Jul 09 02:42:32 n100-pve kernel:  </TASK>
Jul 09 02:42:32 n100-pve kernel: Modules linked in: nfsv3 nfs_acl rpcsec_gss_krb5 auth_rpcgss nfsv4 nfs lockd grace fsc>
Jul 09 02:42:32 n100-pve kernel:  drm_kms_helper snd_timer i2c_algo_bit ov13858 snd v4l2_fwnode syscopyarea v4l2_async >
Jul 09 02:42:32 n100-pve kernel: CR2: 0000000003491674
Jul 09 02:42:32 n100-pve kernel: ---[ end trace 0000000000000000 ]---
Jul 09 02:42:32 n100-pve kernel: RIP: 0010:__apic_accept_irq+0x115/0x2c0 [kvm]
Jul 09 02:42:32 n100-pve kernel: Code: 02 00 00 44 89 e8 44 89 e9 49 8b b4 24 a8 00 00 00 c1 f8 05 83 e1 1f c1 e0 04 05>
Jul 09 02:42:32 n100-pve kernel: RSP: 0018:ffffa9a340d2bc30 EFLAGS: 00010206
Jul 09 02:42:32 n100-pve kernel: RAX: 00000000000001f0 RBX: 0000000000000000 RCX: 000000000000000c
Jul 09 02:42:32 n100-pve kernel: RDX: 0000000000000000 RSI: ffff935e490d9000 RDI: ffff935e41ab4d00
Jul 09 02:42:32 n100-pve kernel: RBP: ffffa9a340d2bc70 R08: 0000000000000000 R09: 0000000000000000
Jul 09 02:42:32 n100-pve kernel: R10: 0000000000000001 R11: 0000000000000000 R12: ffff935e41ab4d00
Jul 09 02:42:32 n100-pve kernel: R13: 00000000000000ec R14: ffff935e48dd4800 R15: 0000000000000000
Jul 09 02:42:32 n100-pve kernel: FS:  00007f76c8d3f6c0(0000) GS:ffff9361afb80000(0000) knlGS:0000000000000000
Jul 09 02:42:32 n100-pve kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jul 09 02:42:32 n100-pve kernel: CR2: 0000000003491674 CR3: 0000000111322000 CR4: 0000000000752ee0
Jul 09 02:42:32 n100-pve kernel: PKRU: 55555554
Jul 09 02:42:32 n100-pve kernel: note: CPU 0/KVM[1005] exited with irqs disabled
```

pve versions:
```
proxmox-ve: 8.0.1 (running kernel: 6.2.16-3-pve)
pve-manager: 8.0.3 (running version: 8.0.3/bbf3993334bfa916)
pve-kernel-6.2: 8.0.2
pve-kernel-6.2.16-3-pve: 6.2.16-3
ceph-fuse: 17.2.6-pve1+3
corosync: 3.1.7-pve3
criu: 3.17.1-2
glusterfs-client: 10.3-5
ifupdown2: 3.2.0-1+pmx3
ksm-control-daemon: 1.4-1
libjs-extjs: 7.0.0-3
libknet1: 1.25-pve1
libproxmox-acme-perl: 1.4.6
libproxmox-backup-qemu0: 1.4.0
libproxmox-rs-perl: 0.3.0
libpve-access-control: 8.0.3
libpve-apiclient-perl: 3.3.1
libpve-common-perl: 8.0.6
libpve-guest-common-perl: 5.0.3
libpve-http-server-perl: 5.0.4
libpve-rs-perl: 0.8.3
libpve-storage-perl: 8.0.2
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 5.0.2-4
lxcfs: 5.0.3-pve3
novnc-pve: 1.4.0-2
proxmox-backup-client: 3.0.1-1
proxmox-backup-file-restore: 3.0.1-1
proxmox-kernel-helper: 8.0.2
proxmox-mail-forward: 0.2.0
proxmox-mini-journalreader: 1.4.0
proxmox-widget-toolkit: 4.0.6
pve-cluster: 8.0.2
pve-container: 5.0.4
pve-docs: 8.0.4
pve-edk2-firmware: 3.20230228-4
pve-firewall: 5.0.2
pve-firmware: 3.7-1
pve-ha-manager: 4.0.2
pve-i18n: 3.0.5
pve-qemu-kvm: 8.0.2-3
pve-xtermjs: 4.16.0-3
qemu-server: 8.0.6
smartmontools: 7.3-pve1
spiceterm: 3.3.0
swtpm: 0.8.0+pve1
vncterm: 1.8.0
zfsutils-linux: 2.1.12-pve1
```

closed 2023-07-12 by BreakVoid

wray-lee, 2023-10-08:
Could you tell me how to solve this problem, please?

BreakVoid, 2023-10-08:
I just found that the reason for the kernel panic is the bad memory module. I used Memtest+ to test the module and got plenty of errors. Then, I replaced it with a new one. The issue never occurs again.

wray-lee, 2023-10-08:
> I just found that the reason for the kernel panic is the bad memory module. I used Memtest+ to test the module and got plenty of errors. Then, I replaced it with a new one. The issue never occurs again.

Ok, thank you. I will try it.

qsx313, 2023-10-09:
I have the same kernel panic problem with CW x86 p5 and Crucial DDR5 4800 32GB,which menory module do you choice now?

BreakVoid, 2023-10-09:
Do you buy the module from Taobao? 
I doubt that the module I purchased is a fake one. The new module was purchased from JD.com, which is Crucial 32GiB 4800. I found that the modules from Taobao and JD.com were packed differently. However, after I asked the official store at Taobao about the module package, I got the former one purchased from Taobao's third-party store, which might be self-packed since it was different from the official one (including Taobao's official and JD.com). Therefore, it should be that some of Taobao's third-party stores sell fake modules, which can cause memory errors and lead to kernel panic.

The initial bad module is 16GiB. And the new good module is 32GiB. Although the official documentation states that Intel N100 only supports 16GiB, the 32GiB module works well after a really long initial memory training. The first startup after installing the memory is quite long, which may take about 10 minutes. But it would finally work well, and the next boots would be very fast.

qsx313, 2023-10-09:
Roger that，I bought it from PDD ,shipping with a simple package.I would like to bug another one.Thank you!