strongtz/i915-sriov-dkms #94: Error! Bad return status for module build on kernel: 6.2.16-4-pve (x86_64)

quejoo, 2023-07-21:
root@pvefo:~# dkms install -m i915-sriov-dkms -v 6.1 -k 6.2.16-4-pve --force

Sign command: /lib/modules/6.2.16-4-pve/build/scripts/sign-file

Signing key: /var/lib/dkms/mok.key

Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:

Cleaning build area...

make -j4 KERNELRELEASE=6.2.16-4-pve -j4 -C /lib/modules/6.2.16-4-pve/build M=/var/lib/dkms/i915-sriov-dkms/6.1/build....(bad exit status: 2)

Error! Bad return status for module build on kernel: 6.2.16-4-pve (x86_64)

Consult /var/lib/dkms/i915-sriov-dkms/6.1/build/make.log for more information.

[make.log](https://github.com/strongtz/i915-sriov-dkms/files/12134625/make.log)

ysymmsj, 2023-07-22:
嗯，一模一样的错误，无法安装了，低版本的时候可以  升级成最新的就不行啦

craschnet, 2023-07-25:
You'll want to install the older kernel versions that I saw that worked 6.2.9-1 but unable to find it in the bookworm repo

`root@deathstar2:/usr/src# apt search pve-kernel-6.2
Sorting... Done
Full Text Search... Done
pve-kernel-6.2/stable,now 8.0.3 all [installed]
  Latest Proxmox VE Kernel Image

pve-kernel-6.2.16-1-pve/stable 6.2.16-1 amd64
  Proxmox Kernel Image

pve-kernel-6.2.16-2-pve/stable 6.2.16-2 amd64
  Proxmox Kernel Image

pve-kernel-6.2.16-3-pve/stable,now 6.2.16-3 amd64 [installed]
  Proxmox Kernel Image

pve-kernel-6.2.16-4-pve/stable,now 6.2.16-5 amd64 [installed]
  Proxmox Kernel Image`

same with pve-headers

craschnet, 2023-07-26:
Ok I enabled the pvetest respo in proxmox and installed 6.2.16-5-pve kernel and headers and that seem to solved the problem.
I think they may have streamed this driver into 6.2.16-5-pve kernel

closed 2023-09-30 by strongtz