strongtz/i915-sriov-dkms #95: Unable to use intel_gpu_top  in pve lxc container

zgqq, 2023-07-22:
pve 8.0 
```
root@jellyfin:/dev# uname -r
6.2.16-4-pve
root@jellyfin:/dev#  intel_gpu_top 
Failed to detect engines! (No such file or directory)
```

zgqq, 2023-07-22:
had the same problem
https://forums.unraid.net/topic/138458-error-using-intel_gpu_top/

icenine9, 2023-08-06:
Similar problem here. On the host: 

```
 root@proxmox:~# intel_gpu_top 
Failed to detect engines! (No such file or directory)
(Kernel 4.16 or newer is required for i915 PMU support.)
Segmentation fault
```
However, when passed with '-d sriov' I get an updating top line:

`intel-gpu-top: Intel Raptorlake_s (Gen12) @ /dev/dri/card0 -   77/ 702 MHz;  55% RC6;  0.25/54.00 W;        0 irqs/s
`
a few of the stats update, but the chart counters are all at 0% even under a load. 

On the LXC, can't get it to work at all: 

```
root@lateralus:~# intel_gpu_top
Failed to detect engines! (No such file or directory)
(Kernel 4.16 or newer is required for i915 PMU support.)
Segmentation fault
root@lateralus:~# intel_gpu_top -d sriov
Failed to initialize PMU! (Permission denied)

```

right-0903, 2023-08-12:
> Similar problem here. On the host:
> 
> ```
>  root@proxmox:~# intel_gpu_top 
> Failed to detect engines! (No such file or directory)
> (Kernel 4.16 or newer is required for i915 PMU support.)
> Segmentation fault
> ```
> 
> However, when passed with '-d sriov' I get an updating top line:
> 
> `intel-gpu-top: Intel Raptorlake_s (Gen12) @ /dev/dri/card0 - 77/ 702 MHz; 55% RC6; 0.25/54.00 W; 0 irqs/s ` a few of the stats update, but the chart counters are all at 0% even under a load.
> 
> On the LXC, can't get it to work at all:
> 
> ```
> root@lateralus:~# intel_gpu_top
> Failed to detect engines! (No such file or directory)
> (Kernel 4.16 or newer is required for i915 PMU support.)
> Segmentation fault
> root@lateralus:~# intel_gpu_top -d sriov
> Failed to initialize PMU! (Permission denied)
> ```
> 
> Seems cosmetic, as I'm able to use HW video features in both an lxc and win10 vm concurrently.

`VFs` do not support PMU,  `intel_gpu_top` rquire that. Only the `PF` support that, I think all workloads are under the `PF`, so `VFs` don't know their workloads. BTW, how can you make it work with win10? I got a code43 on win10 22h2.

```sh
[   67.507835] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
[   67.513240] i915 0000:00:02.2: [drm] PMU not supported for this GPU.
```

right-0903, 2023-08-12:
After you have created Vfs, you should specific the real device or `PF` to monitor, you can go with `intel_gpu_top -d sys:/sys/devices/pci0000:00/0000:00:02.0` there are many ways to do it with `-d` option.

closed 2023-09-30 by strongtz

scyto, 2023-10-07:
> I got a code43 on win10 22h2

change the CPU type to host https://gist.github.com/scyto/e4e3de35ee23fdb4ae5d5a3b85c16ed3#add-vgpu-to-a-windows-11-or-server-2022-vm

scyto, 2023-10-07:
> After you have created Vfs, you should specific the real device or `PF` to monitor, you can go with `intel_gpu_top -d sys:/sys/devices/pci0000:00/0000:00:02.0` there are many ways to do it with `-d` option.

This doesn't seem to work in a unprivileged container

right-0903, 2023-10-07:
> change the CPU type to host https://gist.github.com/scyto/e4e3de35ee23fdb4ae5d5a3b85c16ed3#add-vgpu-to-a-windows-11-or-server-2022-vm

this repo don't work for me, I use another, I have noticed I should disable hidden hypervisor.