strongtz/i915-sriov-dkms #96: [3.973982] i915 0000:00:02.0: driver does not support SR-IOV configuration via sysfs

tckxjzc, 2023-07-29:
`root@pve:~# dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.2.16-5-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7
[    0.026917] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.2.16-5-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7
[    3.547779] i915: unknown parameter 'max_vfs' ignored
[    3.548485] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    3.548512] i915 0000:00:02.0: vgaarb: deactivate vga console
[    3.548570] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    3.548912] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    3.550891] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc_ver2_16.bin (v2.16)
[    3.659430] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_70.bin version 70.5.1
[    3.659436] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc.bin version 7.9.3
[    3.663782] i915 0000:00:02.0: [drm] HuC authenticated
[    3.664254] i915 0000:00:02.0: [drm] GuC submission enabled
[    3.664255] i915 0000:00:02.0: [drm] GuC SLPC enabled
[    3.664556] i915 0000:00:02.0: [drm] GuC RC: enabled
[    3.665245] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    3.665780] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    3.667501] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    3.668068] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    3.668163] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    3.668202] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    3.973957] i915 0000:00:02.0: driver does not support SR-IOV configuration via sysfs
[    3.973982] i915 0000:00:02.0: driver does not support SR-IOV configuration via sysfs`

root@pve:~# dkms status
i915-sriov-dkms/6.1, 6.2.16-3-pve, x86_64: installed

root@pve:~# ls /lib/modules/6.2.16-3-pve/kernel/drivers/gpu/drm/i915
i915.ko  kvmgt.ko

root@pve:~# lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-N [UHD Graphics]

CPU: Intel® Processor N100

ChrisKretschmer, 2023-07-30:
I also followed the tutorial on 
https://www.derekseaman.com/2023/06/proxmox-ve-8-windows-11-vgpu-vt-d-passthrough-with-intel-alder-lake.html

```[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.2.16-5-pve root=/dev/mapper/pve-root ro quiet acpi_enforce_resources=lax intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7
[    0.083838] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.2.16-5-pve root=/dev/mapper/pve-root ro quiet acpi_enforce_resources=lax intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7
[    4.056503] i915: unknown parameter 'max_vfs' ignored
[    4.057530] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    4.057564] i915 0000:00:02.0: vgaarb: deactivate vga console
[    4.057626] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    4.058860] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    4.059666] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[    4.060067] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/rkl_dmc_ver2_03.bin (v2.3)
[    4.163533] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_70.1.1.bin version 70.1.1
[    4.163536] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc_7.9.3.bin version 7.9.3
[    4.165740] i915 0000:00:02.0: [drm] HuC authenticated
[    4.166021] i915 0000:00:02.0: [drm] GuC submission enabled
[    4.166022] i915 0000:00:02.0: [drm] GuC SLPC enabled
[    4.166315] i915 0000:00:02.0: [drm] GuC RC: enabled
[    4.166842] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[    4.166944] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    4.167320] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    4.168769] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    4.169394] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    4.170094] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
```

Is it possible, the new PVE 8.* kernels are missing some required modules?

ChrisKretschmer, 2023-07-30:
in my case, i built the module by accident for the wrong kernel. I had a kernel-update pending, so after restart, it booted the new kernel and the stock module.

now the message with 'max_vfs' is gone, but still nothing happens :/

```
root@pve:~# dmesg |grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.2.16-5-pve root=/dev/mapper/pve-root ro quiet acpi_enforce_resources=lax intel_iommu=on                                                                             i915.enable_guc=3 i915.max_vfs=7
[    0.083894] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.2.16-5-pve root=/dev/mapper/pve-root ro quiet acpi_enforce_resources=lax intel_io                                                                            mmu=on i915.enable_guc=3 i915.max_vfs=7
[    4.121469] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    4.121487] i915 0000:00:02.0: vgaarb: deactivate vga console
[    4.121542] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    4.122559] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    4.123181] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[    4.123419] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/rkl_dmc_ver2_03.bin (v2.3)
[    4.127355] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.1.1.bin version 70.1.1
[    4.127358] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc_7.9.3.bin version 7.9.3
[    4.129558] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads!
[    4.129843] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    4.129844] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    4.130172] i915 0000:00:02.0: [drm] GuC RC: enabled
[    4.130726] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[    4.130835] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    4.131321] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    4.132848] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    4.133502] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    4.134192] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes

```

closed 2023-08-15 by tckxjzc

tckxjzc, 2023-08-15:
> in my case, i built the module by accident for the wrong kernel. I had a kernel-update pending, so after restart, it booted the new kernel and the stock module.
> 
> now the message with 'max_vfs' is gone, but still nothing happens :/
> 
> ```
> root@pve:~# dmesg |grep i915
> [    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.2.16-5-pve root=/dev/mapper/pve-root ro quiet acpi_enforce_resources=lax intel_iommu=on                                                                             i915.enable_guc=3 i915.max_vfs=7
> [    0.083894] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.2.16-5-pve root=/dev/mapper/pve-root ro quiet acpi_enforce_resources=lax intel_io                                                                            mmu=on i915.enable_guc=3 i915.max_vfs=7
> [    4.121469] i915 0000:00:02.0: [drm] VT-d active for gfx access
> [    4.121487] i915 0000:00:02.0: vgaarb: deactivate vga console
> [    4.121542] i915 0000:00:02.0: [drm] Using Transparent Hugepages
> [    4.122559] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
> [    4.123181] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
> [    4.123419] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/rkl_dmc_ver2_03.bin (v2.3)
> [    4.127355] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.1.1.bin version 70.1.1
> [    4.127358] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc_7.9.3.bin version 7.9.3
> [    4.129558] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads!
> [    4.129843] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
> [    4.129844] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
> [    4.130172] i915 0000:00:02.0: [drm] GuC RC: enabled
> [    4.130726] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
> [    4.130835] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
> [    4.131321] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
> [    4.132848] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
> [    4.133502] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
> [    4.134192] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
> ```
> 
> also, the /sriov_numvfs node is missing in the sysfs dir

> 
我解决了,你改下版本号试试,6.1->6.2.
保持内核版本和安装的版本一致
你的内核版本好像是6.2.16-5,但是安装的版本是6.2.16-3
nano dkms.conf
PACKAGE_NAME="i915-sriov-dkms"
PACKAGE_VERSION="6.2"
mv i915-sriov-dkms/ /usr/src/i915-sriov-dkms-6.2
其他步骤一样

rbray89, 2023-10-12:
@ChrisKretschmer Were you able to get this working? I'm seeing an identical issue.

ChrisKretschmer, 2023-10-12:
> @ChrisKretschmer Were you able to get this working? I'm seeing an identical issue.

havent tryed it again tbh