strongtz/i915-sriov-dkms #103: Ubuntu 22.04 [drm] *ERROR* GuC submission disabled Device initialization failed (-19)

scottrus, 2023-08-21:
On my Ubuntu 22.04 guest image I'm not able to get the driver to load. I'm using the same dkms driver on the host as I am on the guest.

Hardware is Intel NUC11PAHi5 with i5-1135G7 (Tiger Lake)
I'm running Proxmox 8.0.4 with kernel 6.2.16-8-pve on the host.

I followed directions and setup dkms on the host and everything appears to be working well. On the PVE host, here is my `dimesg | grep i915`
<details>

```
root@pve-node-01:/etc/modprobe.d# dmesg|grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.2.16-8-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on i915.enable_guc=3 i915.max_vfs=7
[    0.089345] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.2.16-8-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on i915.enable_guc=3 i915.max_vfs=7
[    6.678343] i915: module verification failed: signature and/or required key missing - tainting kernel
[    6.935063] i915 0000:00:02.0: Running in SR-IOV PF mode
[    6.935559] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    6.935576] i915 0000:00:02.0: vgaarb: deactivate vga console
[    6.935622] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    6.986918] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    6.987941] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/tgl_dmc_ver2_12.bin (v2.12)
[    6.988312] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[    6.992159] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.1.1.bin version 70.1.1
[    6.992163] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc_7.9.3.bin version 7.9.3
[    6.995695] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads!
[    6.996062] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    6.996063] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    6.996486] i915 0000:00:02.0: [drm] GuC RC: enabled
[    6.997365] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[    6.997482] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    6.998386] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    7.000052] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    7.000212] i915 0000:00:02.0: 7 VFs could be associated with this PF
[    7.001259] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    7.002268] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    7.560112] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=io+mem
[    7.560173] i915 0000:00:02.1: enabling device (0000 -> 0002)
[    7.560186] i915 0000:00:02.1: Running in SR-IOV VF mode
[    7.560514] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.0.0
[    7.561929] i915 0000:00:02.1: [drm] VT-d active for gfx access
[    7.561946] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[    7.562474] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.0.0
[    7.562781] i915 0000:00:02.1: GuC firmware PRELOADED version 1.0 submission:SR-IOV VF
[    7.562782] i915 0000:00:02.1: HuC firmware PRELOADED
[    7.565607] i915 0000:00:02.1: [drm] Protected Xe Path (PXP) protected content support initialized
[    7.565612] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
[    7.565670] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.1 on minor 1
[    7.566107] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=io+mem
[    7.566113] i915 0000:00:02.1: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[    7.566171] i915 0000:00:02.2: enabling device (0000 -> 0002)
[    7.566187] i915 0000:00:02.2: Running in SR-IOV VF mode
[    7.566526] i915 0000:00:02.2: [drm] GT0: GUC: interface version 0.1.0.0
[    7.566904] i915 0000:00:02.2: [drm] VT-d active for gfx access
[    7.566918] i915 0000:00:02.2: [drm] Using Transparent Hugepages
[    7.567351] i915 0000:00:02.2: [drm] GT0: GUC: interface version 0.1.0.0
[    7.567635] i915 0000:00:02.2: GuC firmware PRELOADED version 1.0 submission:SR-IOV VF
[    7.567637] i915 0000:00:02.2: HuC firmware PRELOADED
[    7.569567] i915 0000:00:02.2: [drm] Protected Xe Path (PXP) protected content support initialized
[    7.569571] i915 0000:00:02.2: [drm] PMU not supported for this GPU.
[    7.569621] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.2 on minor 2
[    7.569930] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=io+mem
[    7.569936] i915 0000:00:02.1: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[    7.569943] i915 0000:00:02.2: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[    7.569989] i915 0000:00:02.3: enabling device (0000 -> 0002)
[    7.569999] i915 0000:00:02.3: Running in SR-IOV VF mode
[    7.570204] i915 0000:00:02.3: [drm] GT0: GUC: interface version 0.1.0.0
[    7.570424] i915 0000:00:02.3: [drm] VT-d active for gfx access
[    7.570439] i915 0000:00:02.3: [drm] Using Transparent Hugepages
[    7.570746] i915 0000:00:02.3: [drm] GT0: GUC: interface version 0.1.0.0
[    7.570884] i915 0000:00:02.3: GuC firmware PRELOADED version 1.0 submission:SR-IOV VF
[    7.570886] i915 0000:00:02.3: HuC firmware PRELOADED
[    7.572099] i915 0000:00:02.3: [drm] Protected Xe Path (PXP) protected content support initialized
[... 91 lines cut ...]
      > i915-sriov-dkms/6.1, 6.2.0-26-generic, x86_64: installed
8.  Edit grub with `vi /etc/default/grub`
      > GRUB_CMDLINE_LINUX_DEFAULT="console=ttyS0 i915.enable_guc=3"
9.  Run `update-grub`
10. Run `update-initramfs -u -k $(uname -r)`
11. Reboot the guest VM and add pci device 02.00.7 to the VM

[image]

Once the guest VM booted again I checked `dmesg` and found the following errors. I'm at a loss what I'm doing wrong where the guest VM will not pick up the VF device.

```
root@green:~# dmesg | grep i915
[    2.144766] i915: loading out-of-tree module taints kernel.
[    2.146426] i915: module verification failed: signature and/or required key missing - tainting kernel
[    2.346320] i915 0000:01:00.0: Running in SR-IOV VF mode
[    2.346640] i915 0000:01:00.0: [drm] *ERROR* GuC submission disabled
[    2.415916] i915 0000:01:00.0: Device initialization failed (-19)
[    2.416267] i915 0000:01:00.0: Please file a bug on drm/i915; see https://gitlab.freedesktop.org/drm/intel/-/wikis/How-to-file-i915-bugs for details.

root@green:~# uname -a
Linux green 6.2.0-26-generic #26~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Jul 13 16:27:29 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

That's where I'm stuck. Anyone have an idea about the `[drm] *ERROR* GuC submission disabled` error?

scottrus, 2023-08-23:
Making a bit more progress. While troubleshooting the VM guest (22.04 with HWE kernel as shown above) I installed sysfsutils and ran 

```
root@red:/# systool -v -m i915 | grep guc
    enable_guc          = "3"
    guc_firmware_path   = "(null)"
    guc_log_level       = "-1"
```

Seeing the `enable_guc` not getting picked up from my `/etc/default/grub` config, I added this to the modules.d with `echo 'options i915 enable_guc=3' > /etc/modprobe.d/i915.conf ` and rebooted.

After reboot I see the i915 loading in dmesg, good output in lspci, and vainfo reporting details.

The dmesg output from the VM guest:
<details>

```
root@red:/# dmesg | grep i915
[    2.125654] i915: loading out-of-tree module taints kernel.
[    2.127335] i915: module verification failed: signature and/or required key missing - tainting kernel
[    2.326574] i915 0000:01:00.0: Running in SR-IOV VF mode
[    2.327136] i915 0000:01:00.0: [drm] GT0: GUC: interface version 0.1.0.0
[    2.328006] i915 0000:01:00.0: [drm] VT-d active for gfx access
[    2.328495] i915 0000:01:00.0: vgaarb: deactivate vga console
[    2.328948] i915 0000:01:00.0: [drm] Using Transparent Hugepages
[    2.330197] i915 0000:01:00.0: [drm] GT0: GUC: interface version 0.1.0.0
[    2.330974] i915 0000:01:00.0: GuC firmware PRELOADED version 1.0 submission:SR-IOV VF
[    2.331598] i915 0000:01:00.0: HuC firmware PRELOADED
[    2.333666] i915 0000:01:00.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    2.334365] i915 0000:01:00.0: [drm] PMU not supported for this GPU.
[    2.334953] [drm] Initialized i915 1.6.0 20201103 for 0000:01:00.0 on minor 0
```

</details>

The lspci output on the VM guest:
<details>

```
root@red:/# lspci -vs 0000:01:00.0
01:00.0 VGA compatible controller: Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] (rev 01) (prog-if 00 [VGA controller])
	Subsystem: Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics]
	Physical Slot: 0
	Flags: bus master, fast devsel, latency 0, IRQ 39
	Memory at c0000000 (64-bit, non-prefetchable) [size=16M]
	Memory at 800000000 (64-bit, prefetchable) [size=512M]
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable+ 64bit-
	Kernel driver in use: i915
	Kernel modules: i915
```

</details>

The vainfo output
<details

```
root@red:/# vainfo
error: XDG_RUNTIME_DIR not set in the environment.
error: can't connect to X server!
libva info: VA-API version 1.14.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_14
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.14 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.3.1 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSliceLP
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSliceLP
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile1            :	VAEntrypointVLD
      VAProfileVP9Profile1            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile3            :	VAEntrypointVLD
      VAProfileVP9Profile3            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileHEVCMain422_10         :	VAEntrypointVLD
      VAProfileHEVCMain422_12         :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain10          :	VAEntrypointVLD
      VAProfileHEVCSccMain10          :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444         :	VAEntrypointVLD
      VAProfileHEVCSccMain444         :	VAEntrypointEncSliceLP
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointEncSliceLP
```

</details>

The next issue is when using the transcoder in Plex, the image is not rendered correctly as shown below. Not sure the cause of this yet.

[image]

blebo, 2023-08-27:
Just a side note: 
It might help to update you VM iso for Ubuntu to 22.04.3 if you need to do this again, as it comes with kernel 6.2 build in (removing your original step 1).

scottrus, 2023-08-28:
I’ll have to try the jammy server iso I guess. I used the latest jammy cloud init amd64 image and was surprised to see the older kernel still there.

tom2201, 2023-09-15:
The same problem arises.  I'm using i3-n305 with pve 7.4 VM guest is ubuntu 23.04 kernel 6.4.0-060400, Run plex under docker.  But in VM guest Windows 11 + Win plex server decoding is fine again.

scottrus, 2023-09-21:
Tone map problems likely related to https://forums.plex.tv/t/pms-1-32-6-hw-transcoding-issues-and-corrections/853757/103 which happened around the same time as this issue being opened. 

Doesn’t seem related.

scottrus, 2023-09-21:
This issue happens on boot of the guest where GuC is never enabled

tom2201, 2023-09-22:
Thank you. I have tested it. After updating to 1.32.6.7521-24e4dff81 still have the same problem. Maybe the linux program is not tuned to be compatible with the latest CPU. I've sent feedback to plex and waiting for the latest solution.