strongtz/i915-sriov-dkms #105: Unable to set sriov_numvfs since kernel 6.4

odror, 2023-08-26:
when typing
```
echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
```
I get the following error:
```
-bash: echo: write error: No such file or directory
```
Journalctl has the following message.
```
Aug 26 16:15:32 i9-13900k-arch kernel: i915 0000:00:02.0: driver does not support SR-IOV configuration via sysfs
```
Any Ideas how to resolve that

I would like to mention also that when trying to assign the value through /etc/sysfs.com nothing happens.

```
00:02.0 VGA compatible controller: Intel Corporation Raptor Lake-S GT1 [UHD Graphics 770] (rev 04)
01:00.0 VGA compatible controller: NVIDIA Corporation AD104 [GeForce RTX 4070 Ti] (rev a1)
05:00.0 VGA compatible controller: NVIDIA Corporation GM204GL [Quadro M4000] (rev a1)
```

gotem, 2023-08-29:
I dont think the module is compatible with that version yet, I had to downgrade to 6.3

scyto, 2023-09-19:
try compiling now, it worked for me on 6.5 with this weeks changes

odror, 2023-09-29:
It was working with 6.4

I have the same issue with 6.5

closed 2024-01-29 by strongtz