strongtz/i915-sriov-dkms #106: Kernel driver in use: vf90-pci

amosyuen, 2023-08-29:
I was following this guide https://www.derekseaman.com/2023/06/proxmox-ve-8-windows-11-vgpu-vt-d-passthrough-with-intel-alder-lake.html to enable SR-IOV in proxmox. But it failed to load.

`dmesg | grep i915` output:
```
[   12.327141] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=io+mem
[   12.327147] i915 0000:00:02.1: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[   12.327153] i915 0000:00:02.2: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[   12.327221] i915 0000:00:02.3: enabling device (0000 -> 0002)
[   12.327245] i915 0000:00:02.3: Running in SR-IOV VF mode
[   12.327371] i915 0000:00:02.3: [drm] GT0: GUC: interface version 0.1.0.0
[   12.327852] i915 0000:00:02.3: [drm] VT-d active for gfx access
[   12.327869] i915 0000:00:02.3: [drm] Using Transparent Hugepages
[   12.328216] i915 0000:00:02.3: [drm] GT0: GUC: interface version 0.1.0.0
[   12.328367] i915 0000:00:02.3: GuC firmware PRELOADED version 1.0 submission:SR-IOV VF
[   12.328368] i915 0000:00:02.3: HuC firmware PRELOADED
[   12.329969] i915 0000:00:02.3: [drm] Protected Xe Path (PXP) protected content support initialized
[   12.329973] i915 0000:00:02.3: [drm] PMU not supported for this GPU.
[   12.330051] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.3 on minor 3
[   12.330240] i915 0000:00:02.0: Enabled 3 VFs
[   21.007018] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=io+mem
[   21.007034] i915 0000:00:02.2: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[   21.007045] i915 0000:00:02.3: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[   21.119017] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=io+mem
[   21.119033] i915 0000:00:02.3: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[   21.239011] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=none,decodes=io+mem:owns=io+mem
[   22.482633] i915 0000:00:02.0: Disabled 3 VFs
```

From https://github.com/strongtz/i915-sriov-dkms/issues/48#issuecomment-1464772926 it sounded like i should verify the kernel driver is `i915`. 

`lspci -vs 00:02.0` output:
```
00:02.0 VGA compatible controller: Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] (rev 01) (prog-if 00 [VGA controller])
        DeviceName: Onboard - Video
        Subsystem: Device 1c6c:078b
        Flags: bus master, fast devsel, latency 0, IRQ 138, IOMMU group 0
        Memory at 6000000000 (64-bit, non-prefetchable) [size=16M]
        Memory at 4000000000 (64-bit, prefetchable) [size=256M]
        I/O ports at 3000 [size=64]
        Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
        Capabilities: [40] Vendor Specific Information: Len=0c <?>
        Capabilities: [70] Express Root Complex Integrated Endpoint, MSI 00
        Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable+ 64bit-
        Capabilities: [d0] Power Management version 2
        Capabilities: [100] Process Address Space ID (PASID)
        Capabilities: [200] Address Translation Service (ATS)
        Capabilities: [300] Page Request Interface (PRI)
        Capabilities: [320] Single Root I/O Virtualization (SR-IOV)
        Kernel driver in use: vfio-pci
        Kernel modules: i915
```

Looks like the kernel driver is `vfio-pci` instead of `i915`, but I'm not sure what is the correct steps to fix that.

amosyuen, 2023-08-29:
Figured it out, the problem was I already had the graphics set to pass through to the VM. Removing it from the VM, fixed the problem.

closed 2023-08-29 by amosyuen

JobberRT, 2024-03-08:
@amosyuen Hi, sorry for interrupt, but do you mean everytime I restart the PVE host, I have to remove the vGPU from VM, then after started, readd it to VM?