strongtz/i915-sriov-dkms #107: Create kmod for Fedora/RPM based systems (especially immutable systems like Silverblue/Kinoite)

sihawken, 2023-08-30:
I have been attempting to use this dkms repository as a base to build an akmod rpm package, so that fedora based systems, including immutable (silverblue/kinoite) can use this kernel module.

Unfortunately, immutable Fedora (Silverblue/Kinoite) is not compatible with dkms. Only akmods are supported.

I gave it my best attempt here [https://github.com/sihawken/i915-sriov-kmod/tree/main](https://github.com/sihawken/i915-sriov-kmod/tree/main) it seems to work and load. However, when I run ``dmesg | grep i915``, I get the dreaded: ``i915: unknown parameter 'max_vfs' ignored`` message.

I have a copr repository here if you want to try it out on your own system
Fedora repository: https://copr.fedorainfracloud.org/coprs/sihawken/akmods/

Immutable Fedora: ``rpm-ostree install akmod-i915-sriov``
Fedora: ``dnf install akmod-i915-sriov``

Perhaps I am missing something obvious here. I would really appreciate it if someone with more experience regarding kernel modules would take a look.

Thanks

sihawken, 2023-09-03:
I managed to install the akmod on Fedora Workstation 38, with a ``depmod -a`` and ``dracut --force``. 

It seems that I manage to solve my own problem, but it would be great to have this be officially supported by this repo.

masiv1001, 2023-09-24:
> I managed to install the akmod on Fedora Workstation 38, with a `depmod -a` and `dracut --force`.
> 
> It seems that I manage to solve my own problem, but it would be great to have this be officially supported by this repo.

How did you manage to install it? I'm trying to install it on Fedora 38 too.

sihawken, 2023-10-23:
@masiv1001 The instructions can be found here: https://github.com/sihawken/i915-sriov-kmod/

EverlastingOS, 2023-11-24:
> @masiv1001 The instructions can be found here: https://github.com/sihawken/i915-sriov-kmod/

I did install from the instructions, did not give any error,  how can i see if it works? Can i create VA-API video acceleration in VM  now?

masiv1001, 2023-11-25:
> I did install from the instructions, did not give any error, how can i see if it works? Can i create VA-API video acceleration in VM now?
@EverlastingOS You can test is by watching the output of the following commands (creating 2 test virtual devices):
```
sudo su
echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
```
It should not output anything, then run:
```
lspci | grep VGA
```
If everything goes well, it should list three devices, 00:02.0 and it's respective sub-devices 2.1 and 2.3.
Beware VM usage requires extra steps, some users could get it working but I couldn't get it working with a windows guest, for further reading go onto some of the other issues like #101, #81 or #8 or go to [vGPU (SR-IOV) with Intel 12th Gen iGPU](https://www.michaelstinkerings.org/gpu-virtualization-with-intel-12th-gen-igpu-uhd-730/), where I learned some of the things necessary to get it up somewhat working.

> @masiv1001 The instructions can be found here: https://github.com/sihawken/i915-sriov-kmod/

I've since successfully installed it but it stopped working ever since I updated the kernel on fedora 39 (now running 6.5)

EverlastingOS, 2023-11-26:
> @EverlastingOS You can test is by watching the output of the following commands (creating 2 test virtual devices):
>
> ```
> sudo su
> echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
> ```
>
> It [...]

hey, thanks for reply, yes im running fedora 39 also, and my laptop do freeze when typing " sudo su
echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs"

masiv1001, 2023-11-27:
@EverlastingOS I could not get it working on the new fedora release so I don't know if it is possible at all. Your pc freezing might indicate it didn't work, still, do your own research on the topic, you might find something I did not, good luck!

resiliencer, 2023-11-30:
I was unable to compile the dkms i915 module in Alpine Linux, so I did that in Arch Linux and then I took successfully compiled kernel and attached it to my Alpine linux (to do this, just create a separate item in the grub menu and copy the desired folder from /lib/ modules). So I am currently running Alpine with Arch kernel with dkms and i915 mod lol) Works fine.

bolzerrr, 2024-04-18:
I did compile it for alpine with following steps:

I had to update minor version in https://github.com/strongtz/i915-sriov-dkms/blob/master/drivers/gpu/drm/i915/gt/uc/abi/guc_version_abi.h and remove a compile flag ( -DCONFIG_DRM_I915_PXP) in https://github.com/strongtz/i915-sriov-dkms/blob/master/Makefile

```sh
make -C /lib/modules/$(uname -r)/build M=$(pwd) modules -j4
mv i915.ko i915_sriov.ko
cp i915_sriov.ko /lib/modules/$(uname -r)/extra/
depmod -a
echo "blacklist i915" | tee /etc/modprobe.d/i915-blacklist.conf
echo "i915_sriov" |  tee -a /etc/modules
 vi /etc/default/grub ## i915.enable_guc=3 
update-grub
```

closed 2024-12-29 by strongtz