strongtz/i915-sriov-dkms #108: IGPU Intel 12th Gen Not Working

HPMSGEN8, 2023-09-03:
Good Day,

I do not know what i am doing wrong or what is missing but i am getting the below:

`[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.2.16-10-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on i915.enable_guc=3 i915.max_vfs=7
[    0.034171] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.2.16-10-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on i915.enable_guc=3 i915.max_vfs=7
[    3.226318] i915: module verification failed: signature and/or required key missing - tainting kernel
[    3.406827] i915 0000:00:02.0: Running in SR-IOV PF mode
[    3.407680] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    3.407767] i915 0000:00:02.0: vgaarb: deactivate vga console
[    3.407810] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    3.408428] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    3.409901] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
[    3.417987] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[    3.422870] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.bin version 70.5.1
[    3.422876] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[    3.425564] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads!
[    3.425794] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    3.425795] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    3.426098] i915 0000:00:02.0: [drm] GuC RC: enabled
[    3.426675] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[    3.426799] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    3.449831] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    3.463711] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    3.463885] i915 0000:00:02.0: 7 VFs could be associated with this PF
[    3.463934] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    3.485183] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes`

Thanks

HPMSGEN8, 2023-09-03:
Here is some more info on my iGPU

`00:02.0 VGA compatible controller: Intel Corporation Alder Lake-S GT1 [UHD Graphics 770] (rev 0c) (prog-if 00 [VGA controller])
        Subsystem: Dell Alder Lake-S GT1 [UHD Graphics 770]
        Flags: bus master, fast devsel, latency 0, IRQ 142, IOMMU group 0
        Memory at 6000000000 (64-bit, non-prefetchable) [size=16M]
        Memory at 4000000000 (64-bit, prefetchable) [size=256M]
        I/O ports at 4000 [size=64]
        Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
        Capabilities: [40] Vendor Specific Information: Len=0c <?>
        Capabilities: [70] Express Root Complex Integrated Endpoint, MSI 00
        Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable+ 64bit-
        Capabilities: [d0] Power Management version 2
        Capabilities: [100] Process Address Space ID (PASID)
        Capabilities: [200] Address Translation Service (ATS)
        Capabilities: [300] Page Request Interface (PRI)
        Capabilities: [320] Single Root I/O Virtualization (SR-IOV)
        Kernel driver in use: i915
        Kernel modules: i915`

Nightmare41241, 2024-10-05:
did u add numvfs in sysfs or try to change the kernel use

closed 2024-12-29 by strongtz