strongtz/i915-sriov-dkms #109: compile failed on kernel 6.5.2

scyto, 2023-09-14:
No surprise, just logging.

This failed to compile on a 6.5.2 kernel (Ubuntus fork) - i had only applied 2 custom thunderbolt fixes to that so that should be no issue.

```
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_breadcrumbs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_context.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_context_sseu.o
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_display_debugfs.c: In function ‘intel_dp_mst_info’:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_display_debugfs.c:537:47: error: ‘struct drm_dp_mst_port’ has no member named ‘has_audio’
  537 |         bool has_audio = intel_connector->port->has_audio;
      |                                               ^~
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_display_debugfs.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/gen8_ppgtt.c:103:12: warning: ‘gen12_pte_encode_shim’ defined but not used [-Wunused-function]
  103 | static u64 gen12_pte_encode_shim(dma_addr_t addr,
      |            ^~~~~~~~~~~~~~~~~~~~~
make[1]: *** [/usr/src/linux-headers-6.5.2-tbfixes-v2+/Makefile:2032: /var/lib/dkms/i915-sriov-dkms/6.1/build] Error 2
make: *** [Makefile:234: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.5.2-tbfixes-v2+'
```

MotherOfTheGracchi, 2023-09-16:
I updated the driver to build on kernel 6.5.3 (PR https://github.com/strongtz/i915-sriov-dkms/pull/111). I'd be grateful for some testing/feedback.

scyto, 2023-09-17:
> I updated the driver to build on kernel 6.5.3 (PR https://github.com/strongtz/i915-sriov-dkms/pull/111). I'd be grateful for some testing/feedback.

Will do, do you need me to use 6.5.3 or will 6.5.2 do?
(Just so I know if I have to recompile the whole kernel).

MotherOfTheGracchi, 2023-09-17:
You can test on 6.5.2.

scyto, 2023-09-17:
No errors on compile, no errors in `demesg | grep i915`
All 7 VFs lit back up as good in y proxmox resource pool
I will test with guest later.

Closing. Will repopen if i get any immediate errors in my windows VM.

scyto, 2023-09-17:
I was inspired to try and get this working with a debian 12 VM (previoulsy i had only done windows)

I get the following, does this mean Debian 12 has the same issues as Debian 11 and needs special instructions?
(if so i will just switch to some other fork for testing the guest).

[image]

scyto, 2023-09-19:
I compiled in a manjaro guest VM and it seems to be working:

[image]

scyto, 2023-09-19:
reopening as i don't think everything is working right, i can;t get vainfo do what it should on host or guest

this is host:

```
[alex@majaroGPU ~]$ sudo vainfo
[sudo] password for alex: 
Trying display: wayland
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
Trying display: x11
error: can't connect to X server!
Trying display: drm
libva error: /usr/lib/dri/virtio_gpu_drv_video.so init failed
vaInitialize failed with error code 2 (resource allocation failed),exit
[alex@majaroGPU ~]$ sudo dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.5-x86_64 root=UUID=bdabaf4a-8237-4466-93af-36beecfa40d5 rw quiet splash udev.log_priority=3 intel_iommu=on iommu=pt i915.enable_guc=3
[    0.014326] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.5-x86_64 root=UUID=bdabaf4a-8237-4466-93af-36beecfa40d5 rw quiet splash udev.log_priority=3 intel_iommu=on iommu=pt i915.enable_guc=3
[    2.228218] i915: loading out-of-tree module taints kernel.
[    2.228243] i915: module verification failed: signature and/or required key missing - tainting kernel
[    2.426704] i915 0000:01:00.0: Running in SR-IOV VF mode
[    2.429224] i915 0000:01:00.0: [drm] GT0: GUC: interface version 0.1.0.0
[    2.430921] i915 0000:01:00.0: [drm] Using Transparent Hugepages
[    2.433317] i915 0000:01:00.0: [drm] GT0: GUC: interface version 0.1.0.0
[    2.434020] i915 0000:01:00.0: GuC firmware PRELOADED version 1.0 submission:SR-IOV VF
[    2.434024] i915 0000:01:00.0: HuC firmware PRELOADED
[    2.439327] i915 0000:01:00.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    2.439336] i915 0000:01:00.0: [drm] PMU not supported for this GPU.
[    2.439401] [drm] Initialized i915 1.6.0 20201103 for 0000:01:00.0 on minor 0
[    5.033551] i915 0000:01:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[alex@majaroGPU ~]$ sudo  lspci | grep -i vga
00:01.0 VGA compatible controller: Red Hat, Inc. Virtio 1.0 GPU (rev 01)
01:00.0 VGA compatible controller: Intel Corporation Raptor Lake-P [Iris Xe Graphics] (rev 04)
[alex@majaroGPU ~]$ lspci -vs 01:00.0
01:00.0 VGA compatible controller: Intel Corporation Raptor Lake-P [Iris Xe Graphics] (rev 04) (prog-if 00 [VGA controller])
        Subsystem: Intel Corporation Raptor Lake-P [Iris Xe Graphics]
        Physical Slot: 0
        Flags: bus master, fast devsel, latency 0, IRQ 46
        Memory at c0000000 (64-bit, non-prefetchable) [size=16M]
        Memory at 7000000000 (64-bit, prefetchable) [size=512M]
        Capabilities: <access denied>
        Kernel driver in use: i915
        Kernel modules: i915

[alex@majaroGPU ~]$ sudo dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.5-x86_64 root=UUID=bdabaf4a-8237-4466-93af-36beecfa40d5 rw quiet splash udev.log_priority=3 intel_iommu=on iommu=pt i915.enable_guc=3
[    0.014326] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.5-x86_64 root=UUID=bdabaf4a-8237-4466-93af-36beecfa40d5 rw quiet splash udev.log_priority=3 intel_iommu=on iommu=pt i915.enable_guc=3
[    2.228218] i915: loading out-of-tree module taints kernel.
[    2.228243] i915: module verification failed: signature and/or required key missing - tainting kernel
[    2.426704] i915 0000:01:00.0: Running in SR-IOV VF mode
[    2.429224] i915 0000:01:00.0: [drm] GT0: GUC: interface version 0.1.0.0
[    2.430921] i915 0000:01:00.0: [drm] Using Transparent Hugepages
[    2.433317] i915 0000:01:00.0: [drm] GT0: GUC: interface version 0.1.0.0
[    2.434020] i915 0000:01:00.0: GuC firmware PRELOADED version 1.0 submission:SR-IOV VF
[    2.434024] i915 0000:01:00.0: HuC firmware PRELOADED
[    2.439327] i915 0000:01:00.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    2.439336] i915 0000:01:00.0: [drm] PMU not supported for this GPU.
[    2.439401] [drm] Initialized i915 1.6.0 20201103 for 0000:01:00.0 on minor 0
[    5.033551] i915 0000:01:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[alex@majaroGPU ~]$ sudo ls /dev/dri/render*
/dev/dri/renderD128  /dev/dri/renderD129
[alex@majaroGPU ~]$ sudo vainfo
Trying display: wayland
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
Trying display: x11
error: can't connect to X server!
Trying display: drm
libva error: /usr/lib/dri/virtio_gpu_drv_video.so init failed
```

scyto, 2023-09-19:
this is host (i assume i failed to install something or compile something right on the host?)

```
root@pve1:/etc/pve/qemu-server# vainfo
error: can't connect to X server!
libva info: VA-API version 1.17.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: va_openDriver() returns -1
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
```

scyto, 2023-09-20:
closing as this a driver issue not a compile on 6.5 issue

closed 2023-09-20 by scyto