strongtz/i915-sriov-dkms #110: update-initramfs -u WARNING

GavinIves, 2023-09-15:
CPU:13-12100
The host is PVE and the kernel is 6.2. The virtual machine is ubuntu server 22.04 and the kernel is 6.2. Everything was normal according to the steps, and finally the virtual machine successfully obtained the nuclear display. At present, it seems that everything is normal, but when I executed update-initramfs -u on the virtual machine, the following warning occurred. I've searched the internet and haven't found a solution so far. I wonder if it affects functionality.

`root@production:/lib/firmware/i915# update-initramfs -uupdate-initramfs: Generating /boot/initrd.img-6.2.0-32-generic
W: Possible missing firmware /lib/firmware/i915/mtl_gsc_102.0.0.1511.bin for module i915
W: Possible missing firmware /lib/firmware/i915/mtl_huc_8.4.3_gsc.bin for module i915
W: Possible missing firmware /lib/firmware/i915/mtl_guc_70.6.4.bin for module i915
W: Possible missing firmware /lib/firmware/i915/adlp_dmc.bin for module i915
root@production:/lib/firmware/i915# `

GavinIves, 2023-09-17:
i find some information
[image]
maybe just name different??

scyto, 2023-09-20:
can you  solve with this https://unix.stackexchange.com/questions/556946/possible-missing-firmware-lib-firmware-i915-for-module-i915

scyto, 2023-09-20:
also try this

```
mkdir firmware
cd firmware
wget -r -nd -e robots=no -A '*.bin' --accept-regex '/plain/' https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915/
sudo mv *.bin /lib/firmware/i915/
sudo update-initramfs -c -k all
```

GavinIves, 2023-09-21:
its work
but we need rename *.bin

closed 2023-09-21 by GavinIves

GavinIves, 2023-09-21:
These missing files must be installed. It will affect emby's hardware transcoding. Although emby will show that hardware transcoding is successfully enabled, its efficiency is very low. For transcoding 4khdr into 1080sdr, there are only 40 frames without installing these files. After that, there are 80 frames.

HibernantBear, 2024-09-08:
> These missing files must be installed. It will affect emby's hardware transcoding. Although emby will show that hardware transcoding is successfully enabled, its efficiency is very low. For [...]

I have met the same problem as you, I try download these files metioned before, but none of them meet the same name as I missing(these files that I'm missing are all the same name as yours), so could you please tell me how to rename them? and do they really acts fine? thank you very much