strongtz/i915-sriov-dkms #115: Failed to compile i915-sriov module in kernel 4.4.302

fm365, 2023-10-06:
I tried to compile i915-sriov module for 4.4.302 kernel ,but failed , and got the following error :

```
DKMS make.log for i915-sriov-dkms-4.4 for kernel 4.4.302 (x86_64)
Fri Oct  6 11:09:36 CST 2023
make: Entering directory '/root/kernel-4.4.302/linux-4.4.302'
  LD      /var/lib/dkms/i915-sriov-dkms/4.4/build/built-in.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/4.4/build/drivers/gpu/drm/i915/i915_driver.o
In file included from include/linux/acpi.h:25:0,
                 from /var/lib/dkms/i915-sriov-dkms/4.4/build/drivers/gpu/drm/i915/i915_driver.c:30:
/var/lib/dkms/i915-sriov-dkms/4.4/build/include/linux/ioport.h:15:26: fatal error: linux/minmax.h: No such file or directory
compilation terminated.
scripts/Makefile.build:280: recipe for target '/var/lib/dkms/i915-sriov-dkms/4.4/build/drivers/gpu/drm/i915/i915_driver.o' failed
make[1]: *** [/var/lib/dkms/i915-sriov-dkms/4.4/build/drivers/gpu/drm/i915/i915_driver.o] Error 1
Makefile:1453: recipe for target '_module_/var/lib/dkms/i915-sriov-dkms/4.4/build' failed
make: *** [_module_/var/lib/dkms/i915-sriov-dkms/4.4/build] Error 2
make: Leaving directory '/root/kernel-4.4.302/linux-4.4.302'
```

fm365, 2023-10-06:
Does it mean  that this module source code is only for kernel 6.1 ?

strongtz, 2023-10-06:
Your kernel is probably way too old.

simcop2387, 2023-10-25:
Kernel series 4.4 was EOL'd back in Feb 2022 so it's not receiving updates from upstream kernel development anymore.  It also originally was released back in 2016.  Almost certainly going to have many kernel api changes since then that just won't be compatible.

closed 2024-01-29 by strongtz